mavlink/mavros #185: PixHawk timestamp instead of ROS time

thedinuka, 2015-01-07:
I was looking at imu_pub.cpp and realized that imu/data_raw  topic is published with the header timestamp set to ROS::Time::now() 
(https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/imu_pub.cpp#L250)

The MAVLINK_MSG_ID_HIGHRES_IMU message from PixHawk does contain a time_usec filed which actually represent the time at which the imu data was captured. For certain applications like for synchronizing a camera and IMU, it maybe useful to have that as the timestamp instead of the ros time. 

I'm wondering whether this would qualify as a new feature request? I could do a PR for this, but the change in the code is really simple anyway.

mhkabir, 2015-01-07:
Hi, 
We have timesync working. I will be updating the plugins soon to take advantage of this. We synchronise Pixhawk timestamps to ROS time and vice-versa.
What you propose would not be ideal, as ROS wall time is calculated from Unix Epoch and Pixhawk time is calculated from boot.

thedinuka, 2015-01-07:
@mhkabir timesync would be ideal, I agree. What sort of accuracies can you achieve with it?

mhkabir, 2015-01-07:
We get microsecond precision.

thedinuka, 2015-01-07:
ahh, excellent. would you be able to comment on when this will be available?

mhkabir, 2015-01-07:
Pretty soon hopefully. Synchronisation works already (check system_time.cpp ) The plugins just need to be updated.

thedinuka, 2015-01-07:
@mhkabir I'm still reading through system_time.cpp (actually  it's sys_time.cpp right?) but as a preliminary question, is the idea to maintain the difference between ros time and PixHawk time and then use that offset in other plugins to set the correct timestamp?

mhkabir, 2015-01-07:
Yes

vooon, 2015-01-16:
I testing PR #186.

Setup:
- Wandboard Quad with Ubuntu 14.04 and latest ROS for UbuntuARM
- Pixhawk with latest master

Results:
First run without custom `-m onboard` mode.
Got timsync, but strange timestamps:

*/mavros/imu/data*
```yaml
---
header: 
  seq: 4316
  stamp: 
    secs: 4294967295
    nsecs: 4294967295
  frame_id: fcu
orientation: 
  x: -0.0430093572484
  y: 0.00982332692738
  z: -0.307822874007
  w: 0.950420315269
orientation_covariance: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
angular_velocity: 
  x: 0.0105894114822
  y: -0.0485693067312
  z: -0.010234859772
angular_velocity_covariance: [1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07]
linear_acceleration: 
  x: 0.0641356706619
  y: -0.864281475544
  z: 10.0234670639
linear_acceleration_covariance: [8.999999999999999e-08, 0.0, 0.0, 0.0, 8.999999999999999e-08, 0.0, 0.0, 0.0, 8.999999999999999e-08]
---
header: 
  seq: 4317
  stamp: 
    secs: 4294967295
    nsecs: 4294967295
  frame_id: fcu
orientation: 
  x: -0.0430188834137
  y: 0.00986130963341
  z: -0.307839425409
  w: 0.950414129948
orientation_covariance: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
angular_velocity: 
  x: 0.00931540410966
  y: -0.0476437397301
  z: -0.00976792909205
angular_velocity_covariance: [1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07]
linear_acceleration: 
  x: 0.0641356706619
  y: -0.864281475544
  z: 10.0234670639
linear_acceleration_covariance: [8.999999999999999e-08, 0.0, 0.0, 0.0, 8.999999999999999e-08, 0.0, 0.0, 0.0, 8.999999999999999e-08]
---

```

*/diagnostics*
```yaml
  - 
    level: 0
    name: mavros: Time Sync
    message: Normal
    hardware_id: Mavlink
    values: 
      - 
        key: Events in window
        value: 100
      - 
        key: Events since startup
        value: 1361
      - 
        key: Duration of window (s)
        value: 10.007385
      - 
        key: Actual frequency (Hz)
        value: 9.992621
      - 
        key: Last dt (ms)
        value: 0.550172
      - 
        key: Mean dt (ms)
        value: -0.000695
      - 
        key: Last system time (s)
        value: 204.922809000
      - 
        key: Time offset (s)
        value: 1421389669.363838673
```

vooon, 2015-01-16:
I placed this to */fs/microsd/etc/extras.txt*:
```sh
#!nsh
#
# USB MAVLink start
#

mavlink start -r 20000 -d /dev/ttyACM0 -x -m onboard
# Enable a number of interesting streams we want via USB
mavlink stream -d /dev/ttyACM0 -s PARAM_VALUE -r 200
usleep 100000
mavlink stream -d /dev/ttyACM0 -s MISSION_ITEM -r 50
usleep 100000
mavlink stream -d /dev/ttyACM0 -s NAMED_VALUE_FLOAT -r 10
usleep 100000
mavlink stream -d /dev/ttyACM0 -s OPTICAL_FLOW -r 10
usleep 100000
mavlink stream -d /dev/ttyACM0 -s VFR_HUD -r 20
usleep 100000
mavlink stream -d /dev/ttyACM0 -s ATTITUDE -r 20
usleep 100000
mavlink stream -d /dev/ttyACM0 -s ATTITUDE_CONTROLS -r 30
usleep 100000
mavlink stream -d /dev/ttyACM0 -s RC_CHANNELS_RAW -r 5
usleep 100000
mavlink stream -d /dev/ttyACM0 -s SERVO_OUTPUT_RAW_0 -r 20
usleep 100000
mavlink stream -d /dev/ttyACM0 -s POSITION_TARGET_GLOBAL_INT -r 10
usleep 100000
mavlink stream -d /dev/ttyACM0 -s LOCAL_POSITION_NED -r 30
usleep 100000
mavlink stream -d /dev/ttyACM0 -s MANUAL_CONTROL -r 5

# Exit shell to make it available to MAVLink
exit
```

And checked that mavlink starts from fcu bootup (px4 quirk not needed, but enabled).
But get similar resuilt: 4294967295 constant in timestamp

Also strange operation errors:
```
[ INFO] [1421390635.313224536]: MAVROS started. MY ID [1, 240], TARGET ID [1, 50]
[ WARN] [1421390635.318253761]: TM: Clock skew detected (-1421390607.381244898 s). Hard syncing clocks.
[ INFO] [1421390635.388229088]: CON: Got HEARTBEAT, connected.
[ WARN] [1421390646.390623651]: PR: request list timeout, retries left 2
[ WARN] [1421390647.392587524]: PR: request list timeout, retries left 1
[ WARN] [1421390648.393595986]: PR: request list timeout, retries left 0
[ WARN] [1421390651.391841876]: WP: timeout, retries left 2
[ INFO] [1421390651.907876513]: WP: seems GCS requesting mission
[ INFO] [1421390651.909298605]: WP: sheduling pull after GCS is done
[ WARN] [1421390652.401791065]: WP: timeout, retries left 2
[ INFO] [1421390652.414816520]: WP: seems GCS requesting mission
[ INFO] [1421390652.415519210]: WP: sheduling pull after GCS is done
[ INFO] [1421390652.936081782]: WP: seems GCS requesting mission
[ INFO] [1421390652.936835230]: WP: sheduling pull after GCS is done
[ WARN] [1421390653.404124672]: WP: timeout, retries left 1
[ INFO] [1421390653.451881471]: WP: seems GCS requesting mission
[ INFO] [1421390653.452306833]: WP: sheduling pull after GCS is done
[ INFO] [1421390653.973946167]: WP: seems GCS requesting mission
[ INFO] [1421390653.974379332]: WP: sheduling pull after GCS is done
[ WARN] [1421390654.405906896]: WP: timeout, retries left 0
[ INFO] [1421390654.479033400]: WP: seems GCS requesting mission
[ INFO] [1421390654.479452293]: WP: sheduling pull after GCS is done
[ INFO] [1421390654.995982050]: WP: seems GCS requesting mission
[ INFO] [1421390654.997416385]: WP: sheduling pull after GCS is done
[ERROR] [1421390655.409012207]: WP: timed out.
[ INFO] [1421390655.499020671]: WP: seems GCS requesting mission
[ INFO] [1421390655.499445457]: WP: sheduling pull after GCS is done
[ INFO] [1421390656.006200442]: WP: seems GCS requesting mission
[ INFO] [1421390656.006822432]: WP: sheduling pull after GCS is done
[ERROR] [1421390656.452320919]: FCU: Operation timeout
[ WARN] [1421390662.010983050]: WP: timeout, retries left 2
[ WARN] [1421390663.012967000]: WP: timeout, retries left 1
[ WARN] [1421390664.015000102]: WP: timeout, retries left 0
[ERROR] [1421390665.017205422]: WP: timed out.
```

vooon, 2015-01-16:
Bug in `syncronise_stamp()`. I will fix it.

vooon, 2015-01-16:
@mhkabir pull request closed, your patch rebased in master.

Time is close to ros wall time in range ±5 ms by my quick test.

mhkabir, 2015-01-16:
Awesome.

vooon, 2015-01-24:
I think we may close this.

closed 2015-01-24 by vooon