mavlink/mavros #188: Pixhawk, QGroundControl, Mavros

fabiorubino, 2015-01-19:
Hi all,
I am trying to reading the IMU data from the Pixhawk through Mavros. The problem is that at every restart of Pixhawk I have to connect it to QGroundControl, disconnect it, and then I can run the mavros node and read the data. If I don't connect the autopilot to QGC the mavros node only run the plugins. Why?
How can I solve this problem?

bests 
Fabio Rubino

LorenzMeier, 2015-01-19:
Because the USB link is not intended to be used with a companion computer and defaults to the NSH shell. Please use a FTDI cable on TELEM2 instead and start a MAVLink instance on that port. Please refer to the px4 users mailing list for further assistance on how to do this.

closed 2015-01-19 by LorenzMeier

fabiorubino, 2015-01-19:
Thanks Lorenz for your fast support.
But can I read the data with Mavros using the ftdi cable? If yes how? Can I reach the 120hz of rate?

Bests
Fabio