mavlink/mavros #189: Mavros readme and mavlink-gbp-release

fabiorubino, 2015-01-20:
Hi all,
I need to install the mavros package by source in a catkin workspace in a PC with INDIGO and UBUNTU 14.04 . I read in the README.md that the mavros package is dependent on ros-*-mavlink that is build from mavlink-gbp-release. I tried to install by source, because I have to do in this way, the mavlink package and the mavros package. After I tried to compile with catkin_make the workspace. CMake fails because  the mavlink package is not a catkin package.

I used for the mavlink package:
-this URL https://github.com/mavlink/mavlink-gbp-release.git
-this branch release/indigo/mavlink

I used for the mavros package:
-this URL  https://github.com/mavlink/mavros.git
-this branch master

Can you help me, please?

Best wishes

Fabio

vooon, 2015-01-21:
Do step-by-step script listed in installation instruction. Mavlink package require their own workspace and isolated build (like ros source installation).

fabiorubino, 2015-01-24:
Hi vooon, 
yes, installing step-by-step it works.Thank you very much.

Bests
Fabio

closed 2015-01-24 by vooon