mavlink/mavros #190: Pixhawk driver

fabiorubino, 2015-01-23:
Hi all,
I need to create an interface for the Pixhawk. It has to be connected to a flight control stack different from the PX4 one. What I need is to send the roll, pitch, thrust and the derivative of yaw commands and from it I need to read the rotation angles, battery status, all data. I need a driver package. The stack is based on UBUNTU 14.04 and on ROS INDIGO. 

Can you help me?

Bests
Fabio Rubino

vooon, 2015-01-23:
Mavros can communicate with any FCU with common.xml message set (dialect).
But you must declare what messages are used for RPY-T.

fabiorubino, 2015-01-26:
@vooon 
Do yo know what Mavros file or files I have to manage for sending roll, pitch, derivative of yaw and thrust commands to the Pixhawk directly from my stack? And for sending back the IMU sensor data directly to my stack bypassing the Mavros controller?
Thank you for your help.

bests
Fabio

vooon, 2015-01-27:
@fabiorubino it depends on what mavlink message you want to send.

fabiorubino, 2015-01-27:
Thank you for your help but now I want to fly autonomously using only the mavros stack. I have a quadrotor with Odroid that is connected to Pixhawk and PX4FLOW. In the Odroid I can read through mavros the imu data from Pixhawk and the optical_flow from PX4FLOW in the same time using two different launchers. How can I send commands to Pixhawk for flying?

bests 
Fabio

vooon, 2015-01-27:
1. Please describe connections of Odroid, PX4 and PX4Flow. What module connects with other, and what channel it uses
2. What exactly commands exactly you want to send?

fabiorubino, 2015-01-27:
Well, I connect to Odroid through one USB port the Pixhawk and through another USB port the PX4FLOW. Both of them use the usb port for communicating. To read the data I use for the PIxhawk the command "roslaunch mavros px4.launch" and for the PX4FLOW the command "roslaunch mavros_extra px4flow.launch". 
The Odroid is powered by means the cable, the pixhawk is powered by means the POWER port, the PX4FLOW is powered by means UART3 port.
 I want to take off, hover and land. It would be useful also to send the waypoints.

bests
Fabio

mhkabir, 2015-01-28:
@fabiorubino Please post about this on the px4users mailing list. If you had actually gone through the wiki as I keep telling you repeatedly, you would've seen that there is so need for all the shenanigans. All information is on the wiki : 
1. https://pixhawk.org/modules/px4flow
2. https://pixhawk.org/dev/offboard_control
You don't need to connect the flow via USB. Just connect it via the i2c bus on the Pixhawk. Please continue this on Gitter if you face further issues. You need mavros just for sending the control commands, not flow position estimation.

fabiorubino, 2015-01-28:
Thank you for your support. I have already put the same question on the px4users mailing list. In any case I tried to connect the PX4FLOW through i2c, but in mavros I can't read data from mavros/optical_flow  topic. The topic is subscribed, but I can't see anything. In future if I will have other problems I will write on gitter as you want.
Thanks for all

bests
Fabio

mhkabir, 2015-01-28:
/mavros/optical_flow is deprecated. It doesn't work with the new interface on i2c. I can fix that and put in a message type for the new interface on Saturday though. 
Is there any particular reason you need to view the flow data? Position estimation will work fine even without mavros.

fabiorubino, 2015-01-28:
So using the I2C interface don't I need to subscribe to mavros/optical_flow to sends data to the Pixhawk controller? Can I fly autonomously only sending command through mavros to the Pixhawk controller?
If I want to collect the PX4FLOW data how can I do?

vooon, 2015-01-28:
As i remember px4 resends flow data via telemetry link, so just remove px4flow plugin from black list.
Via I2C you only have flow data, no image, but actually i think it is not needed.

mhkabir, 2015-01-28:
Actually Vladimir, no. It does not. Also the new OPTICAL_FLOW_RAD message
is not support in mavros yet?

I think I forgot to do a PR. I'll do it later.

Fabio, you don't need to do anything of that sort to fly with the PX4Flow.
Just connect i2c and send commands from mavros.

Why do you need to collect flow data? You can get current computed local
position in mavros/position/local if you need that.

fabiorubino, 2015-01-28:
Thank you all. I will do as you described.

Bests
Fabio

mhkabir, 2015-01-29:
Closing this. Please ask on the mailing list or Gitter IM if you have further questions. Thanks :)
https://gitter.im/mavlink/mavros
https://gitter.im/PX4/Firmware

closed 2015-01-29 by mhkabir