mavlink/mavros #191: Can't locate mavrod_node

dpnebert, 2015-01-23:
I am able to install and use ROS (http://wiki.ros.org/indigo/Installation/Ubuntu)

Then I install MAVROS (https://github.com/mavlink/mavros) using the console command, 'sudo git clone https://github.com/mavlink/mavros.git'

I add the directory I installed mavros to the environment variable ROS_PACKAGE_PATH.

I am able to build the package using, 'rosmake mavros' which echos:

============================================================
dan@dan-HP-ProBook-6470b:~/indigo_workspace/mavros/mavros$ rosmake mavros
[ rosmake ] rosmake starting...                                                                                                                
[ rosmake ] Packages requested are: ['mavros']                                                                                                 
[ rosmake ] Logging to directory /home/dan/.ros/rosmake/rosmake_output-20150123-141534                                                         
[ rosmake ] Expanded args ['mavros'] to:
['mavros']                                                                                            
[rosmake-0] Starting >>> catkin [ make ]                                                                                                       
[rosmake-0] Finished <<< catkin ROS_NOBUILD in package catkin
 No Makefile in package catkin                                                   
[rosmake-0] Starting >>> genmsg [ make ]                                                                                                       
[rosmake-1] Starting >>> cpp_common [ make ]                                                                                                   
[rosmake-2] Starting >>> cmake_modules [ make ]                                                                                                
[rosmake-3] Starting >>> libmavconn [ make ]                                                                                                   
[rosmake-2] Finished <<< cmake_modules ROS_NOBUILD in package cmake_modules
 No Makefile in package cmake_modules                              
[rosmake-0] Finished <<< genmsg ROS_NOBUILD in package genmsg
 No Makefile in package genmsg                                                   
[rosmake-0] Starting >>> genlisp [ make ]                                                                                                      
[rosmake-2] Starting >>> genpy [ make ]                                                                                                        
[rosmake-1] Finished <<< cpp_common ROS_NOBUILD in package cpp_common
 No Makefile in package cpp_common                                       
[rosmake-1] Starting >>> gencpp [ make ]                                                                                                       
[rosmake-0] Finished <<< genlisp ROS_NOBUILD in package genlisp
 No Makefile in package genlisp                                                
[rosmake-0] Starting >>> rostime [ make ]                                                                                                      
[rosmake-0] Finished <<< rostime ROS_NOBUILD in package rostime
 No Makefile in package rostime                                                
[rosmake-0] Starting >>> roscpp_traits [ make ]                                                                                                
[rosmake-3] Finished <<< libmavconn ROS_NOBUILD in package libmavconn
 No Makefile in package libmavconn                                       
[rosmake-3] Starting >>> rospack [ make ]                                                                                                      
[rosmake-2] Finished <<< genpy ROS_NOBUILD in package genpy [ roscpp_traits: 0.0 sec ] [ rospack: 0.0 sec ]           [ 4 Active 7/58 Complete ]
 No Makefile in package genpy                                                      
[rosmake-1] Finished <<< gencpp ROS_NOBUILD in package gencpp
 No Makefile in package gencpp                                                   
[rosmake-1] Starting >>> roslang [ make ]                                                                                                      
[rosmake-0] Finished <<< roscpp_traits ROS_NOBUILD in package roscpp_traits
 No Makefile in package roscpp_traits                              
[rosmake-0] Starting >>> roscpp_serialization [ make ]                                                                                         
[rosmake-2] Starting >>> message_generation [ make ]                                                                                           
[rosmake-3] Finished <<< rospack ROS_NOBUILD in package rospack
 No Makefile in package rospack                                                
[rosmake-3] Starting >>> roslib [ make ]                                                                                                       
[rosmake-1] Finished <<< roslang ROS_NOBUILD in package roslang
 No Makefile in package roslang                                                
[rosmake-1] Starting >>> xmlrpcpp [ make ]                                                                                                     
[rosmake-0] Finished <<< roscpp_serialization ROS_NOBUILD in package roscpp_serialization
 No Makefile in package roscpp_serialization         
[rosmake-0] Starting >>> message_runtime [ make ]                                                                                              
[rosmake-3] Finished <<< roslib ROS_NOBUILD in package roslib
 No Makefile in package roslib                                                   
[rosmake-3] Starting >>> rosunit [ make ]                                                                                                      
[rosmake-1] Finished <<< xmlrpcpp ROS_NOBUILD in package xmlrpcpp
 No Makefile in package xmlrpcpp                                             
[rosmake-1] Starting >>> class_loader [ make ]                                                                                                 
[rosmake-2] Finished <<< message_generation ROS_NOBUILD in package message_generation
 No Makefile in package message_generation               
[rosmake-2] Starting >>> rosgraph [ make ]                                                                                                     
[rosmake-3] Finished <<< rosunit ROS_NOBUILD in package rosunit
 No Makefile in package rosunit                                                
[rosmake-3] Starting >>> rosclean [ make ]                                                                                                     
[... 154 lines cut ...]
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/mission/pull_after_gcs: True
 * /mavros/plugin_blacklist: ['3dr_radio']
 * /mavros/startup_px4_usb_quirk: True
 * /mavros/sys/min_voltage: 10.0
 * /mavros/target_component_id: 50
 * /mavros/target_system_id: 1
 * /rosdistro: indigo
 * /rosversion: 1.11.10

NODES
  /
    mavros (mavros/mavros_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
ERROR: cannot launch node of type [mavros/mavros_node]: can't locate node [mavros_node] in package [mavros]
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete
=============================================================
Note: I am not connected to the drone in anyway.  I am simply trying to get far enough through this that I can see the motors spin on the drone.  At what point should I try to get the FCU and GCS address entered in?  Or is that my problem?
For some reason, I can't figure out how to create a MAVROS_NODE object.  If anyone has any suggestion, I'd appreciate it.

vooon, 2015-01-24:
Did you read this: [README Installation](https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation)?
Mavros is catkin project, not rosbuild.

dpnebert, 2015-01-26:
I did not read that far, I guess.  I see that now with the link you gave.  Thanks for the help, still new to all of this.

closed 2015-01-28 by dpnebert