mavlink/mavros #193: Convert body-fixed IMU data to Inertial frame for Pose messages

mhkabir, 2015-01-24:


LorenzMeier, 2015-04-09:
@vooon @mhkabir Are you using this message yet? http://mavlink.org/messages/common#LOCAL_POSITION_NED_COV

I'm wondering if we should stick acceleration in earth frame into it as well, and cut the covariance to the upper triangular. That would end up being 36 floats, which is still manageable (144 bytes).

TSC21, 2015-04-09:
@LorenzMeier that was an open issue (#94). From what I know, it's still not in use.
Seems a good add though, since it manages with position estimate uncertainty.

mhkabir, 2015-04-09:
No we aren't. No covariance from the PX4 estimation yet.

mhkabir, 2015-04-09:
And this issue is separate, I think. The thing here, the ROS Pose message expects attitude in the inertial frame. My flight visualiser thus was getting the track inverted. 
The IMU message expects it in body frame.

TSC21, 2015-04-09:
@mhkabir from what I remember in my thesis I had to invert the yaw axis so to sync both frames, as I was getting na invertion on Rviz.

mhkabir, 2015-04-09:
Yeah. As I said. It expects it in inertial frame.

TSC21, 2015-04-09:
Yeah #195 seems to solve it then :)

mhkabir, 2015-04-09:
It would. If it worked. Which it doesn't. Using tf won't work here.

closed 2015-06-28 by mhkabir