mavlink/mavros #197: From ARMED to TAKEOFF without GPS

dpnebert, 2015-01-28:
Hello again,

I want to make question as simple as I can, and I apologize for being a noob in advance.

My goal is to use my drone indoors without GPS using Optitrack.  I currently do not have any information from Optitrack coming in (yet), I will need to connect these two pieces together once I know what format I need to present MAVROS data in.

I run roscore, then I launch a node with, 'roslaunch mavros px_radio.launch' (I have found the need to add the two commands once it is launched: 'rosparam set use_sim_time true', 'rosservice call /mavros/set_stream_rate 0 10 1', though I don't fully understand why I need that, but I have found it helps.

Through all of my research, I have learned I want to be in GUIDED mode, though I can't find the page I read that on to cite it.  But I used, 'rosrun mavros mavsys -v mode -c 4' to put it in GUIDED mode.

Which brings me to MOCAP.  I set the param ~mocap/use_pose to true, since I want to use Optitrack.  I see two subscribed topics, ~mocap/use_pose and ~mocap/tf.  I assume I want to use ~mocap/pose.  I see the definition contains Point position and Quaternion orientation.  While I understand how to create a msgs file, I don't understand how to create one with that contains two msgs files definitions (I do remember reading something about this, but I don't remember where.)

In short, without GPS and not having the Optitrack info available yet, I just want my drone to take off to about a meter off the ground, then return to landing.  Am I on the right track?

You guys have been a live safer with this forum and I've been trying really hard to avoid getting on here and asking a bunch of noob questions, but I am getting really stuck here.  

You should have seen my face the first time I (successfully) armed the drone and the motors started spinning!! :D

Thanks in advance!

vooon, 2015-01-28:
I think that `rosparam set use_sim_time true` should not be used.
Also instead on calling `rosservice` use `mavsys rate`.

As i understand you use APM (at least same custom modes as ArduCopter), and i don't know actual status of `VICON_POSITION_ESTIMATE` message support.
Perhaps only PX4 native stack support this?

FYI: You can do `mavsys mode -c GUIDED`, if FCU known by mavros. Check `/mavros/state` topic.

dpnebert, 2015-01-28:
I am stepping out for the day, but I will be back tomorrow.

I will avoid setting use_sim_time.
I will try mavsys rate.
I am attemping to use PX4, not APM.  I understand PX4 has more functionality.  Sound right?

I did check the /mavros/state topic, and it did report the mode as GUIDED.

mhkabir, 2015-01-29:
1. What you need is offboard mode. https://pixhawk.org/dev/offboard_control
2. Push the pose estimate from Optitrack to the /mavros/position/vision topic for now. The mocap plugin isn't fully supported on PX4 side yet. The visual position estimate plugin is in mavros_extras. You should use that for now 
3. You cannot take off or hover without a positioning source. 
4. OPTITRACK should publish some form of a tf or Pose message. You just need to reroute this to Mavros. 
@tonybaltovski Can you help him with mocap please. Thanks. 

We'd appreciate it if you can post such issues on the px4users mailing list and not here. I'll be there to help as well, but this project bugtracker isn't a forum. Thanks :)

tonybaltovski, 2015-01-29:
I will give the mocap plugin a test later this week or early next week.  Like @mhkabir said, once you have Optitrack running, you get a TF and/or PoseStamped.  Remap the topic or set the topic name in your config to the `/mavros/position/vision`.  I assume you should be able to just send set point values once in OFFBOARD mode to your desired pose which is a position and orientation (only yaw).  I will report back once I have done some testing.

mhkabir, 2015-01-29:
The mocap plugin doesn't work as inav in PX4 doesn't handle VICON_POSITION_ESTIMATE.

But, the vision position plugin does.

dpnebert, 2015-01-29:
@mhkabir 
1. That link was perfect!  I don't know how I didn't find that in my search previously.  Thank you!  I have found a bunch of missing pages/information when I was on pixhawk.org website.  When I went to the link you gave, and clicked the link at the bottom of that page, 'Offboard Control with Mavros', it brings me to a page with lots of good information.  Although, it seems to be missing a command at the bottom of the page, under the text, 'If the above conditions are satisfied, you can switch to offboard mode using your transmitter switch or using the mavlink mode change command from mavros by running this : '  I want to assume it says, 'mavsys mode -c OFFBOARD'.

2. I will use /mavros/position/vision topic.  I assume I need to start another node, mavros_extras_node

3. I will work to get the Optitrack data to publish data to a topic so I can publish it back out to /mavros/position/vision

4. I think I understand what @tonybaltovski and yourself said about publishing the Optitrack pose data to a topic.  I will work to get the Optitrack data published and I will hopefully be able to trail and error my way from there. 

I will close this issue at the end of the day (giving chance for a few last comments, in case anyone wanted to correct anything I've said), I will start posting to px4users mailing list.  

Again, thank you, @vooon, @mhkabir, and @tonybaltovski for the help, even though I came to the wrong place!

mhkabir, 2015-01-29:
1. I'll add the mavsys command to the wiki.
2. No need to start another node. vision_position is just another plugin.

Glad that we could help. You can also use the mavros gitter here : https://gitter.im/PX4/Firmware if you would like to ask further questions.

closed 2015-01-29 by mhkabir

LorenzMeier, 2015-02-10:
I did throw a few notes in here:
http://www.pixhawk.org/dev/ros/motion_capture

It would be great to get some momentum going to turn this into a real tutorial. All the pointers are there.

mindThomas, 2017-02-15:
@LorenzMeier What happened to the link? We are working on the same topic at the moment therefore looking for ressources.

mhkabir, 2017-02-16:
Everything is here now : https://dev.px4.io/external-position.html