mavlink/mavros #199: gcs_image_bridge.cpp compiling error

TSC21, 2015-01-29:
Guys,

How's things going? I'm coming back to work at the project finally :D Doing some clean up on my laptop. Installed 14.04 with Indigo.
But in my workspace, when doing `catkin_make`, I get this on `gcs_image_bridge.cpp` compilation:
```
[ 98%] Built target mavros_plugins
CMakeFiles/gcs_image_bridge.dir/src/gcs_image_bridge.cpp.o: In function `image_cb(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&)':
gcs_image_bridge.cpp:(.text+0x1077): undefined reference to `cv::imencode(cv::String const&, cv::_InputArray const&, std::vector<unsigned char, std::allocator<unsigned char> >&, std::vector<int, std::allocator<int> > const&)'
CMakeFiles/gcs_image_bridge.dir/src/gcs_image_bridge.cpp.o: In function `cv::String::String(char const*)':
gcs_image_bridge.cpp:(.text._ZN2cv6StringC2EPKc[_ZN2cv6StringC5EPKc]+0x4f): undefined reference to `cv::String::allocate(unsigned long)'
CMakeFiles/gcs_image_bridge.dir/src/gcs_image_bridge.cpp.o: In function `cv::String::~String()':
gcs_image_bridge.cpp:(.text._ZN2cv6StringD2Ev[_ZN2cv6StringD5Ev]+0x14): undefined reference to `cv::String::deallocate()'
CMakeFiles/gcs_image_bridge.dir/src/gcs_image_bridge.cpp.o: In function `cv::String::operator=(cv::String const&)':
gcs_image_bridge.cpp:(.text._ZN2cv6StringaSERKS0_[_ZN2cv6StringaSERKS0_]+0x28): undefined reference to `cv::String::deallocate()'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/nuno/AIMAV_Project/devel/lib/mavros_extras/gcs_image_bridge] Error 1
make[1]: *** [mavros/mavros_extras/CMakeFiles/gcs_image_bridge.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

Maybe opencv related? I installed the version which comes with the indigo repo.
Any advice?

Hope to continue getting some improvements on the project :)

Best,
Nuno

TSC21, 2015-01-29:
BTW,
`pkg-config --libs opencv` on terminal gives me:
```
/opt/ros/indigo/lib/libopencv_calib3d.so /opt/ros/indigo/lib/libopencv_core.so /opt/ros/indigo/lib/libopencv_features2d.so /opt/ros/indigo/lib/libopencv_flann.so /opt/ros/indigo/lib/libopencv_highgui.so /opt/ros/indigo/lib/libopencv_imgcodecs.so /opt/ros/indigo/lib/libopencv_imgproc.so /opt/ros/indigo/lib/libopencv_ml.so /opt/ros/indigo/lib/libopencv_objdetect.so /opt/ros/indigo/lib/libopencv_photo.so /opt/ros/indigo/lib/libopencv_shape.so /opt/ros/indigo/lib/libopencv_stitching.so /opt/ros/indigo/lib/libopencv_superres.so /opt/ros/indigo/lib/libopencv_ts.a /opt/ros/indigo/lib/libopencv_video.so /opt/ros/indigo/lib/libopencv_videoio.so /opt/ros/indigo/lib/libopencv_videostab.so /opt/ros/indigo/lib/libopencv_viz.so
```

TSC21, 2015-01-29:
Update:
Uninstalled the opencv version of Indigo, and installed opencv from source: http://docs.opencv.org/doc/tutorials/introduction/linux_install/linux_install.html
Same result! What can be wrong then?

vooon, 2015-01-29:
Remove your build and devel dirs.

TSC21, 2015-01-30:
Already did it. same stuff

vooon, 2015-01-30:
Why do you use ros version? As i know it wants system library.

My libs:
```
$ ls /usr/lib/x86_64-linux-gnu/libopencv_*
/usr/lib/x86_64-linux-gnu/libopencv_calib3d.a
/usr/lib/x86_64-linux-gnu/libopencv_calib3d.so@
/usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_contrib.a
/usr/lib/x86_64-linux-gnu/libopencv_contrib.so@
/usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_core.a
/usr/lib/x86_64-linux-gnu/libopencv_core.so@
/usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_features2d.a
/usr/lib/x86_64-linux-gnu/libopencv_features2d.so@
/usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_flann.a
/usr/lib/x86_64-linux-gnu/libopencv_flann.so@
/usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_gpu.a
/usr/lib/x86_64-linux-gnu/libopencv_gpu.so@
/usr/lib/x86_64-linux-gnu/libopencv_gpu.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_gpu.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_gpu.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_highgui.a
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so@
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_imgproc.a
/usr/lib/x86_64-linux-gnu/libopencv_imgproc.so@
/usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_legacy.a
/usr/lib/x86_64-linux-gnu/libopencv_legacy.so@
/usr/lib/x86_64-linux-gnu/libopencv_legacy.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_legacy.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_legacy.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_ml.a
/usr/lib/x86_64-linux-gnu/libopencv_ml.so@
/usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_objdetect.a
/usr/lib/x86_64-linux-gnu/libopencv_objdetect.so@
/usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_ocl.a
/usr/lib/x86_64-linux-gnu/libopencv_ocl.so@
/usr/lib/x86_64-linux-gnu/libopencv_ocl.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_ocl.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_ocl.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_photo.a
/usr/lib/x86_64-linux-gnu/libopencv_photo.so@
/usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_stitching.a
/usr/lib/x86_64-linux-gnu/libopencv_stitching.so@
/usr/lib/x86_64-linux-gnu/libopencv_stitching.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_stitching.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_stitching.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_superres.a
/usr/lib/x86_64-linux-gnu/libopencv_superres.so@
/usr/lib/x86_64-linux-gnu/libopencv_superres.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_superres.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_superres.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_ts.a
/usr/lib/x86_64-linux-gnu/libopencv_ts.so@
/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_video.a
/usr/lib/x86_64-linux-gnu/libopencv_video.so@
/usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_videostab.a
/usr/lib/x86_64-linux-gnu/libopencv_videostab.so@
/usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4@
/usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8@
/usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.9
```

Where 2.4.8 links is a fix for building mavros on 14.10 + Indigo.

TSC21, 2015-01-30:
Solved! I had 2.4.8 instead of 2.4.9, Installing 2.4.9 solved. Closing now!

closed 2015-01-30 by TSC21

vooon, 2015-01-30:
Hmm, 2.4.8 is a version in 14.04, and i did that links to solve build deps.

blackcoder, 2015-02-18:
@TSC21  I am facing the same issue on Ubuntu 14.04, how did you install the 2.4.9 Ubuntu 14.04 has only the 2.4.9 !

mhkabir, 2015-02-18:
@blackcoder Do a source install please

vooon, 2015-02-18:
Problem not in opencv version! I used 2.4.9 because it shipped in Ubuntu 14.10.
And it builds fine on jenkins ([indigo trusty 0.10.1][it]), so 2.4.8 is ok.

Not ok is mixing ros opencv package (placed in `/opt/ros`) and system.
Because ros package provides experimental opencv3.

[it]: http://jenkins.ros.org/job/ros-indigo-mavros-extras_binarydeb_trusty_i386/32/consoleFull

blackcoder, 2015-02-21:
@vooon  you are right ! I simply removed ros-indigo-opencv3 and it compiled correctly. 
For future reference, people facing the same problem should do the same: 
```
sudo apt-get remove ros-indigo-opencv3
```

lesleyhaha, 2018-11-20:
> your

Thank you so much!

lesleyhaha, 2018-11-20:
> Remove your build and devel dirs.
Thank you so much!

lesleyhaha, 2018-11-20:
> Remove your build and devel dirs.

Thank you so much!
I solved this problem after:
1. $ sudo apt-get install ros-indigo-opencv3
2. delete /ros_ws/devel and /ros_ws/build
3. $ catkin_make