mavlink/mavros #203: MAVROS Rc Override

dottantgal, 2015-01-30:
hi all
i have an issue using mavros with the PX4.
The "rc chan override" doesn't work using the ArduRover firmware on the PX4: the mavros/rc/override appears on the topic list when i publish and it is full, but the rc/in and rc/out don't change (they have the values sent using the Fly Sky controller).
I was able to override the channel using Roscopter without problems.
Do you have any suggestions?
let me know if you need more infos.
Thanks a lot

vooon, 2015-01-30:
Did you setup GCS system id property on APM?

Overrides only accepts from GCS system Id, so you should tell APM that number.
By default it GCS is 255, mavros desined to be run at OBC as part of the system, so their default 1 (same as APM).

You could:
1. Setup FCU property: `rosrun mavros mavparam set SYSID_MYGCS 1`
2. Use connection URL with `ids=` attribute, e.g. `serial:///dev/ttyACM0:57600?ids=255,240`

dottantgal, 2015-01-30:
Thank you, i've read about it in other issues but if i run that command ther result is
` Couldn't find executable named mavparam below /home/pc-casy/catkin_ws/src/mavros/mavros
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /home/pc-casy/catkin_ws/src/mavros/mavros/scripts/mavparam` 
so to do that i have to reach the `/home/pc-casy/catkin_ws/src/mavros/mavros/scripts/mavparam ` directory and type  `python mavparam set SYSID_MYGCS 1`
Now the override works, but you have some suggestions about the fail when i try to launch the mavparam node??
Thank you

vooon, 2015-01-30:
Maybe `rosrun` missed?
In my current install of Indigo on 14.10 it don't exist, but i blamed my setup (because i must wait Jade for 14.10...).

Also mavparam has shebang, so explicit calling python not required.

vooon, 2015-01-30:
Oh, or check file permissions all scripts should have exec perm (x).

dottantgal, 2015-01-30:
Yes, thanks, it's all about the permissions!

vooon, 2015-01-30:
But **git** know what permissions should be...

TODO: need add note about `SYSID_MYGCS`.

closed 2015-01-30 by vooon

erikasnow, 2019-11-01:
I'm running into this same problem with Ardupilot. I'm sending rc override messages and I can see the change in rqt. But I'm not getting anything from the rc in or rc out topics, even when I have a transmitter plugged in and I can see values in mavlink inspector. 

I tried running the mavparam command, but I get an IOError from the /mavros/param/set service. Any ideas?

idychenxi, 2020-03-09:
have you solved this issue? I met the same issue about setting SYSID_MYGCS