mavlink/mavros #209: Plugins not publishing when using TELEM2

pmukherj, 2015-02-06:
Hey guys, 

I just recently switched to using TELEM2 over an FTDI and mavros. Everything connects up fine, I get the heartbeat and the /mavlink/from topic clearly publishes topics. However, none of the plugins seem to be publishing data. Its odd, because using the USB port and the same settings/launch files, all the data seems to be publishing fine (albeit slowly, which is why I'm switching to TELEM2). This is using the px4 stack. 

I have links to a bag and the diagnostic output. Any thoughts?
https://www.dropbox.com/s/f5mu0n9lz8vwmhf/plugins_not_functioning.bag?dl=0
https://www.dropbox.com/s/rpm3oibddr89mkh/ros_debug_dump.csv?dl=0

-PJ

vooon, 2015-02-06:
Check your PX4 configuration. I see that status publushed at very high rate (~30 Hz) also heartbeat at 50 Hz. Usually it published at 1 Hz.
Almost all publish rates defined by FCU, not mavros.

I don't see most of topics in bag file. Try use something like this launch:
```xml
<!-- bixler2_apm_radio.launch -->
<launch>
	<!-- vim: ft=xml -->
	<include file="$(find mavros)/launch/apm2_radio.launch">
		<arg name="gcs_url" value="tcp-l://" />
	</include>

	<!-- log all data for now -->
	<node pkg="rosbag" type="record" name="rosbag" args="-a" />
</launch>
```

But change it for px4.

Also i don't see mavros startup messages, where it reports what plugins it load.

mhkabir, 2015-02-06:
@vooon As issue stands, QGC is not connecting either.

vooon, 2015-02-06:
And direct connection QGC-TELEM2 too?

mhkabir, 2015-02-06:
Yes, direct connection.

The issue isn't mine, its @pmukherj, we tried debugging it this
morning, to no avail.

pmukherj, 2015-02-06:
Yes. @mhkabir is correct. QGC is still not connecting. This is on a Dev build of px4. The connections are fine over USB and the FTDI cable is fine as I'm able to communicate using it in APM. However APM only goes upto baud 115 over TELEM. 

I am using the default px4.launch but with the port settings changed to be correct (USB0/921600)

Under 'rostopic list I see all the too

pmukherj, 2015-02-06:
Sorry, unfinished comment. 
Under 'rostopic list' I see all the topics but when I bag '-a', its just the ones you see that pop in. Must be a rosbag oddity. 

It loaded all the plugins reported a heartbeat but did not report RC signals or High Res IMU. 

This is the full rostopic list
diagnostics
/mavlink/from
/mavlink/to
/mavros/battery
/mavros/camera_image
/mavros/camera_image/compressed
/mavros/camera_image/compressed/parameter_descriptions
/mavros/camera_image/compressed/parameter_updates
/mavros/camera_image/compressedDepth
/mavros/camera_image/compressedDepth/parameter_descriptions
/mavros/camera_image/compressedDepth/parameter_updates
/mavros/camera_image/theora
/mavros/camera_image/theora/parameter_descriptions
/mavros/camera_image/theora/parameter_updates
/mavros/fix
/mavros/global_position/compass_hdg
/mavros/global_position/global
/mavros/global_position/gps_vel
/mavros/global_position/local
/mavros/global_position/rel_alt
/mavros/gps_vel
/mavros/imu/atm_pressure
/mavros/imu/data
/mavros/imu/data_raw
/mavros/imu/mag
/mavros/imu/temperature
/mavros/mission/waypoints
/mavros/mocap/pose
/mavros/position/local
/mavros/position/vision
/mavros/px4flow/ground_distance
/mavros/px4flow/raw/optical_flow_rad
/mavros/px4flow/temperature
/mavros/rc/in
/mavros/rc/out
/mavros/rc/override
/mavros/safety_area/set
/mavros/setpoint/accel
/mavros/setpoint/att_throttle
/mavros/setpoint/cmd_vel
/mavros/setpoint/local_position
/mavros/state
/mavros/time_reference
/mavros/vfr_hud
/mavros/vision_speed/speed_vector
/mavros/visualization/track_markers
/mavros/visualization/vehicle_marker
/mavros/wind_estimation
/rosout
/rosout_agg
/tf

jgoppert, 2015-02-06:
I had this setup working last night. A couple of times I ran into a similar issue. What worked for me:

 1. Login via NuttX shell. Make sure position_estimator_inav and attitude_estimator_ekf are running.
 2. Make sure SYSTEM_COMPANION is set to 921600

pmukherj, 2015-02-06:
ooo. Hmm. 
1. I'll try that. Is there anything specific you do to test that these are working?
2. I did verify this. Just to be a pendant it is SYS_COMPANION (not SYSTEM_COMPANION) correct?

mhkabir, 2015-02-07:
Hi Guys,

The reason that you don't get HIGHRES_IMU is that rcS on PX4 side is
scripted to start-up mavlink of TELEM2 in onboard mode.

I never probably never added HIGHRES_IMU to the stream list when I updated
that interface. I'll be doing that soon.

PJ, As James says, you estimation apps might not be running for some
reason. Did you try starting them manually?

pmukherj, 2015-02-07:
Yup. Just checked it, @mhkabir and @jgoppert are right. My apps weren't running, I'm guessing it will be added to a startup script at some point?

pmukherj, 2015-02-07:
Aaaand that was because my airframe was not set. My bad. @jgoppert Maybe you had the same issue. Either way. Closing.

closed 2015-02-07 by pmukherj