mavlink/mavros #215: mavros/state frequency

dottantgal, 2015-02-10:
Hi all,
I'm using a pixhawk with the ardurover firmware and i connect the autopilot with ROS using the wifi antennas.
is there any way to pump up the frequency of the mavros/state topic??
i see that the mavros/from come at 100Hz but the mavros/state at 1Hz.
Thank you

mhkabir, 2015-02-10:
State data is pushed at 1hz. You probably can use the mavsys call to
increase the stream rate. It's not very important though.

What'd you need state for?

dottantgal, 2015-02-10:
i need it in my project to change the operation mode from auto to manual and viceversa
I've tried to use the mavsys but nothing changes

vooon, 2015-02-10:
Why clog the channel with unnecessary information?
You want change modes several times per second?

dottantgal, 2015-02-11:
I have wrote a node as bridge between simulink and mavros so i can override the RC channels using a control algorithm in matlab. I use a channel of the radio controller to switch the operation mode from manual to auto and viceversa so with my node i check continuously the state. Matlab send to my node the pwm frequency for the channels and it's connected via UDP.
With my bridge node I subscribe to the mavros/state and based on the state the control goes to the radio controller or to the simulink algorithm, so all the things depend on the mavros/state publishing frequency.
Do you have any suggestions??

dottantgal, 2015-02-11:
I've fixed the problem about the publishing of the RC_CHANNELS so now i don't have delays to control the rover from matlab, but if i switch from auto to manual mode i have to wait the read of the state and in the meanwhile i could crash with my rover if something goes wrong with the matlab algorithm because the 1HZ frequency is too much low :)
Could you tell me how to pump up the rate frequency of the state?
thank you

vooon, 2015-02-11:
You can subscribe to `~rc/in` and check mode channel state. Rate on that topic can be modified by mavsys.
But actual mode can be differ, so check `~state` too.

dottantgal, 2015-02-11:
Thank you.
could you write me the command line to use mavsys on RC frequency? Actually i don't remember about it
thank u so much!

dottantgal, 2015-02-11:
I ve found how to do it 
` rosrun mavros mavsys rate --rc-channels 50` 
but nothing changes, rc/in is published at 3HZ and it's too low

dottantgal, 2015-02-11:
Ok, i think that the problem is because i'm using the telemetry antennas to comunicate with the pixhawk.
If i use the mavsys command when the pixhawk is wired with a usb cable i can pump up the frequency to 25HZ and it's a really good value.
I can't do anything if i use the antennas...probably it is a problem on the pixhawk side or not?
thank you

mhkabir, 2015-02-11:
The radios are rate-limited. You cannot do anything about that.

dottantgal, 2015-02-11:
MMMM do you know why the mavros/from is published at 100HZ using the antennas and the other stuff from  5 to 1 HZ????

mhkabir, 2015-02-11:
mavros/from is the raw mavlink datastream. Not a particular message.

vooon, 2015-02-11:
`/mavlink/from` consists all messages come from FCU, so it's rate is a sum.

Note that telemetry bandwidth is limited, so probably if you can setup faster datalink you could set higher rates.
But i don't know does APM have similar bandwidth manager as PX4 have.

closed 2015-03-09 by vooon

dottantgal, 2015-03-10:
Sorry about the delay.
Now I'm using the channel frequency that i use to change the operation mode, so i can reach the 25Hz and it's enough.
About the mavros/state: I'haven't tried to push up the frequency, so I think it is unsolved.