mavlink/mavros #217: log debugger

rakeshkaupu, 2015-02-13:
Hello,
I'm trying to "OFFBOARD CONTROL the drone using my linux machine using optitrack"
Every time I try to arm the drone (px4) I get this error message "Request Failed. Check the logs.".
I'm not sure if I'm looking at the right log, but the one I'm looking at says "/use_sim_time" is not set.

When I try and set the "use_sim_time", I'm not able to launch itself. I'm guessing the message setting to true/false isn't the problem.

So, my question is how best can I debug the issue/logs. Why do I see so many files in the <ROS_folder>/.ros/log/
Is it a separate folder for separate session/command ?

vooon, 2015-02-13:
Sim time is a simulated ros wall time, don't use.

Logs you interested is mavros log (on console that runs mavros launch, or in `~/.ros` folder for `mavros_node`).
Usually what you want, is FCU log messages ('FCU: xxxx'). And because i still didn't run my UAV with pixhawk i don't know does it write those messages to SD log.

mhkabir, 2015-02-13:
> many files in the /.ros/log/

You're probably saftied.

Press the safety switch and then send the command.

closed 2015-03-02 by mhkabir