mavlink/mavros #219: Edit rc.usb with mavftp

dani-carbonell, 2015-02-18:
Hi, 
I want to publish trough "/dev/ttyACM0" HIGHRES_IMU messages at 100 Hz since default config is 1Hz. I am able to set it with "mavlink stream" command but I want to make it permanent. 

I would like to edit the rc.usb file with the mavftp but I am getting errors, I am afraid is still not fully implemented.

Is there a way to modify/append/upload files without flashing the Pixhawk?

vooon, 2015-02-18:
FTP fully implemented, but you tried modify ROMFS entry (read only).
You can edit `etc/extras.txt` on SD, and disable px4 startup quirk.

vooon, 2015-02-18:
https://pixhawk.org/dev/system_startup

vooon, 2015-02-18:
Note: be careful with mode options, it may disable FTP :)

dani-carbonell, 2015-02-18:
Thanks vooon :+1: 
I will try once I solve rencent problems with the USB.

mstuettgen, 2015-03-01:
@dani-carbonell, i suggest using TELEM2 for communication, as USB is quite unrelieable und unstable. 
 here is my extras.txt:

usleep 100000
mavlink start -m custom -d /dev/ttyS2 -b 921600 -x
usleep 100000
mavlink stream -d /dev/ttyS2 -s HEARTBEAT -r 5
usleep 100000
mavlink stream -d /dev/ttyS2 -s STATUSTEXT -r 1
usleep 100000
mavlink stream -d /dev/ttyS2 -s SYS_STATUS -r 1
usleep 100000
mavlink stream -d /dev/ttyS2 -s PARAM_VALUE -r 200
usleep 100000
mavlink stream -d /dev/ttyS2 -s HIGHRES_IMU -r 50
usleep 100000
mavlink stream -d /dev/ttyS2 -s ATTITUDE -r 50
usleep 100000
mavlink stream -d /dev/ttyS2 -s ATTITUDE_QUATERNION -r 50
usleep 100000
mavlink stream -d /dev/ttyS2 -s VFR_HUD -r 10
usleep 100000
mavlink stream -d /dev/ttyS2 -s GPS_RAW_INT -r 5
usleep 100000
mavlink stream -d /dev/ttyS2 -s GLOBAL_POSITION_INT -r 5
usleep 100000
mavlink stream -d /dev/ttyS2 -s LOCAL_POSITION_NED -r 5

note that /dev/ttyS2 is the internal port for TELEM2

mstuettgen, 2015-03-01:
-x will activate FTP

mhkabir, 2015-03-02:
Just use this : https://pixhawk.org/dev/companion_link

vooon, 2015-03-09:
I think we may close.

closed 2015-03-09 by vooon