mavlink/mavros #220: Add BSD license option

vooon, 2015-02-25:
@LorenzMeier wants more permissive BSD license, so mavros can be deeply integrated with PX4 codebase.

In that case i prefer not to change license, but add a second license.

- [ ] libmavconn - triple licensed LGPLv3, BSD and GPLv3
- [ ] mavros - GPLv3
- [ ] mavros_extras - GPLv3

LorenzMeier, 2015-02-25:
Actually its not so much about PX4 compatibility (we don't link against mavros anyway) then about ROS compatibility. ROS is BSD licensed and a lot of ROS users depend on this licensing for their usage. For them having a GPL licensed communication layer as part of their setup is problematic and and has led to some of them using their own MAVLink to ROS bridges for licensing reasons only.

I therefore would also suggest to ask the other contributors to MAVROS and MAVROS_EXTRAS if they would be willing to re-license under the same model.

vooon, 2015-02-25:
List of contributors for `mavros`:

- [ ] @mhkabir Mohammed Kabir
- [ ] @TSC21 Nuno Marques
- [ ] @tonybaltovski Tony Baltovski
- [ ] @mthz 
- [ ] @ggregory8 Glenn Gregory

List of contributors for `mavros_extras`:

- [ ] @mhkabir 
- [ ] @tonybaltovski 
- [ ] @TSC21

vooon, 2015-02-25:
FYI: i just released mavros 0.10.2 for *Indigo* and *Jade* (new). Really it's small release for new distro.
In that version libmavconn becomes BSD.

tonybaltovski, 2015-02-25:
@vooon Sounds good to me.

TSC21, 2015-02-25:
Ok, I'm not really into to this licenses thing, so do I just say "OK"? :D

vooon, 2015-02-25:
Generally in the BSD I do not like that my code can be closed. But on the other hand mavros does not have any complex code...

mstuettgen, 2015-03-02:
I would be finde with BSD Licence, because I thinkt the advantage of every ROS user beeing able to user mavros outweighs the "limitations" to the code

ggregory8, 2015-03-04:
@vooon sorry about not giving a response earlier. Agree BSD for ROS compatibility is good.

vooon, 2015-03-04:
Well, I got the agreement of all the authors of the code.
@mthz i don't have any contact information, so thanks for wp bug fix, but I will not wait for you.

mthz, 2015-03-04:
sry for the late response.
BSD works for me

vooon, 2015-03-04:
Ok, thanks!

vooon, 2015-03-04:
All done, next release (0.11) will be dual licensed GPL/BSD.

closed 2015-03-04 by vooon

LorenzMeier, 2015-03-04:
@vooon This is really great news! I think it will greatly help the continuing adoption of mavros. The current situation with a top-level license file and LGPL in the headers is quite confusing. I would propose to remove the license text, and change it to a short paragraph along the lines below.
This simplifies the source code files and ensures the terms remain clear to everybody.

```
/**************
 * This file is part of the mavros package and subject to the license terms
 * in the top-level LICENSE file of the mavros repository.
 * https://github.com/mavlink/mavros/tree/master/LICENSE.md
 */
```
Then in the top level license file:
```
mavros is available triple-licensed as BSD, GPLv3 and LGPLv3.
Its use is compatible with any of these licenses. However, contributions to the upstream
repository must be made available under all three licenses and therefore have
to be compatible to BSD, GPLv3 and LGPLv3.
```

I would also suggest to add a CONTRIBUTIONS.md file similar to what other projects have, e.g. the PX4 one is here, which again restates the contributions requirements.
https://github.com/PX4/Firmware/blob/master/CONTRIBUTING.md

vooon, 2015-03-04:
Ok, i will do it later.