mavlink/mavros #229: Add support for direct actuator control

mhkabir, 2015-02-28:


mstuettgen, 2015-02-28:
Thanks for opening. 
where can i find the definitions of the actuator control messages? do you have some example code to start?

mstuettgen, 2015-02-28:
https://pixhawk.ethz.ch/mavlink/#SET_ACTUATOR_CONTROL_TARGET

thats our baby

vooon, 2015-02-28:
Description looks like [RCOut.msg](http://docs.ros.org/indigo/api/mavros/html/msg/RCOut.html).

mstuettgen, 2015-02-28:
Hmm RCout unfortunately uses uint16t... i need float
i might begin with std_msgs/float for testing

mstuettgen, 2015-02-28:
http://docs.ros.org/indigo/api/std_msgs/html/msg/Float32MultiArray.html should do for testing

mhkabir, 2015-03-02:
@mstuettgen Awesome job on the plugin :) I just woke and saw the semicolon thing CD
I'll be closing this once you confirm that you have it working in Firmware

mhkabir, 2015-03-02:
Also, please make a generic wiki page in the PX4 wiki regarding off board actuator control. I'll patch it up and integrate with the rest of the off board docs.
Also, regarding the rover page, I think you can remove it from ROS as of now. I might take the current controller up sometime later, add positiom controller and turn it into a new standalone platform ;)

Thanks!!

closed 2015-03-02 by mhkabir