mavlink/mavros #236: Possible command plugin ack bug

vooon, 2015-03-02:
Seems that waiting ACK is broken.

```
[ INFO] [1425259196.659292601]: CON: Got HEARTBEAT, connected.
[ WARN] [1425259197.653793168]: FCU: Beginning INS calibration; do not move plane
[ INFO] [1425259197.686509503]: FCU: command received: 
[ INFO] [1425259202.665872072]: VER: Capabilities 0x0000000000000000
[ INFO] [1425259202.666041584]: VER: Flight software:     00000000 (999710d0)
[ INFO] [1425259202.666132250]: VER: Middleware software: 00000000 (        )
[ INFO] [1425259202.666275503]: VER: OS software:         00000000 (        )
[ INFO] [1425259202.666399389]: VER: Board hardware:      00000000
[ INFO] [1425259202.666535555]: VER: VID/PID: 0000:0000
[ INFO] [1425259202.666683343]: VER: UID: 0000000000000000
[ WARN] [1425259202.667359257]: Unexpected command 520, result 0
[ INFO] [1425259202.668310696]: FCU: Initialising APM...
```

vooon, 2017-04-04:
Don't know. Anyway that warning is not something very important.

TSC21, 2017-04-07:
Closing this as outdated.

closed 2017-04-07 by TSC21