mavlink/mavros #237: Error compiling mavros for indigo (master)

vmayoral, 2015-03-02:
@vooon I'm having issues compiling mavros. Does this sound familiar to you?

```bash 
Linking CXX executable /home/victor/indigo/devel_isolated/mavros/lib/mavros/mavros_node
[ 87%] Built target mavros_node
[ 87%] Building CXX object CMakeFiles/mavros_plugins.dir/src/plugins/sys_time.cpp.o
[ 88%] Building CXX object CMakeFiles/mavros_plugins.dir/src/plugins/imu_pub.cpp.o
[ 89%] Building CXX object CMakeFiles/mavros_plugins.dir/src/plugins/gps.cpp.o
[ 89%] Building CXX object CMakeFiles/mavros_plugins.dir/src/plugins/param.cpp.o
[ 90%] Building CXX object CMakeFiles/mavros_plugins.dir/src/plugins/waypoint.cpp.o
/home/victor/indigo/src/mavros/src/plugins/param.cpp: In static member function 'static mavplugin::Parameter::param_t mavplugin::Parameter::from_param_value(mavlink_param_value_t&)':
/home/victor/indigo/src/mavros/src/plugins/param.cpp:63:14: error: cannot bind packed field 'uv.param_union::<anonymous>.param_union::<anonymous union>::param_uint8' to 'unsigned char&'
    return uv.param_uint8;
              ^
/home/victor/indigo/src/mavros/src/plugins/param.cpp:65:14: error: cannot bind packed field 'uv.param_union::<anonymous>.param_union::<anonymous union>::param_int8' to 'signed char&'
    return uv.param_int8;
              ^
/home/victor/indigo/src/mavros/src/plugins/param.cpp:67:14: error: cannot bind packed field 'uv.param_union::<anonymous>.param_union::<anonymous union>::param_uint16' to 'short unsigned int&'
    return uv.param_uint16;
...
CMakeFiles/mavros_plugins.dir/build.make:169: recipe for target 'CMakeFiles/mavros_plugins.dir/src/plugins/param.cpp.o' failed
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/param.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:2120: recipe for target 'CMakeFiles/mavros_plugins.dir/all' failed
make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'mavros': 
  Command '/home/victor/indigo/install_isolated/env.sh make -j2 -l2' returned non-zero exit status 2

```

vooon, 2015-03-02:
What OS version? Looks like old error on trusty (but it more compiler specific).
And i fixed that. What version of mavlink used?

vmayoral, 2015-03-02:
My bad, just noticed it was `debian/hydro/mavlink`. Updated to `debian/indigo/mavlink` and recompiling. Seems to progress.

vmayoral, 2015-03-02:
Got a similar error again. This time using https://github.com/mavlink/mavlink-gbp-release/tree/debian/indigo/mavlink
```
[ 90%] Building CXX object CMakeFiles/mavros_plugins.dir/src/plugins/waypoint.cpp.o
/home/victor/indigo/src/mavros/src/plugins/param.cpp: In static member function 'static mavplugin::Parameter::param_t mavplugin::Parameter::from_param_value(mavlink_param_value_t&)':
/home/victor/indigo/src/mavros/src/plugins/param.cpp:63:14: error: cannot bind packed field 'uv.param_union::<anonymous>.param_union::<anonymous union>::param_uint8' to 'unsigned char&'
    return uv.param_uint8;
              ^
/home/victor/indigo/src/mavros/src/plugins/param.cpp:65:14: error: cannot bind packed field 'uv.param_union::<anonymous>.param_union::<anonymous union>::param_int8' to 'signed char&'
    return uv.param_int8;
              ^
/home/victor/indigo/src/mavros/src/plugins/param.cpp:67:14: error: cannot bind packed field 'uv.param_union::<anonymous>.param_union::<anonymous union>::param_uint16' to 'short unsigned int&'
    return uv.param_uint16;
              ^
/home/victor/indigo/src/mavros/src/plugins/param.cpp:69:14: error: cannot bind packed field 'uv.param_union::<anonymous>.param_union::<anonymous union>::param_int16' to 'short int&'
    return uv.param_int16;
              ^
/home/victor/indigo/src/mavros/src/plugins/param.cpp:71:14: error: cannot bind packed field 'uv.param_union::<anonymous>.param_union::<anonymous union>::param_uint32' to 'unsigned int&'
    return uv.param_uint32;
              ^
/home/victor/indigo/src/mavros/src/plugins/param.cpp:73:14: error: cannot bind packed field 'uv.param_union::<anonymous>.param_union::<anonymous union>::param_int32' to 'int&'
    return uv.param_int32;
              ^
/home/victor/indigo/src/mavros/src/plugins/param.cpp:75:14: error: cannot bind packed field 'uv.param_union::<anonymous>.param_union::<anonymous union>::param_float' to 'float&'
    return uv.param_float;
              ^
CMakeFiles/mavros_plugins.dir/build.make:169: recipe for target 'CMakeFiles/mavros_plugins.dir/src/plugins/param.cpp.o' failed
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/param.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:2120: recipe for target 'CMakeFiles/mavros_plugins.dir/all' failed
make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'mavros': 
  Command '/home/victor/indigo/install_isolated/env.sh make -j2 -l2' returned non-zero exit status 2
```

vooon, 2015-03-02:
Hmm, try add type casting: `(type)field` on error lines.

vooon, 2015-03-03:
Also try to update mavlink package, look #240.

vmayoral, 2015-03-04:
Fixed after adding:

>Hmm, try add type casting: (type)field on error lines.

Thanks!

closed 2015-03-04 by vmayoral

vooon, 2015-03-04:
Please send me a patch.

vmayoral, 2015-03-04:
@vooon tried to make a pull request but then noticed that i was using the hydro branch of the code while trying to compile indigo. 

Here's what i did in the `mavros/mavros/src/plugins/param.cpp` file of the hydro version to compile in indigo:
```cpp
 61                 switch (pmsg.param_type) {
 62                 case MAV_PARAM_TYPE_UINT8:
 63                         return (uint8_t) uv.param_uint8;
 64                 case MAV_PARAM_TYPE_INT8:
 65                         return (uint8_t) uv.param_int8;
 66                 case MAV_PARAM_TYPE_UINT16:
 67                         return (uint16_t) uv.param_uint16;
 68                 case MAV_PARAM_TYPE_INT16:
 69                         return (uint16_t) uv.param_int16;
 70                 case MAV_PARAM_TYPE_UINT32:
 71                         return (uint32_t) uv.param_uint32;
 72                 case MAV_PARAM_TYPE_INT32:
 73                         return (uint32_t) uv.param_int32;
 74                 case MAV_PARAM_TYPE_REAL32:
 75                         return (float) uv.param_float;

```

just added the casting but for what i've seen you already took this into account in newer versions of the code. Do you still want a patch for the hydro branch?

vooon, 2015-03-04:
Hm, i think i release that bugfix. But yes, you should use master for Indigo and Jade.