mavlink/mavros #239: Send a Mission_Item message using mavros

clementbesson, 2015-03-03:
Hi,

I am using mavros to send commands to a Pixhawk from an Odroid XU3. I am currently trying to send a Mission_Item (#39) message with command id #113 to perform a change in altitude. My message has the following payload:

frame = 3
command = 113
current = 3 
autocontinue = 1
and then the parameters and x, y, z data.

This message has been successfully tested in SITL and real life while sending it from Mission Planner . I am now trying to send it from my Odroid XU3

To do so my plan was to use "rosrun mavros mavwp goto" since it is a Mission_Item and since the Waypoint.msg  has an "is_current" value that can be accessed and modified. My thought was to change it from 2 to 3. 

By modifying the mavwp script I am now able of setting the is_current value in the terminal by running the following command: rosrun mavros mavwp goto 3 113 is_3 p1 p2 p3 x y z

On the SITL side, I do receive the message BUT when I print the packet.current value of my message is always 2 no matter what I set it to form the Odroid.

My question is as follows: Where in mavros is the current value automatically set to 2 when I run "mavwp goto" and how can I modify that?

Many thanks in advance for your time!

Best,

Clement

vooon, 2015-03-03:
See the code! https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/waypoint.cpp#L815

For GOTO current has magic value 2.
Also i don't see that APM supports `COMMAND_LONG` with `MAV_CMD_CONDITION_CHANGE_ALT`.

vooon, 2015-03-03:
You may add change alt quirk to waypoint plugin (same as GOTO).

clementbesson, 2015-03-03:
Thank you for your quick reply. I remove the version of mavros I was using and I am trying to build mavros from source (to try and modify that magic value for the seek of testing) as explained here: 

https://github.com/mavlink/mavros/tree/master/mavros

I created a workspace in my home directory as explained in ROS tutorials and when comes the time to perform the catkin_make it crashes and it gives me the following message:

[ 84%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_position.cpp.o
/home/clementbesson/catkin_ws/src/mavros/mavros_extras/src/plugins/mocap_pose_estimate.cpp: In member function ‘void mavplugin::MocapPoseEstimatePlugin::mocap_pose_send(uint64_t, float*, float, float, float)’:
/home/clementbesson/catkin_ws/src/mavros/mavros_extras/src/plugins/mocap_pose_estimate.cpp:101:6: error: ‘mavlink_msg_att_pos_mocap_pack_chan’ was not declared in this scope
     z);
      ^
make[2]: *** [mavros/mavros_extras/CMakeFiles/mavros_extras.dir/src/plugins/mocap_pose_estimate.cpp.o] Error 1
make[1]: *** [mavros/mavros_extras/CMakeFiles/mavros_extras.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 84%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/safety_area.cpp.o
/home/clementbesson/catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp: In member function ‘void mavplugin::SystemStatusPlugin::autopilot_version_cb(const ros::TimerEvent&)’:
/home/clementbesson/catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp:619:26: error: ‘MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES’ was not declared in this scope
    cmd.request.command = MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES;
                          ^
make[2]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Any thoughts on what is happening?

Clement

clementbesson, 2015-03-03:
Thanks for your quick reply. I am going to make the modification to waypoint.cpp and build mavros from source. I will let you know if it worked

clementbesson, 2015-03-03:
It worked, thank you very much for your help

closed 2015-03-04 by clementbesson