mavlink/mavros #240: Fail to build mavros from source with ROS indigo and Ubuntu trusty

clementbesson, 2015-03-03:
Hi

I am trying to build mavros from source as explained here:

https://github.com/mavlink/mavros/tree/master/mavros

I created a workspace in my home directory as explained in ROS tutorials and when comes the time to perform the catkin_make it crashes and it gives me the following message:

[ 84%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_position.cpp.o
/home/clementbesson/catkin_ws/src/mavros/mavros_extras/src/plugins/mocap_pose_estimate.cpp: In member function ‘void mavplugin::MocapPoseEstimatePlugin::mocap_pose_send(uint64_t, float, float, float, float)’:
/home/clementbesson/catkin_ws/src/mavros/mavros_extras/src/plugins/mocap_pose_estimate.cpp:101:6: error: ‘mavlink_msg_att_pos_mocap_pack_chan’ was not declared in this scope
z);
^
make[2]: ** [mavros/mavros_extras/CMakeFiles/mavros_extras.dir/src/plugins/mocap_pose_estimate.cpp.o] Error 1
make[1]: *** [mavros/mavros_extras/CMakeFiles/mavros_extras.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 84%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/safety_area.cpp.o
/home/clementbesson/catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp: In member function ‘void mavplugin::SystemStatusPlugin::autopilot_version_cb(const ros::TimerEvent&)’:
/home/clementbesson/catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp:619:26: error: ‘MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES’ was not declared in this scope
cmd.request.command = MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES;
^
make[2]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Any thoughts on what is happening?

Thanks in advance for your time

Clement

vooon, 2015-03-03:
Please install fresh mavlink package from shadow repo: http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-indigo-mavlink

For Ubuntu 14.04 amd64:
```
cd /tmp
wget http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-indigo-mavlink/ros-indigo-mavlink_2015.3.3-0trusty-20150303-1144-+0000_amd64.deb
sudo dpkg -i *mavlink*deb
```

clementbesson, 2015-03-03:
Great thank you

clementbesson, 2015-03-03:
Also, once I catkin_make in my workspace, what should I do to have ROS knowing that its there? Because right now I get error saying "ResourceNotFound: mavros" when I trying to rosrun when of my script. Thanks again for your help

vooon, 2015-03-03:
`source <myws>/devel/setup.bash`

vooon, 2015-03-03:
For example there part of my `~/.zshenv` file (for bash change file extension):
```sh
# ROS
if [[ -e "$HOME/ros/setup.zsh" ]]; then
	source $HOME/ros/setup.zsh
elif [[ -e "$HOME/ros/devel/setup.zsh" ]]; then
	source $HOME/ros/devel/setup.zsh
elif [[ -e "/opt/ros/jade/setup.zsh" ]]; then
	source /opt/ros/jade/setup.zsh
elif [[ -e "/opt/ros/indigo/setup.zsh" ]]; then
	source /opt/ros/indigo/setup.zsh
elif [[ -e "/opt/ros/hydro/setup.zsh" ]]; then
	source /opt/ros/hydro/setup.zsh
fi
```

clementbesson, 2015-03-03:
Ok so I added: source <myws>/devel/setup.bash at the end of my .bashrc file (in my home directory)
Then I open a new terminal, go to a different directory and run "roslaunch mylaunchfile.launch"
And I still get the same error...

Here is my launch file:

<launch>
 	
	<arg name="fcu_url" default="udp://192.168.0.107:14552@192.168.0.107:14554" />
	<arg name="gcs_url" default="" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="50" />

	<node name="mavros" pkg="mavros" type="mavros_node" output="screen">
		<param name="fcu_url" value="$(arg fcu_url)" />
		<param name="gcs_url" value="$(arg gcs_url)" />
		<param name="target_system_id" value="$(arg tgt_system)" />
		<param name="target_component_id" value="$(arg tgt_component)" />
 
		<rosparam command="load" file="$(find mavros)/launch/px4_blacklist.yaml" />
 
		
		<param name="conn_heartbeat" value="5.0" />
		<param name="conn_timeout" value="5.0" />
                
		<param name="startup_px4_usb_quirk" value="true" />
 
	</node>
 
 
</launch>

vooon, 2015-03-03:
Check that you sourced right setup.bash (and dot overwrite it with /opt/ros variant).

vooon, 2015-03-03:
So your `env` should be similar to my:
```
ROS_DISTRO=jade
ROS_PACKAGE_PATH=/home/vovan/ros/src:/opt/ros/jade/share:/opt/ros/jade/stacks
ROS_ETC_DIR=/opt/ros/jade/etc/ros
ROS_ROOT=/opt/ros/jade/share/ros
ROS_MASTER_URI=http://localhost:11311
```

clementbesson, 2015-03-03:
Ok so its works when I source to the directory where I built mavros. If I go there and run the my launch file, its all good. If I open a new terminal, enter "rosrun mavros ......." it works. Therefore I think my question is as follows: how do I do to be able to use mavros wherever I am? 

For instance I have a other directory: /home/clementbesson/dir and in dir, I need to source devel/setup.bash in order to be able to use the command "rosrun my_package my_script". When I source in dir, I am not capable of using mavros anymore.

clementbesson, 2015-03-03:
I ended up chaining the second directory:  /home/clementbesson/dir to the one that contains mavros and it works fine. Thanks again for your time

closed 2015-03-03 by clementbesson