mavlink/mavros #243: Building Mavros from source from Odroid XU-3 running Ubuntu 14.04.1 LTS

clementbesson, 2015-03-04:
Hi,

I am trying to build mavros from source as mentioned in issue #240 the difference being that I am doing it from an Odroid. I get an "internal compiler error" message mentioned below:

It looks like my system is running out of memory... Here is my memory specs:

                    total       used       free     shared    buffers     cached
Mem:          1995        843       1151         23         17        164
-/+ buffers/cache:        662       1333
Swap:            0          0          0

Thanks in advance for your time,

Clement

[ 86%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/dummy.cpp.o
g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [mavros/mavros/CMakeFiles/gcs_bridge.dir/src/gcs_bridge.cpp.o] Error 4
make[1]: *** [mavros/mavros/CMakeFiles/gcs_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [mavros/mavros_extras/CMakeFiles/mavros_extras.dir/src/plugins/vision_pose_estimate.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [mavros/mavros_extras/CMakeFiles/mavros_extras.dir/src/plugins/px4flow.cpp.o] Error 4
/home/odroid/mavros_source/src/mavros/mavros_extras/src/plugins/mocap_pose_estimate.cpp: In member function 'void mavplugin::MocapPoseEstimatePlugin::mocap_pose_send(uint64_t, float*, float, float, float)':
/home/odroid/mavros_source/src/mavros/mavros_extras/src/plugins/mocap_pose_estimate.cpp:101:6: error: 'mavlink_msg_att_pos_mocap_pack_chan' was not declared in this scope
     z);
      ^
g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [mavros/mavros_extras/CMakeFiles/mavros_extras.dir/src/plugins/image_pub.cpp.o] Error 4
make[2]: *** [mavros/mavros_extras/CMakeFiles/mavros_extras.dir/src/plugins/mocap_pose_estimate.cpp.o] Error 1
make[1]: *** [mavros/mavros_extras/CMakeFiles/mavros_extras.dir/all] Error 2
[ 87%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/param.cpp.o
Linking CXX executable /home/odroid/mavros_source/devel/lib/mavros/mavros_node
Linking CXX executable /home/odroid/mavros_source/devel/lib/mavros_extras/gcs_image_bridge
[ 87%] Built target mavros_node
[ 87%] Built target gcs_image_bridge
[ 88%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/command.cpp.o
[ 88%] [ 88%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/rc_io.cpp.o
Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/local_position.cpp.o
[ 89%] [ 89%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/waypoint.cpp.o
Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/gps.cpp.o
[ 90%] [ 91%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/imu_pub.cpp.o
Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/sys_time.cpp.o
[ 91%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o
[ 92%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/global_position.cpp.o
g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/local_position.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/rc_io.cpp.o] Error 4
[... 186 lines cut ...]
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro 'ROSCONSOLE_PRINT_AT_LOCATION'
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:516:35: note: in expansion of macro 'ROS_LOG_COND'
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:100:35: note: in expansion of macro 'ROS_LOG'
 #define ROS_INFO_NAMED(name, ...) ROS_LOG(::ros::console::levels::Info, std::string(ROSCONSOLE_NAME_PREFIX) + "." + name, __VA_ARGS__)
                                   ^
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp:603:3: note: in expansion of macro 'ROS_INFO_NAMED'
   ROS_INFO_NAMED("sys", "VER: UID: %016llx", (long long int)apv.uid);
   ^
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp: In member function 'void mavplugin::SystemStatusPlugin::autopilot_version_cb(const ros::TimerEvent&)':
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp:632:26: error: 'MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES' was not declared in this scope
    cmd.request.command = MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES;
                          ^
g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/imu_pub.cpp.o] Error 4
make[2]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o] Error 1
make[1]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
Invoking "make" failed

mhkabir, 2015-03-04:
Limit number of paralell tasks : catkin_make -j1

vooon, 2015-03-04:
1. please wrap dumps with ```
2. try `-j1`

clementbesson, 2015-03-04:
Thanks @mhkabir and @vooon for your replies. I tried catkin_make -j1 and now I get the following:
``` [ 79%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o
In file included from /opt/ros/indigo/include/ros/assert.h:35:0,
                 from /opt/ros/indigo/include/ros/common.h:36,
                 from /opt/ros/indigo/include/ros/publisher.h:32,
                 from /opt/ros/indigo/include/ros/node_handle.h:32,
                 from /opt/ros/indigo/include/diagnostic_updater/diagnostic_updater.h:42,
                 from /home/odroid/mavros_source/src/mavros/mavros/include/mavros/mavros_plugin.h:31,
                 from /home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp:27:
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp: In member function 'void mavplugin::SystemStatusPlugin::handle_autopilot_version(const mavlink_message_t*, uint8_t, uint8_t)':
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp:593:9: error: 'mavlink_autopilot_version_t' has no member named 'flight_sw_version'
     apv.flight_sw_version,
         ^
/opt/ros/indigo/include/ros/console.h:342:165: note: in definition of macro 'ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER'
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro 'ROSCONSOLE_PRINT_AT_LOCATION'
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:516:35: note: in expansion of macro 'ROS_LOG_COND'
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:100:35: note: in expansion of macro 'ROS_LOG'
 #define ROS_INFO_NAMED(name, ...) ROS_LOG(::ros::console::levels::Info, std::string(ROSCONSOLE_NAME_PREFIX) + "." + name, __VA_ARGS__)
                                   ^
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp:592:3: note: in expansion of macro 'ROS_INFO_NAMED'
   ROS_INFO_NAMED("sys", "VER: Flight software:     %08x (%*s)",
   ^
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp:594:12: error: 'mavlink_autopilot_version_t' has no member named 'flight_custom_version'
     8, apv.flight_custom_version);
            ^
/opt/ros/indigo/include/ros/console.h:342:165: note: in definition of macro 'ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER'
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro 'ROSCONSOLE_PRINT_AT_LOCATION'
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:516:35: note: in expansion of macro 'ROS_LOG_COND'
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:100:35: note: in expansion of macro 'ROS_LOG'
 #define ROS_INFO_NAMED(name, ...) ROS_LOG(::ros::console::levels::Info, std::string(ROSCONSOLE_NAME_PREFIX) + "." + name, __VA_ARGS__)
                                   ^
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp:592:3: note: in expansion of macro 'ROS_INFO_NAMED'
   ROS_INFO_NAMED("sys", "VER: Flight software:     %08x (%*s)",
   ^
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp:596:9: error: 'mavlink_autopilot_version_t' has no member named 'middleware_sw_version'
     apv.middleware_sw_version,
         ^
/opt/ros/indigo/include/ros/console.h:342:165: note: in definition of macro 'ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER'
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro 'ROSCONSOLE_PRINT_AT_LOCATION'
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:516:35: note: in expansion of macro 'ROS_LOG_COND'
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:100:35: note: in expansion of macro 'ROS_LOG'
 #define ROS_INFO_NAMED(name, ...) ROS_LOG(::ros::console::levels::Info, std::string(ROSCONSOLE_NAME_PREFIX) + "." + name, __VA_ARGS__)
                                   ^
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp:595:3: note: in expansion of macro 'ROS_INFO_NAMED'
   ROS_INFO_NAMED("sys", "VER: Middleware software: %08x (%*s)",
   ^
[... cut ...]
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp: In member function 'void mavplugin::SystemStatusPlugin::autopilot_version_cb(const ros::TimerEvent&)':
/home/odroid/mavros_source/src/mavros/mavros/src/plugins/sys_status.cpp:632:26: error: 'MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES' was not declared in this scope
    cmd.request.command = MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES;
                          ^
make[2]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o] Error 1
make[1]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
Invoking "make" failed ```

vooon, 2015-03-04:
Build mavlink deb from sources (look readme).

clementbesson, 2015-03-04:
Is this what I'm suppose to do? Not sure I'm looking at the correct readme
- Clone into a user-writable directory.
- Add the repository directory to your PYTHONPATH

clementbesson, 2015-03-04:
Ok so I followed the instructions in the readme file and I get stuck when I have to run 

```catkin_make_isolated --install-space $ROSINSTALL --install -DCMAKE_BUILD_TYPE=Release```

I get this error:

```catkin_make_isolated: error: argument --install-space: expected one argument```

jgoppert, 2015-03-04:
You need to define ROSINSTALL. Usually it is defined by sourcing setup.bash in devel or /opt/ros/indigo

clementbesson, 2015-03-04:
I tried both and when I echo $ROSINSTALL its still empty...

jgoppert, 2015-03-04:
Just give it a directory  like install, it is just where the installed
files are put.
catkin_make_isolated --install-space install

clementbesson, 2015-03-04:
Ok so I successfully ran 

```catkin_make_isolated --install-space /opt/ros/indigo --install -DCMAKE_BUILD_TYPE=Release```

when I go back to my mavros source code directory and run ```catkin_make -j1``` I have the same error message as previously

clementbesson, 2015-03-04:
Ok so I ran:

```catkin_make_isolated --install-space /opt/ros/indigo --install -DCMAKE_BUILD_TYPE=Release```

and then I get:

```[100%] Built target ualberta.xml-v1.0
Install the project...
-- Install configuration: "Release"
-- Installing: /opt/ros/indigo/include/mavlink/config.h
CMake Error at cmake_install.cmake:36 (FILE):
  file INSTALL cannot copy file
  "/home/odroid/ros_deps/build_isolated/mavlink/install/config.h" to
  "/opt/ros/indigo/include/mavlink/config.h".

make: *** [install] Error 1
<== Failed to process package 'mavlink': 
  Command 'make install' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/odroid/ros_deps/build_isolated/mavlink && make install

Command failed, exiting.```

vooon, 2015-03-04:
Better do deb build instruction (dh binary). You should have permissions to write to /opt/ros, so you must be root.

clementbesson, 2015-03-04:
Yes I'm root. Ok so I ran:
```
cd /tmp
git clone https://github.com/mavlink/mavlink-gbp-release.git -b debian/indigo/trusty/mavlink
cd mavlink-gbp-release
fakeroot dh binary
```

Then I go back to ``` /home/odroid/mavros_source ``` which is the directory where I put mavros' source code. When I run ``` catkin_make -j1 ``` I get the same error as before... I'm sorry its taking so much time, I really appreciate your help though

vooon, 2015-03-04:
Don't forget install deb with `sudo dpkg -i *mavlink*.deb`

clementbesson, 2015-03-04:
Awesome thanks! I really appreciate that you took the time

vooon, 2015-03-05:
I think that i may close.

closed 2015-03-05 by vooon