mavlink/mavros #246: FCU: command unsupported: 520

blackcoder, 2015-03-09:
New version of mavros is causing an error while trying to execute unsupported command 520 !

```
[ INFO] [1425901259.386226474]: FCU URL: /dev/ttyACM0:57600
[ INFO] [1425901259.386682342]: serial0: device: /dev/ttyACM0 @ 57600 bps
[ INFO] [1425901259.387312990]: GCS URL: udp://10.0.0.207:14555@10.0.0.207:14550
[ INFO] [1425901259.387727830]: udp1: Bind address: 10.0.0.207:14555
[ INFO] [1425901259.387805730]: udp1: Remote address: 10.0.0.207:14550
[ INFO] [1425901259.433151337]: Plugin 3dr_radio blacklisted
[ INFO] [1425901259.523706310]: Plugin actuator_control loaded and initialized
[ INFO] [1425901259.575039567]: Plugin command loaded and initialized
[ INFO] [1425901259.641702359]: Plugin ftp loaded and initialized
[ INFO] [1425901259.693883662]: Plugin global_position loaded and initialized
[ INFO] [1425901259.726677464]: Plugin gps loaded and initialized
[ INFO] [1425901259.726858108]: Plugin image_pub blacklisted
[ INFO] [1425901259.809435051]: Plugin imu_pub loaded and initialized
[ INFO] [1425901259.835208040]: Plugin local_position loaded and initialized
[ INFO] [1425901259.871940866]: udp1: Remote address: 10.0.0.207:14550
[ INFO] [1425901259.962554377]: Plugin mocap_pose_estimate loaded and initialized
[ INFO] [1425901259.989023867]: Plugin param loaded and initialized
[ INFO] [1425901260.023602318]: Plugin px4flow loaded and initialized
[ INFO] [1425901260.058890499]: Plugin rc_io loaded and initialized
[ INFO] [1425901260.090650122]: Plugin safety_area loaded and initialized
[ INFO] [1425901260.118216852]: Plugin setpoint_accel loaded and initialized
[ INFO] [1425901260.200700197]: Plugin setpoint_attitude loaded and initialized
[ INFO] [1425901260.248916057]: Plugin setpoint_position loaded and initialized
[ INFO] [1425901260.273909516]: Plugin setpoint_velocity loaded and initialized
[ INFO] [1425901260.320949272]: Plugin sys_status loaded and initialized
[ INFO] [1425901260.353703465]: Plugin sys_time loaded and initialized
[ INFO] [1425901260.362294760]: Plugin vfr_hud loaded and initialized
[ INFO] [1425901260.413919444]: Plugin vision_pose_estimate loaded and initialized
[ INFO] [1425901260.444317206]: Plugin vision_speed_estimate loaded and initialized
[ INFO] [1425901260.466406354]: Plugin visualization loaded and initialized
[ INFO] [1425901260.497647832]: Plugin waypoint loaded and initialized
[ INFO] [1425901260.497816985]: Autostarting mavlink via USB on PX4
[ INFO] [1425901260.497883951]: MAVROS started. MY ID [1, 240], TARGET ID [1, 50]
[ WARN] [1425901260.504751956]: TM: Clock skew detected (-1425901255.151658058 s). Hard syncing clocks.
[ INFO] [1425901260.510590456]: CON: Got HEARTBEAT, connected.
[ INFO] [1425901260.510951488]: High resolution IMU detected!
[ INFO] [1425901260.942613921]: FCU: data link 1 regained
[ERROR] [1425901261.610020429]: FCU: command unsupported: 520
[ WARN] [1425901262.515657339]: VER: request timeout, retries left 1
[ERROR] [1425901262.575620273]: FCU: command unsupported: 520
[ WARN] [1425901263.516305625]: VER: request timeout, retries left 0
[ERROR] [1425901263.543427227]: FCU: command unsupported: 520
[ WARN] [1425901264.514879048]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[ERROR] [1425901264.569913281]: FCU: command unsupported: 520
```

vooon, 2015-03-09:
Athat not error, just your FCU don't support version request. Duplicates #236.

closed 2015-03-09 by vooon

blackcoder, 2015-03-09:
I am using the latest px4 code, is it implemented yet ? 
Does this error cause any serious issues while using mavros ?

mhkabir, 2015-03-09:
Its not an error. Just a warning. PX4 doesn't support the request yet.

It doesn't affect runtime in any manner.

blackcoder, 2015-03-09:
Weird, because I just updated my mavros code, last commit I used was 16f03eaf22a6757406001fb1f94b904c9971c6b5  for some reason I can't seem to have the vision estimation working with the most recent commits (localX,Y,Z are not being updated) and he light doesn't blink green. /mavros/position/vision is getting the location as usual. Any idea why this might be happening ?

vooon, 2015-03-09:
Because #241 and #233 -> `/mavros/vision_pose/vision`.

mhkabir, 2015-03-09:
Actually, also PX4 has removed inav from being default estimator for
multicopter.

Now ekf_att_pos_estimator is default but it doesn't support vision
estimated.

Change the default estimator in the ROMFS script or use the NuttShell to
stop the EKF and start inav.

@vooon Perhaps add a note to wiki.

mhkabir, 2015-03-09:
Also perhaps we should change topic name to
``` /mavros/vision_pose/estimate``` so that it seems a but more clear?

blackcoder, 2015-03-09:
Thank you @vooon  and @mhkabir , now it makes sense and it works just like before. 
@mhkabir  you might have to update the wiki here https://pixhawk.org/dev/offboard_control/testing and change the topic names:

/mavros/position/local -> /mavros/local_position/local
/mavros/position/local -> /mavros/setpoint_position/local_position

mhkabir, 2015-03-09:
@blackcoder. Its a wiki. Feel free :) I'm all booked for this month.

blackcoder, 2015-03-09:
Done , @mhkabir I was more than happy to do it, just I wasn't sure that it's appropriate to do it at this stage since renaming/re-factoring is still in the process

mhkabir, 2015-03-09:
Now that you say it, I think we might have to discuss the naming once again.

Currently, @vooon , the names don't sound quite right. Perhaps discuss this
on Gitter?

vooon, 2015-03-09:
I think better in tracker (so we could link them as #49).

sahara-RAL, 2021-06-13:
I launched gazebo using make px4_sitl_default gazebo.
I launched mavros by this command roslaunch mavros px4.launch fcu_url:="udp://:14540@192.168.1.36:14557".

but when i am trying to rostopic echo /mavros/state that is not printing any output.

version: Kinetic 
ubuntu 16.04

codingCoffee, 2021-11-16:
@sahara-RAL were you able to resolve the issue?