mavlink/mavros #247: Better names for plugin topics

vooon, 2015-03-09:
We need some naming convention.
For now i only define that better use namespace with same name as plugin alias (where it is useful) (e.g. plugin `gps`: ns `~gps`).
That was common rule, but not strict. Also plugin with single topic (better with same name as plugin) may use `~`.

Current ns exceptions:

| plugin | namespace |
| -------- | --------------- |
| `imu_pub` | `~imu` |
| TBD | |

tonybaltovski, 2015-03-12:
Perhaps use velocity instead of speed? Everything else +1.

vooon, 2015-03-12:
My list of "bad" topic names and possible new names:

| plugin | topic | new name | comment |
| --- | --- | --- | --- |
| `actuator_control` | `~actuator_controls` | `~actuator_control` | plural |
| `local_position` | `~local_position/local` | ??? | |
| `px4flow` | `~px4flow/raw/optical_flow_rad` | ??? | raw? |
| `safety_area` | `~safety_area/set` | ??? | maybe place it to `~`? |
| `setpoint_position` | `~setpoint_position/local_position` | ??? | do we need `local_`? |
| `vision_pose_estimate` | `~vision_pose/vision` | ? | |
| `vision_speed_estimate` | `~vision_speed/speed_vector` | ? | |
| `global_position` | `~global_position/global` | `global_position/fix`? | |
| | `~global_position/local` | ? | |

TSC21, 2015-03-12:
@vooon in the case of `~setpoint_position/local_position`, `local_` is needed if we want to add a plugin to send `global_position` (`SET_POSITION_TARGET_GLOBAL_INT` message).

TSC21, 2015-03-12:
Agree with `~safety_area/set` be in `~`.

vooon, 2015-03-12:
Hmm, i think better keep `~safety_area` ns for params, so i keep `set` for now.

vooon, 2015-03-23:
Changed only `actuator_control`. Safety area are ok.

@mhkabir please look too.

tonybaltovski, 2015-03-23:
Just throwing out ideas:

- `~vision_pose/vision`  -> `~vision/pose`
- `~vision_speed/speed_vector` -> `~vision/velocity`
- `~setpoint_position/local_position` -> `~setpoint_position/local`
-  `~local_position/local` -> `~local_position/pose`

vooon, 2015-03-23:
I don't want merge namespaces between plugins anymore (`~setpoint/`).
But other looks ok. Table updated.

vooon, 2015-03-24:
Updated table. Decided keep current names in `local_position`, `global_position` and `px4flow`.

vooon, 2015-03-24:
Table updated.

Plugins updated too. And because no new changes proposed i close issue.

closed 2015-03-24 by vooon