mavlink/mavros #250: GPS OVERRIDE

dottantgal, 2015-03-10:
Hi all,
do you think it is possible to override the gps signal (getting datas from an optitrack system) and use mavros/pixhawk/ardupilot machine indoor to develop control loop algorithms ???
thank you

vooon, 2015-03-10:
It is possible, but anyway you must change firmware first. Also note that extras alredy has mocap plugin.

dottantgal, 2015-03-10:
Ok and how can i use them? Is it possible to substitute this data with gps then we can use the loiter mode (or to access it)?

vooon, 2015-03-10:
I mean that currently FCU don't handle any mocap messages, only PX4 can handle vision position estimates (with inav).
So you should implement that in firmware, and for that you may use message implemented in mocap plugin.

mhkabir, 2015-03-10:
Use the PX4 stack and not APM.

dottantgal, 2015-03-12:
I've spent a lot of time whitin the code...What do you think if I'm going to write an internal app that  publishes the PX4s uORB vehicle_gps_position fake topic for the AP_GPS_PX4 driver that will present the data received to the APM?? I don't think that there is some kind of check about the data (maybe something about the uart connection of the hardware), so it could be possible.
Let me know 
Thanks

vooon, 2015-03-12:
Better ask at [drone-discuss](https://groups.google.com/forum/#!forum/drones-discuss).

mhkabir, 2015-03-12:
Or px4users

Depending on the flight stack you end up using.

closed 2015-03-23 by vooon

dottantgal, 2015-03-23:
Yes, i'm creating a ROS node that is able to send GPS fake messages. At the moment i can get the green light using costants value, but i'm going to fix it using variable values!! It  seems to work because i can switch to Loiter mode without problems :dancer: