mavlink/mavros #251: google übersetzer'DeviceError:udp: resolve: Bind address resolve failed'

meuchel, 2015-03-15:
mavros/libmavconn/src/udp.cpp

vooon, 2015-03-16:
Ok, i will check. As i understood "bug" exists with `gcs_url`.

meuchel, 2015-03-16:
Yes, it occures with configured gcs_url using udp://...

vooon, 2015-03-16:
Bug should be there, but for what purpose you changed bind address? Try to use default `0.0.0.0` (empty host).

meuchel, 2015-03-16:
I didn't changed bind adress.
I start the node like this
<arg name="fcu_url" default="/dev/ttyACM0:921600" />
<arg name="gcs_url" default="udp://@gcshost:14550" />
where gcshost is another computer running apmplanner2 (gcshost is configured in /etc/hosts).
Both computer are connected via WLAN and when the connection gets lost the mavros node shuts down and will be respawned. During the respawn attemps when the networks gets back above error can happen and then it stucks.

vooon, 2015-03-18:
Hmm, try do set static dhcp address for your host. In that case router will add it to hosts, so dns will be successfull.

Also checked sources. Wrong issue title: mavros should say "Remote address resolve failed.", not bind.
And it is intentional that remote could not present at startup. Other side may send some data and then this address will be used. No need to restart mavros.
Just send some data from your GCS UDP port.

vooon, 2015-03-18:
If DeviceError occurs, mavros should stop, but it might interferre with #130.

meuchel, 2015-03-18:
Hi Voon,
not sure if I got you.
Ok, I will configure now static IP addresses on both sides:
- Host which is running mavros and roscore connected via USB to FCU. This host is connected to network via WLAN.
- and the other host (you name it remote?) where in my case apmplanner2 is running waiting for data from UDP. This host is connected to network with RJ45 cable.
Will let you know the results.

What means "Just send some data from your GCS UDP port."?
As far as I know GCS send nothing to mavros?

Thanks for you efforts!

vooon, 2015-03-18:
Unfortunately GCC does not send anything until it receives a heartbeat. So both ends (mavros and gcs) waiting some interaction from other side.
You may write simple program that sends one message or use `socat` with right parameters (please read man page).
Something like this `UDP:mavros-host:14555,sourceport=14550`.

I mean setup static DHCP IP's on router.

meuchel, 2015-03-18:
So, I've setup static IPs on the router for both hosts but this didn't help.
Sometimes respawning of mavros works but sometimes I get still the known error (see below).
In addition assign fix IP on mavros host makes of course no differnce too.

Sending something with socat didn't change anything. 
When mavros is running normaly I can see the traffic with socat.
But if below error occurs and I try to send something with socat I get of course Connection refused.

...
[mavros-1] restarting process
process[mavros-1]: started with pid [952]
[ INFO] [1426709298.142242343]: FCU URL: /dev/ttyACM0:921600
[ INFO] [1426709298.143429583]: device: /dev/ttyACM0 @ 921600 bps
[ INFO] [1426709298.145488764]: GCS URL: udp://@gcshost:14550
[ INFO] [1426709298.146118739]: Bind address: 0.0.0.0:1455 
[ INFO] [1426709298.146601460]: Remote address: 192.168.1.9:14550
[ INFO] [1426709298.162749895]: Plugin [alias 3dr_radio] blacklisted
[ERROR] [1426709298.165779435]: Network is unreachable
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
[mavros-1] process has died [pid 952, exit code -6, cmd /opt/ros/indigo/lib/mavros/mavros_node __name:=mavros __log:=/home/ros/.ros/log/17e19d04-cda2-11e4-bb2c-7cdd9079f6fe/mavros-1.log].
log file: /home/ros/.ros/log/17e19d04-cda2-11e4-bb2c-7cdd9079f6fe/mavros-1*.log
[mavros-1] restarting process
process[mavros-1]: started with pid [10043]
[ INFO] [1426707252.300334482]: FCU URL: /dev/ttyACM0:921600
[ INFO] [1426707252.301156414]: device: /dev/ttyACM0 @ 921600 bps
[ INFO] [1426707252.302946190]: GCS URL: udp://@gcshost:14550
[ WARN] [1426707252.303491756]: resolve error: Host not found (authoritative)
[FATAL] [1426707252.303916451]: GCS: DeviceError:udp: resolve: Bind address resolve failed
-> stuck :)

vooon, 2015-03-18:
Please try latest version build from sources.

meuchel, 2015-03-19:
I built from sources and the behaviour remains the same with a slightly different error message (...udp1: resolve error...).
...
[mavros-1] restarting process
process[mavros-1]: started with pid [28528]
[ INFO] [1426799268.745845168]: FCU URL: /dev/ttyACM0:921600
[ INFO] [1426799268.747444908]: serial0: device: /dev/ttyACM0 @ 921600 bps
[ INFO] [1426799268.749947642]: GCS URL: udp://@gcshost:14550
[ WARN] [1426799268.751049385]: udp1: resolve error: Host not found (authoritative)
[FATAL] [1426799268.751911921]: GCS: DeviceError:udp: resolve: Bind address resolve failed

vooon, 2015-03-19:
Try `udp://@` and then send something from gcshost to udp 14555.

vooon, 2015-03-19:
What you want is `Remote address: ` message with gcshost address.
Bind resolve failures looks weird (why it fail to resolve `0.0.0.0`?) and seems that related to your local DNS cache.
Present versions of Ubuntu uses `dnsmasq` for caching, sometimes it may break ddns (until TTL).

meuchel, 2015-03-20:
Unfortunatley I can try what you mention at the earliest tomorrow.

You mean: arg name="gcs_url" default="udp://@" ?
And this will do what?

Why do you think sending something from gcshost changes anything, especilly when binding failed before?

Regarding "Bind resolve failures looks weird (why it fail to resolve 0.0.0.0?) ":
For me this is strange too. 
As I worte before the issue doesn't happen always.
The issue comes up let say around every 10 times. Other restarting/respawn attemps usually works.
When ROS tries to respawn and the network is still not back restarting of node by ROS happens fairly quickly and repetitive.
So I assume when restarting of the node happens in the moment the network comes back but is still in an intermediate undefined state the  bind fails.

I want also mention that the first time I recorgnized this issue with bind failure was during a "normal" network problem. (don't know what happened, maybe bad WLAN connection or router hiccup or whatever)
To reproduce this network failure I simply unplug the WLAN stick and replug that network gets back automatically. It's not the same as the "normal" network problem but almost the same.
Nevertheless, the issue respectively stucking of the node should not occur.

BTW I use Ubuntu 14.04.1 LTS + Indigo on odroid xu3

vooon, 2015-03-20:
Any received message on udp connection changes remote address (gcs host), so after sending some data mavros gets remote and sends hb, gcs receives hb → connection established.

meuchel, 2015-03-20:
Ok, will try this soon.
Regarding "...after sending some data mavros gets remote and sends hb..."
This is a nice thing but makes no sense if GCS software itself sends nothing to mavros. (as you mentioned above it waits first for an heartbeat)
So always an additional tool would be neccessary wich sends sometimes something like a heartbeat to the mavros host?"
This is a nice thing but makes no sense if GCS software sends nothing to mavros. (as you mentioned above it waits first for an heartbeat
So an additional tool is neccessary wich sends sometimes something like a heartbeatto the mavros host?

meuchel, 2015-03-20:
With using udp://@ the bind issue seems to be gone (yet I dropped only several times the wlan connection) but this causes problems at GCS side now.
If the connecton to mavros host got lost first I need to recognize this.
Stop apmplanner2.
Needs to check if the connection to the mavros host is available again.
Send something with socat - UDP:192.168.1.32:14555,sourceport=14550
Then kill/end socat and then restart APM and hope that mavros is really up and running.
This workaround is too complicate to be usefull. :) 
More sense would make to let respawn on mavros side doing its job 
or GCS software would have to send heartbeats up to mavros. Not sure if one of them already provide this. 

I ried too 
socat - UDP:192.168.1.32:14555
and keep apmplanner2 running.
But then the sourceport is randomly and only socat receives something and apmplanner2 nothing.

meuchel, 2015-03-20:
want to add:
With using udp://@ and when wlan connection gets lost the mavros node still shuts down and will be respawned.

vooon, 2015-03-20:
That requires changes in GCS, connection dialog has some heartbeat options, but it seems that don't work.

closed 2015-04-09 by vooon