mavlink/mavros #258: Mavros not publishing in topics

dani-carbonell, 2015-03-26:
I run the `rosrun mavros px4.launch` on my laptop with Ubuntu 12.04 - ROS hydro and it works perfect and I can see the msgs with `rostopic echo /mavros/data/imu` but when I try to run it by ssh on a robot with the same Ubuntu - ROS - Kernel - Permissions on the USB  it just do not publish in any topic. I am able to list them but there is no way to see them.

```
ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[mavros-1]: started with pid [4010]
[ INFO] [1427394121.369656032]: FCU URL: /dev/ttyACM0:57600
[ INFO] [1427394121.371629414]: serial0: device: /dev/ttyACM0 @ 57600 bps
[ INFO] [1427394121.375517764]: GCS bridge disabled
[ INFO] [1427394121.390476378]: Plugin 3dr_radio blacklisted
[ INFO] [1427394121.738336512]: Plugin command loaded and initialized
[ INFO] [1427394121.821480747]: Plugin ftp loaded and initialized
[ INFO] [1427394121.873008581]: Plugin global_position loaded and initialized
[ INFO] [1427394121.899902572]: Plugin gps loaded and initialized
[ INFO] [1427394121.979681474]: Plugin imu_pub loaded and initialized
[ INFO] [1427394121.995987006]: Plugin local_position loaded and initialized
[ INFO] [1427394122.028336023]: Plugin param loaded and initialized
[ INFO] [1427394122.044832066]: Plugin rc_io loaded and initialized
[ INFO] [1427394122.060336470]: Plugin safety_area loaded and initialized
[ INFO] [1427394122.073096128]: Plugin setpoint_accel loaded and initialized
[ INFO] [1427394122.118930056]: Plugin setpoint_attitude loaded and initialized
[ INFO] [1427394122.144879531]: Plugin setpoint_position loaded and initialized
[ INFO] [1427394122.154468675]: Plugin setpoint_velocity loaded and initialized
[ INFO] [1427394122.208824310]: Plugin sys_status loaded and initialized
[ INFO] [1427394122.236453451]: Plugin sys_time loaded and initialized
[ INFO] [1427394122.251287137]: Plugin vfr_hud loaded and initialized
[ INFO] [1427394122.304000369]: Plugin waypoint loaded and initialized
[ INFO] [1427394122.305376417]: Autostarting mavlink via USB on PX4
[ INFO] [1427394122.306795673]: Built-in mavlink dialect: ardupilotmega
[ INFO] [1427394122.307112687]: MAVROS started. MY ID [1, 240], TARGET ID [1, 50]

```

blankered, 2015-03-26:
You might need to change your launch file to /dev/ttyUSB0:57600. Rostopic will list topics even if it is not connected to the pixhawk. You can verify that the pixhawk is connected with a heartbeat message which I do not see above.

vooon, 2015-03-27:
Check diagnostics ("FCU connection"). I don't see *CON* notification.

Use rqt plugin for that: http://wiki.ros.org/rqt_runtime_monitor
It is very useful (just set master uri on your PC)!

Check `dmesg` what usb-com device used for px, maybe wrong ACM?

vooon, 2015-03-27:
Also note, all topics are created before handling mavlink messages, so many of them may be silent.

dani-carbonell, 2015-03-27:
Thanks @vooon fr the recomendation of rqt tool. It was very useful to detect that a node of a laser scanner (not present) was blocking the port. I was really sure about the port since `/dev/ttyACM0` disappears when unplug the Pixhawk. As you say, I wasn't receiving the heartbeat but I didn't know why.

Thanks both of you :)

closed 2015-03-27 by dani-carbonell

voladoddi, 2015-04-24:
Hello guys, 

even I am stuck somewhere like this so I went over to this - http://answers.ros.org/question/193411/mavros-topics-not-publishing/?answer=207827#post-id-207827
and just posted my problem as an answer.

Then I was searching more and found this.
I have already changed to "ttyUSB0" after verifying that my radio is plugged in to that port.
However, 
when I do rosrun rqt_runtime_monitor [same], I see that 
[image]

Please advise what I could try to solve this issue.

vooon, 2015-04-24:
@voladoddi  No heartbeats => wrong port or packets is garbage (sad that because #269 we can't diagnose that).

sdipendra, 2015-05-08:
@voladoddi : I had the same problem, this might help.
$ cd ~/catkin_ws/src
Assuming catkin_ws is your catkin workspace.
$ git clone https://github.com/y22ma/mavlink_ros
$ cd ~/catkin_ws/
$ catkin_make
.....
.....
$ roscore
In a new terminal window.
$ rosrun mavlink_ros mavlink_ros_serial -p /dev/ttyACM0 -b 115200
Assuming your Device is at ttyACM0.
$ rostopic list -v

Published topics:
 * /mavlink/from [mavlink_ros/Mavlink] 1 publisher
 * /fcu/mag [sensor_msgs/MagneticField] 1 publisher
 * /rosout [rosgraph_msgs/Log] 1 publisher
 * /rosout_agg [rosgraph_msgs/Log] 1 publisher
 * /fcu/imu [sensor_msgs/Imu] 1 publisher
 * /fcu/raw/imu [sensor_msgs/Imu] 1 publisher

Subscribed topics:
 * /mavlink/to [mavlink_ros/Mavlink] 1 subscriber
 * /rosout [rosgraph_msgs/Log] 1 subscriber
$ rostopic echo /fcu/imu

This will give you angular_velocity.

For more information refer http://erlerobotics.gitbooks.io/erlerobot/content/en/mavlink/ros/mavros.html.

vooon, 2015-05-08:
@sdipendra `mavlink_ros` is a completely different package (and now deprecated by mavros).

@voladoddi In latest release (0.11.2) bug #269 fixed, please update and check FCU connection diag.

saikrn112, 2016-05-18:
Hello Guys, 
Even I am facing the same issue but I have connected to px4 via USB I have connected to correct ACM. I am using mavros (0.17.2) latest release. 
[image]
.

vooon, 2016-05-19:
@saikrn112 

Check ACM port, test by `minicom` (or `miniterm.py`) that it is nsh shell.
Then check that `sh /etc/init.d/rc.usb` in nsh starts mavlink streams (binary data, not printable).

anandbalakrishnan, 2016-07-12:
Hey. i'm having the same problem, but with a PXFmini on a RaspberryPi Zero. Wondering if you could help me out. Thanks.

HobbySingh, 2018-03-22:
I am having the exact same problem using  odroid and pixhawk connected through usb at ttyACM0
It does shows pixhawk at ACM0 in dmesg. Please help I have been stuck for 3 days.

There are no mavros topics being published, no heartbeat, vehicle shows disconnected.
Same setup works when I connect pixhawk via usb to my laptop having same mavros, ros and mavlink version but connecting it via odroid usb doesn't works.

HobbySingh, 2018-03-25:
Solved it by doing kernel update on odroid us3+.

sergioma295, 2019-06-07:
I am having the exact same problem. I have created a new issue. Can somebody help me please? https://github.com/mavlink/mavros/issues/1248