mavlink/mavros #259: source install

blankered, 2015-03-26:
I am trying to compile from source the new mavros 0.11 on indigio armhf. I get stuck at 54% when I run catkin_make. I understand that this is all still under development and hasn't been officially released yet through ros but I thought I would post the error I am having anyway so to help with the development.

[ 54%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o
In file included from /opt/ros/indigo/include/ros/assert.h:35:0,
                 from /opt/ros/indigo/include/ros/common.h:36,
                 from /opt/ros/indigo/include/ros/publisher.h:32,
                 from /opt/ros/indigo/include/ros/node_handle.h:32,
                 from /opt/ros/indigo/include/diagnostic_updater/diagnostic_updater.h:42,
                 from /opt/ros/indigo/share/mavros/workspace/src/mavros/mavros/include/mavros/mavros_plugin.h:21,
                 from /opt/ros/indigo/share/mavros/workspace/src/mavros/mavros/src/plugins/sys_status.cpp:17:
/opt/ros/indigo/share/mavros/workspace/src/mavros/mavros/src/plugins/sys_status.cpp: In member function 'void mavplugin::SystemStatusPlugin::handle_autopilot_version(const mavlink_message_t*, uint8_t, uint8_t)':
/opt/ros/indigo/share/mavros/workspace/src/mavros/mavros/src/plugins/sys_status.cpp:589:9: error: 'mavlink_autopilot_version_t' has no member named 'flight_sw_version'
     apv.flight_sw_version,
         ^
/opt/ros/indigo/include/ros/console.h:342:165: note: in definition of macro 'ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER'
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro 'ROSCONSOLE_PRINT_AT_LOCATION'
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:516:35: note: in expansion of macro 'ROS_LOG_COND'
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:100:35: note: in expansion of macro 'ROS_LOG'
 #define ROS_INFO_NAMED(name, ...) ROS_LOG(::ros::console::levels::Info, std::string(ROSCONSOLE_NAME_PREFIX) + "." + name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/share/mavros/workspace/src/mavros/mavros/src/plugins/sys_status.cpp:588:3: note: in expansion of macro 'ROS_INFO_NAMED'
   ROS_INFO_NAMED("sys", "VER: Flight software:     %08x (%*s)",
   ^
/opt/ros/indigo/share/mavros/workspace/src/mavros/mavros/src/plugins/sys_status.cpp:590:12: error: 'mavlink_autopilot_version_t' has no member named 'flight_custom_version'
     8, apv.flight_custom_version);
            ^
/opt/ros/indigo/include/ros/console.h:342:165: note: in definition of macro 'ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER'
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro 'ROSCONSOLE_PRINT_AT_LOCATION'
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:516:35: note: in expansion of macro 'ROS_LOG_COND'
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:100:35: note: in expansion of macro 'ROS_LOG'
 #define ROS_INFO_NAMED(name, ...) ROS_LOG(::ros::console::levels::Info, std::string(ROSCONSOLE_NAME_PREFIX) + "." + name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/share/mavros/workspace/src/mavros/mavros/src/plugins/sys_status.cpp:588:3: note: in expansion of macro 'ROS_INFO_NAMED'
   ROS_INFO_NAMED("sys", "VER: Flight software:     %08x (%*s)",
   ^
/opt/ros/indigo/share/mavros/workspace/src/mavros/mavros/src/plugins/sys_status.cpp:592:9: error: 'mavlink_autopilot_version_t' has no member named 'middleware_sw_version'
     apv.middleware_sw_version,
         ^
/opt/ros/indigo/include/ros/console.h:342:165: note: in definition of macro 'ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER'
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro 'ROSCONSOLE_PRINT_AT_LOCATION'
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:516:35: note: in expansion of macro 'ROS_LOG_COND'
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:100:35: note: in expansion of macro 'ROS_LOG'
 #define ROS_INFO_NAMED(name, ...) ROS_LOG(::ros::console::levels::Info, std::string(ROSCONSOLE_NAME_PREFIX) + "." + name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/share/mavros/workspace/src/mavros/mavros/src/plugins/sys_status.cpp:591:3: note: in expansion of macro 'ROS_INFO_NAMED'
   ROS_INFO_NAMED("sys", "VER: Middleware software: %08x (%*s)",
   ^
/opt/ros/indigo/share/mavros/workspace/src/mavros/mavros/src/plugins/sys_status.cpp:593:12: error: 'mavlink_autopilot_version_t' has no member named 'middleware_custom_version'
     8, apv.middleware_custom_version);
            ^
/opt/ros/indigo/include/ros/console.h:342:165: note: in definition of macro 'ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER'
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
[... 129 lines cut ...]
Invoking "make" failed
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ _FileRemoveDir
bash: _FileRemoveDir: command not found
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ Scanning dependencies of target _mavros_generate_messages_check_deps_ActuatorControl
bash: Scanning: command not found
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ [  2%] Built target _mavros_generate_messages_check_deps_ActuatorControl
bash: [: missing `]'
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ Scanning dependencies of target _mavros_generate_messages_check_deps_WaypointPull
bash: Scanning: command not found
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ [  2%] Built target _mavros_generate_messages_check_deps_WaypointPull
bash: [: missing `]'
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ Scanning dependencies of target _mavros_generate_messages_check_deps_FileRemove
bash: Scanning: command not found
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ _FileRemoveDir
bash: _FileRemoveDir: command not found
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ Scanning dependencies of target _mavros_generate_messages_check_deps_ActuatorControl
bash: Scanning: command not found
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ [  2%] Built target _mavros_generate_messages_check_deps_ActuatorControl
bash: [: missing `]'
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ Scanning dependencies of target _mavros_generate_messages_check_deps_WaypointPull
bash: Scanning: command not found
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ [  2%] Built target _mavros_generate_messages_check_deps_WaypointPull
bash: [: missing `]'
ubuntu@tegra-ubuntu:/opt/ros/indigo/share/mavros/workspace$ Scanning dependencies of target _mavros_generate_messages_check_deps_FileRemove
bash: Scanning: command not found

MarcGyongyosi, 2015-03-26:
> /opt/ros/indigo/include/rosconsole/macros_generated.h35: note: in expansion of macro 'ROS_LOG'

There's something wrong with the mavlink version as far as I know and the message definitions mess up. There is a workaround described in one of the closed/solved issues on here. You have to install mavlink separately. I got it working on a jetson tk1.

blankered, 2015-03-26:
was this your solution?
https://github.com/mavlink/mavros/issues/240

blankered, 2015-03-26:
I installed mavlink from source but catkin_make in /opt/ros/indigo/share/mavros/workspace still fails

blankered, 2015-03-26:
I was trying to compile a modified local_position.cpp file and think this was the cause of the error.

closed 2015-03-26 by blankered