mavlink/mavros #261: Stupid question for the purpose of mavros

ymkim92, 2015-04-01:
Hi,

I cannot understand why you use the mavros.
Just for using convenient monitoring and control tool of ROS such as rqt and rostopic?
Can you share the purposes with me?

vooon, 2015-04-01:
Because ROS is great for robotic application development, thats why.
In confirmation here is one example: PX4 SITL built as a bunch ROS nodes.

ymkim92, 2015-04-01:
Thanks a lot!
I have one more stupid question.
If ROS is great for robotic application development, why are PX4 and APM working on MAVLink?
I think ground control station such as mission planner and mavproxy is great! Is that all?

vooon, 2015-04-01:
Probably because ROS require linux computer, while APM initially uses 16 MIPS AVR8?
Also, do not confuse a framework with telemetry protocol.

PX4 uses MAVLink as a telemetry and command protocol, internally they use ORB that similar to ROS topics (thats why they easily use ROS topics in SITL).

ymkim92, 2015-04-01:
I want to use ROS as the framework of my own drone (both H/W and S/W)  and MAVLink as telemetry protocol.
To do that, can I use mavros? I am working on my own hardware that is not commercial like PX4.
Can you give some hint?

vooon, 2015-04-01:
For what purposes you wants atmega? I think better use STM32 or even run AP software on CPU (see erle-brain project).

Mavros node desined to speak with FCU and provide transparent link for GCS software. So in general it not fit if you wants provide ROS-AP <-> MAVLink GCS.

But if unload all standard plugins and replace by yours you could use fcu_url as gcs.
Or better use `gcs_bridge` node for creating link and use `to` and `from` topics to send/receive messages.
Pro: you may run multiple gcs links
Con: additional network io if you want only one, require node for mavlink message processing

Third way: use libmavconn in your mavlink message processing node
Pro: no extra net io, no external node
Con: number of gcs links limited by your software.
Note: current libmavconn has limitation on links count (for linux 16) per one process (defined in mavlink c-lib).

ymkim92, 2015-04-02:
In my hardware structure, ATMega takes charge in controlling of IMU, PWM, and GPIO and Arm CPU computes navigation algorithm. But I did not design this structure so I am not sure this is good or not.
Thank you very much for your advice.

closed 2015-04-02 by ymkim92