mavlink/mavros #264: Add support to DISTANCE_SENSOR msg coder/decoder

TSC21, 2015-04-04:
Related to https://github.com/PX4/Firmware/issues/1986.

TSC21, 2015-04-04:
Ideally, this has the following features:
- [ ] rosserial connection
- [ ] `DISTANCE_SENSOR` coder (decoder may only make sense if sensor is connected on the FCU)

mhkabir, 2015-04-05:
A decoder probably makes more sense since it is better to connect whatever low level peripherals you want (e.g Laser ranger)  to your flight controller. 
Not much point using the companion computer for this. It would use up a serial port too. 
And serial isn't the best bus for external sensors. Use a more appropriate bus  (SPI,  UAVCAN) for interfacing external sensors. 

For the decoder to ROS, see the flow plugin. Add a appropriate handler in a new plugin and send in a a PR :D

mhkabir, 2015-04-05:
PX4 implementation would involve adding a mavlink message stream to read from the distance sensor uORB topic.

TSC21, 2015-04-05:
@mhkabir thanks for your inputs. I wasn't really asking to this to be implemented, more like this is an ongoing implementation where ideas are welcomed :D 
The decoder is certainly easy to get done, also as the coder. The PX4 implementation though can involve adding support to this message I think.

vooon, 2015-05-13:
Plugin almost done, by #292 . Need test and fix todo.

vooon, 2015-05-13:
(#292 merged, my fixes also uses that number. Perhaps better use this issue, but too late)

TSC21, 2015-05-17:
@vooon new sensor type was added: https://github.com/mavlink/mavlink/pull/381/files. So, ideally @tfoote could add `LASER` `radiation_type` to `Range.msg` enum (reason: as I explained in the comment on https://github.com/mavlink/mavros/pull/292#issuecomment-101754507).

TSC21, 2015-05-17:
Another thing: https://github.com/mavlink/mavlink/pull/383/files - can this enum be added then to `/mavros/mavros/include/mavros` in `sensor_orientation.h` or similar?

vooon, 2015-05-17:
@TSC21 as soon as it merged i will do release of mavlink package.

TSC21, 2015-05-17:
OK. Any ideas how to then include it here: https://github.com/TSC21/mavros/blob/external_sensors_connect/mavros_extras/src/plugins/distance_sensor.cpp#L231?

TSC21, 2015-05-17:
Update: https://github.com/mavlink/mavros/pull/296#issuecomment-102821032

vooon, 2015-05-17:
Make a std::array with orientations then choose one by orientation enum.
Also add param for setting parent frame_id, child frame_id same as in Range.header.

As i understand tf: you don't need actually rotate FCU frame_id, only publish sensor orientation value.

TSC21, 2015-05-18:
OK I think I will do this in a next PR. You can merge this one in the mean time, if everything's ok.

TSC21, 2015-05-18:
@vooon https://github.com/mavlink/mavlink/pull/383 was merged. You can do mavlink release now.

vooon, 2015-05-18:
Already done.

vooon, 2015-05-18:
- https://github.com/ros/rosdistro/pull/8482
- https://github.com/ros/rosdistro/pull/8483
- https://github.com/ros/rosdistro/pull/8484

TSC21, 2015-05-18:
> Also add param for setting parent frame_id, child frame_id same as in Range.header.

Why do we need parent frame_id, if we only want tf between `fcu` and the sensor? Doesn't make sense to add other tfs, I think.

> As i understand tf: you don't need actually rotate FCU frame_id, only publish sensor orientation value.

We don't rotate fcu frame. We rotate the sensor id accoriding to the enum and then the tf between the fcu and the sensor is published, with the proper orientation of the sensor being updated.

TSC21, 2015-05-18:
And do you merge now or are you waiting for the orientation stuff?

vooon, 2015-05-18:
Waiting orientation and **test results**.

TSC21, 2015-05-18:
Results from which side? Cause ROS side working ok. I'm getting the PX4 side worked on now.

vooon, 2015-05-18:
Ok, if you say that plugin works, it's enough for now. I can merge current PR if you need that. But probably better merge when all features done.

TSC21, 2015-05-18:
No need to merge, unless some other PR is added. I'm advancing to create the `std::array sensor_orientation`. One question: isn't it better to add it to `uas` definition, so to be used by other plugins if needed?

vooon, 2015-05-18:
Because it not depend on uas data it may be simple helper function in
libmavros. Please make new cpp.

TSC21, 2015-05-18:
You mean here: https://github.com/mavlink/mavros/tree/master/mavros/include/mavros?

vooon, 2015-05-18:
I mean src/lib, and add function definition to utils.h
19.05.2015 0:24 пользователь "TSC21" <notifications@github.com> написал:

TSC21, 2015-05-18:
so what do I put in `/src/lib/sensor_orientation.h` and what do I put in `/include/mavros/utils.h`?

vooon, 2015-05-18:
C++ and C split definition and implementation in `.h` and `.cpp` (`.c`) files. So you should put function definition in `utils.h` and implementation in `/src/lib/sensor_orientation.cpp`.

Then just use `<mavros/utils.h>`.

TSC21, 2015-05-18:
> C++ and C split definition and implementation in .h and .cpp (.c) files. So you should put function definition in utils.h and implementation in /src/lib/sensor_orientation.cpp.
Then just use <mavros/utils.h>.

OK that was what I was already doing :) thanks

TSC21, 2015-05-18:
In `utils.h`, do I have to use `inline`, since the method implementation is in `/src/lib/sensor_orientation.cpp`?

vooon, 2015-05-18:
No! Inline for `.h` files. And there regular function, so no `static`, no `inline` no `constexpr`.

TSC21, 2015-05-18:
Willco :+1:

TSC21, 2015-06-11:
This is tested and working ;) so can be closed, since https://github.com/mavlink/mavros/pull/292 has been merged!

closed 2015-06-11 by TSC21