mavlink/mavros #267: Warn appears when using mavros through 3dr radio telemetery

wangsen1312, 2015-04-08:
When I  want to connect to my pixhawk with px4  using the command line below:
roslaunch mavros px4.launch fcu_url:='/dev/ttyUSB0:57600'

some warn and error appears 

[ INFO] [1428526279.843410104]: Plugin vision_speed_estimate loaded and initialized
[ INFO] [1428526279.850209078]: Plugin visualization loaded and initialized
[ INFO] [1428526279.858127256]: Plugin waypoint loaded and initialized
[ INFO] [1428526279.858220369]: Autostarting mavlink via USB on PX4
[ INFO] [1428526279.858268901]: Built-in mavlink dialect: ardupilotmega
[ INFO] [1428526279.858299536]: MAVROS started. MY ID [1, 240], TARGET ID [1, 50]
[ WARN] [1428526280.043101645]: TM: Clock skew detected (-1428525963.312890291 s). Hard syncing clocks.
[ INFO] [1428526280.507462306]: CON: Got HEARTBEAT, connected.
[ INFO] [1428526280.681539646]: High resolution IMU detected!
[ WARN] [1428526280.703567434]: GPS: no fix
[ INFO] [1428526280.759456458]: RC_CHANNELS message detected!
[ERROR] [1428526281.947240470]: FCU: command unsupported: 520
[ WARN] [1428526282.512012283]: VER: request timeout, retries left 1
[ERROR] [1428526282.812494516]: FCU: command unsupported: 520
[ WARN] [1428526283.511819166]: VER: request timeout, retries left 0
[ERROR] [1428526284.080156548]: FCU: command unsupported: 520
[ WARN] [1428526284.511943177]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[ERROR] [1428526284.769493511]: FCU: command unsupported: 520
[ WARN] [1428526290.047965821]: TM: Clock skew detected (0.035652834 s). Hard syncing clocks.
[ INFO] [1428526290.920454296]: FCU: [pm] sending list
[ INFO] [1428526295.804754278]: WP: mission received
[ INFO] [1428526295.852786731]: FCU: WPM: mission is empty
[ERROR] [1428526296.093861315]: FCU: REJ. WP CMD: partner id mismatch
[ WARN] [1428526300.124791705]: TM: Clock skew detected (-0.011057769 s). Hard syncing clocks.
[ WARN] [1428526310.206588022]: TM: Clock skew detected (-0.090429983 s). Hard syncing clocks.
[ WARN] [1428526320.319566391]: TM: Clock skew detected (-0.061695946 s). Hard syncing clocks.
[ WARN] [1428526330.432523301]: TM: Clock skew detected (0.086793319 s). Hard syncing clocks.
[ INFO] [1428526335.802996067]: PR: parameters list received
[ WARN] [1428526340.634352970]: TM: Clock skew detected (0.036648383 s). Hard syncing clocks.
[ WARN] [1428526341.407710771]: GPS: no fix
[ WARN] [1428526350.731952500]: TM: Clock skew detected (0.086464618 s). Hard syncing clocks.
[ WARN] [1428526360.807835889]: TM: Clock skew detected (-0.027864879 s). Hard syncing clocks.
[ WARN] [1428526371.177861656]: TM: Clock skew detected (-0.161407933 s). Hard syncing clocks.
[ WARN] [1428526381.246760665]: TM: Clock skew detected (0.085835654 s). Hard syncing clocks.
[ WARN] [1428526391.346681837]: TM: Clock skew detected (-0.034732230 s). Hard syncing clocks.

Can you give me some suggestion? Thank you very much!

vooon, 2015-04-08:
1. Please next use markdown code block (```)
2. `command unsupported: 520` - not a bug, just ignore
3. Default `px4_config.yaml` tuned for onboard direct connection (with low delay), you must disable time synchronization by setting `~conn/timesync` to 0

wangsen1312, 2015-04-10:
@vooon  Thank you for your help. It's my first time use mavros. Can you give me a simple sample such as read the attitude of px4 or write something into it. Thank you very much!

vooon, 2015-04-11:
PX4 wiki has examples: https://pixhawk.org/dev/ros/mavros_offboard

Altitude etc: http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.global_position
Attitude: http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.imu_pub

closed 2015-04-30 by wangsen1312