mavlink/mavros #268: mavros_extras do not compile

vmayoral, 2015-04-09:
Greetings,

I'm trying to compile mavros_extras in Indigo and here's the output:
```bash
Building CXX object CMakeFiles/mavros_extras.dir/src/plugins/image_pub.cpp.o
In file included from /home/victor/indigo/install_isolated/include/class_loader/class_loader_core.h:39:0,
                 from /home/victor/indigo/install_isolated/include/class_loader/class_loader_register_macro.h:33,
                 from /home/victor/indigo/install_isolated/include/class_loader/class_loader.h:39,
                 from /home/victor/indigo/install_isolated/include/pluginlib/class_list_macros.h:40,
                 from /home/victor/indigo/src/mavros_extras/src/plugins/px4flow.cpp:19:
/home/victor/indigo/install_isolated/include/class_loader/meta_object.h: In instantiation of 'B* class_loader::class_loader_private::MetaObject<C, B>::create() const [with C = mavplugin::PX4FlowPlugin; B = mavplugin::MavRosPlugin]':
/home/victor/indigo/src/mavros_extras/src/plugins/px4flow.cpp:129:1:   required from here
/home/victor/indigo/install_isolated/include/class_loader/meta_object.h:193:17: error: invalid new-expression of abstract class type 'mavplugin::PX4FlowPlugin'
      return new C;
                 ^
/home/victor/indigo/src/mavros_extras/src/plugins/px4flow.cpp:31:7: note:   because the following virtual functions are pure within 'mavplugin::PX4FlowPlugin':
 class PX4FlowPlugin : public MavRosPlugin {
       ^
In file included from /home/victor/indigo/src/mavros_extras/src/plugins/px4flow.cpp:18:0:
/home/victor/indigo/install_isolated/include/mavros/mavros_plugin.h:71:15: note: 	virtual void mavplugin::MavRosPlugin::initialize(mavros::UAS&, ros::NodeHandle&, diagnostic_updater::Updater&)
  virtual void initialize(UAS &uas,
               ^
/home/victor/indigo/install_isolated/include/mavros/mavros_plugin.h:78:28: note: 	virtual const string mavplugin::MavRosPlugin::get_name() const
  virtual const std::string get_name() const = 0;
                            ^
CMakeFiles/mavros_extras.dir/build.make:54: recipe for target 'CMakeFiles/mavros_extras.dir/src/plugins/px4flow.cpp.o' failed
make[2]: *** [CMakeFiles/mavros_extras.dir/src/plugins/px4flow.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/victor/indigo/install_isolated/include/class_loader/class_loader_core.h:39:0,
                 from /home/victor/indigo/install_isolated/include/class_loader/class_loader_register_macro.h:33,
                 from /home/victor/indigo/install_isolated/include/class_loader/class_loader.h:39,
                 from /home/victor/indigo/install_isolated/include/pluginlib/class_list_macros.h:40,
                 from /home/victor/indigo/src/mavros_extras/src/plugins/image_pub.cpp:18:
/home/victor/indigo/install_isolated/include/class_loader/meta_object.h: In instantiation of 'B* class_loader::class_loader_private::MetaObject<C, B>::create() const [with C = mavplugin::ImagePubPlugin; B = mavplugin::MavRosPlugin]':
/home/victor/indigo/src/mavros_extras/src/plugins/image_pub.cpp:216:1:   required from here
/home/victor/indigo/install_isolated/include/class_loader/meta_object.h:193:17: error: invalid new-expression of abstract class type 'mavplugin::ImagePubPlugin'
      return new C;
                 ^
/home/victor/indigo/src/mavros_extras/src/plugins/image_pub.cpp:34:7: note:   because the following virtual functions are pure within 'mavplugin::ImagePubPlugin':
 class ImagePubPlugin : public MavRosPlugin {
       ^
In file included from /home/victor/indigo/src/mavros_extras/src/plugins/image_pub.cpp:17:0:
/home/victor/indigo/install_isolated/include/mavros/mavros_plugin.h:71:15: note: 	virtual void mavplugin::MavRosPlugin::initialize(mavros::UAS&, ros::NodeHandle&, diagnostic_updater::Updater&)
  virtual void initialize(UAS &uas,
               ^
/home/victor/indigo/install_isolated/include/mavros/mavros_plugin.h:78:28: note: 	virtual const string mavplugin::MavRosPlugin::get_name() const
  virtual const std::string get_name() const = 0;
                            ^
CMakeFiles/mavros_extras.dir/build.make:77: recipe for target 'CMakeFiles/mavros_extras.dir/src/plugins/image_pub.cpp.o' failed
make[2]: *** [CMakeFiles/mavros_extras.dir/src/plugins/image_pub.cpp.o] Error 1
CMakeFiles/Makefile2:560: recipe for target 'CMakeFiles/mavros_extras.dir/all' failed
make[1]: *** [CMakeFiles/mavros_extras.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

```

mhkabir, 2015-04-09:
Is this hydro? I think I remember this from back then.

vmayoral, 2015-04-09:
@vooon this is Indigo.

vooon, 2015-04-09:
What version you tried to compile? In latest 0.11.1 there is no more `get_name()` (API changes).

vmayoral, 2015-04-09:
I can try reproducing it in Hydro if you consider necessary.

vmayoral, 2015-04-09:
I compiled master. Let me clean the package sources and builds and re do it all over.

vooon, 2015-04-09:
Test new build tool: `catkin` (not sure that it preinstalled with indigo)

VERY useful tool, can be called anywhere in ws, has `catkin clean --all`

vmayoral, 2015-04-09:
Rebuilding `mavros` i get an error with it:
```bash
/home/victor/indigo/src/mavros/src/plugins/sys_status.cpp: In member function 'void mavplugin::SystemStatusPlugin::autopilot_version_cb(const ros::TimerEvent&)':
/home/victor/indigo/src/mavros/src/plugins/sys_status.cpp:628:26: error: 'MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES' was not declared in this scope
    cmd.request.command = MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES;
                          ^
CMakeFiles/mavros_plugins.dir/build.make:77: recipe for target 'CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o' failed
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o] Error 1

```

vooon, 2015-04-09:
Update mavlink, current 2015.4.4, but 2015.3.13 fine too.

vmayoral, 2015-04-09:
That's perfect.

closed 2015-04-09 by vmayoral