mavlink/mavros #271: mavros_extras: setup PX4FLOW

vmayoral, 2015-04-09:
Trying to set up the PX4Flow with Erle-Brain. `mavros` and `mavros_extras` have been properly compiled and all dependencies satisfied.

I launch APM with the following flags:
```bash
./ArduCopter.elf -A udp:127.0.0.1:6001 -B /dev/ttyO5 
```

`mavros` launches with:
```bash
rosrun mavros mavros_node _fcu_url:="udp://:6001@"
```
and outputs something like:
```bash
(ErleRobotics)root@localhost:/home/ubuntu/indigo/install_isolated# rosrun mavros mavros_node _fcu_url:="udp://:6001@"
[ INFO] [1426599286.701109289]: FCU URL: udp://:6001@
[ INFO] [1426599286.712283331]: udp0: Bind address: 0.0.0.0:6001
[ INFO] [1426599286.720940081]: GCS URL: udp://@
[ INFO] [1426599286.726135831]: udp1: Bind address: 0.0.0.0:14555
[ INFO] [1426599286.732669123]: udp0: Remote address: 127.0.0.1:46436
[ INFO] [1426599287.598695581]: Plugin 3dr_radio loaded and initialized
[ INFO] [1426599287.687727915]: Plugin actuator_control loaded and initialized
[ INFO] [1426599287.824488498]: Plugin command loaded and initialized
[ INFO] [1426599288.031085665]: Plugin ftp loaded and initialized
[ INFO] [1426599288.192630540]: Plugin global_position loaded and initialized
[ WARN] [1426599288.239407665]: TF to MSG: Quaternion Not Properly Normalized
[ INFO] [1426599288.252693081]: Plugin gps loaded and initialized
[ WARN] [1426599288.310199331]: GPS: no fix
[ WARN] [1426599288.464124040]: TF to MSG: Quaternion Not Properly Normalized
[ WARN] [1426599288.649995415]: TF to MSG: Quaternion Not Properly Normalized
[ WARN] [1426599288.855937915]: TF to MSG: Quaternion Not Properly Normalized
[ WARN] [1426599289.061734123]: TF to MSG: Quaternion Not Properly Normalized
[ WARN] [1426599289.246490248]: TF to MSG: Quaternion Not Properly Normalized
[ WARN] [1426599289.461780165]: TF to MSG: Quaternion Not Properly Normalized
[ WARN] [1426599289.644748665]: TF to MSG: Quaternion Not Properly Normalized
[ WARN] [1426599289.850126415]: TF to MSG: Quaternion Not Properly Normalized
[ WARN] [1426599290.076556040]: TF to MSG: Quaternion Not Properly Normalized
[ INFO] [1426599290.164956706]: Plugin image_pub loaded and initialized
[ WARN] [1426599290.252427290]: TF to MSG: Quaternion Not Properly Normalized
[ INFO] [1426599290.325711706]: Plugin imu_pub loaded and initialized
[ WARN] [1426599290.386178415]: RAW_IMU: linear acceleration known on APM only
[ INFO] [1426599290.392302540]: Plugin local_position loaded and initialized
[ INFO] [1426599290.510106665]: Plugin mocap_pose_estimate loaded and initialized
[ INFO] [1426599290.583699707]: Plugin param loaded and initialized
[ INFO] [1426599290.696697748]: Plugin px4flow loaded and initialized
[ INFO] [1426599290.837424457]: Plugin rc_io loaded and initialized
[ INFO] [1426599290.943830957]: Plugin safety_area loaded and initialized
[ INFO] [1426599291.046982790]: Plugin setpoint_accel loaded and initialized
[ INFO] [1426599291.321804498]: Plugin setpoint_attitude loaded and initialized
[ INFO] [1426599291.467325207]: Plugin setpoint_position loaded and initialized
[ INFO] [1426599291.556707415]: Plugin setpoint_velocity loaded and initialized
[ INFO] [1426599291.685578498]: Plugin sys_status loaded and initialized
[ INFO] [1426599291.762444082]: Plugin sys_time loaded and initialized
[ INFO] [1426599291.800729332]: Plugin vfr_hud loaded and initialized
[ WARN] [1426599291.845430623]: TM: Wrong FCU time.
[ INFO] [1426599291.971965373]: Plugin vision_pose_estimate loaded and initialized
[ INFO] [1426599292.075278832]: Plugin vision_speed_estimate loaded and initialized
[ INFO] [1426599292.155975832]: Plugin visualization loaded and initialized
[ INFO] [1426599292.256839040]: CON: Got HEARTBEAT, connected.
[ INFO] [1426599292.281264373]: Plugin waypoint loaded and initialized
[ INFO] [1426599292.287390540]: Built-in mavlink dialect: ardupilotmega
[ INFO] [1426599292.293125748]: MAVROS started. MY ID [1, 240], TARGET ID [1, 1]
[ INFO] [1426599295.243614082]: FCU: PERF: 5/1000 23933 1222

[ WARN] [1426599303.262409250]: PR: request list timeout, retries left 2
[ WARN] [1426599303.319563833]: VER: request timeout, retries left 1
[ INFO] [1426599303.322016250]: VER: Capabilities 0x0000000000000000
[ INFO] [1426599303.322943833]: VER: Flight software:     00000000 (0ca56f1c)
[ INFO] [1426599303.326137416]: VER: Middleware software: 00000000 (        )
[ INFO] [1426599303.336139625]: VER: OS software:         00000000 (        )
[ INFO] [1426599303.342431083]: VER: Board hardware:      00000000
[... 101 lines cut ...]
/mavros/rc/override
/mavros/safety_area/set
/mavros/setpoint_accel/accel
/mavros/setpoint_attitude/att_throttle
/mavros/setpoint_attitude/cmd_vel
/mavros/setpoint_position/local
/mavros/setpoint_velocity/cmd_vel
/mavros/state
/mavros/time_reference
/mavros/vfr_hud
/mavros/vision_pose/pose
/mavros/vision_speed/speed_vector
/mavros/visualization/track_markers
/mavros/visualization/vehicle_marker
/mavros/wind_estimation
/rosout
/rosout_agg
/tf
```

Unfortunately, nothing gets published in nodes such as `px4flow/ground_distance`
```bash
rostopic echo /mavros/px4flow/ground_distance
```
 @mhkabir, @vooon, mind looking at this walkthrough? I'm expecting the PX4FLOW module to work however it's not. Maybe the drivers? (my Linux distribution might have shipped them)

vooon, 2015-04-09:
Please use `roslaunch mavros apm.launch fcu_url:=udp://:6001@ gcs_url:=udp://@` (if you need gcs bridge).

Then start **another** instance for px4flow: `roslaunch mavros_extras px4flow.launch`

vooon, 2015-04-09:
Also note, current px4flow plugin only gets data from device.

mhkabir, 2015-04-10:
You need two mavros instances to stream from PX4Flow and APM.

Architecturally I'd say you should use the i2c connection on the flight
controller and stream a flow message from there using mavlink if you need
to do stuff with it offboard.

vmayoral, 2015-04-10:
@vooon and @mhkabir thanks for the help.

Followed @vooon tips and launched `mavros` ([log](https://gist.github.com/vmayoral/7a4915eb6fa35715aa04)) and then `mavros_extras` ([log](https://gist.github.com/vmayoral/aac1ee74ad7ff88625ad)) through `roslaunch`.

The topics available now are:
```bash
(ErleRobotics)root@localhost:/home/ubuntu# rostopic list
/diagnostics
/mavlink/from
/mavlink/to
/mavros/battery
/mavros/global_position/compass_hdg
/mavros/global_position/global
/mavros/global_position/gp_vel
/mavros/global_position/local
/mavros/global_position/rel_alt
/mavros/gps/fix
/mavros/gps/gps_vel
/mavros/imu/atm_pressure
/mavros/imu/data
/mavros/imu/data_raw
/mavros/imu/mag
/mavros/imu/temperature
/mavros/mission/waypoints
/mavros/radio_status
/mavros/rc/in
/mavros/rc/out
/mavros/rc/override
/mavros/setpoint_accel/accel
/mavros/setpoint_position/local
/mavros/setpoint_velocity/cmd_vel
/mavros/state
/mavros/time_reference
/mavros/vfr_hud
/mavros/wind_estimation
/px4flow/battery
/px4flow/image/camera_image
/px4flow/mavlink_from
/px4flow/mavlink_to
/px4flow/px4flow/ground_distance
/px4flow/px4flow/raw/optical_flow_rad
/px4flow/px4flow/temperature
/px4flow/state
/px4flow/time_reference
/rosout
/rosout_agg
/tf
```

LEDs on the PX4FLOW module behave the same way:
>The LEDs combination of the PX4FLOW also notices that something might not be good. It starts with a green solid LED and a orange blinking one. After a few seconds it switches into a green solid LED, orange solid, red blinking and blue blinking even faster

Unfortunately `rostopic echo /px4flow/px4flow/ground_distance` returns still nothing. I'm expecting to see some output when *echoing* this topic. Is my expectation mistaken? If not this way, how can i test/verify that the optical flow is working correctly?

mhkabir, 2015-04-10:
Hi Victor,

Could you please update the flow Firmware from Qgroundcontrol. Something
seems to be off with the board Firmware. The ERR led shouldn't be on.

Check if you can connect to the board in QGC after that.

vmayoral, 2015-04-10:
@mhkabir thanks. Did it and the PX4FLOW now seems to behave better (`red` LED is not blinking).

Tried also using `roslaunch mavros apm.launch fcu_url:=udp://:6001@` (thanks for the bridge pointer, quite useful) and `roslaunch mavros_extras px4flow.launch` and now finally i can see the topics publishing stuff :).

**Our goal from here is to interface PX4FLOW with Linux autopilots**. @ahcorde and I are willing to put time but we first need to understand the situation:

>Also note, current px4flow plugin only gets data from device.

We presume there's no way for the FCU to use the information being published by ROS thereby in order to make it work with a Linux autopilot we'll need to:
 1. connect the PX4FLOW using I2C as @mhkabir suggested and code a driver for the `AP_HAL_Linux` at https://github.com/diydrones/ardupilot/tree/master/libraries/AP_OpticalFlow (not sure how to do this though, I'd appreciate a few pointers).
 2.  extend the `mavros_extras` code and (interfacing through `/mavlink/to`) send the flow information to the autopilot.

@vooon and @mhkabir what's your opinion? Which is the best way to go? Or even, is there an alternative option?

Thanks!

-----

A few notes:
 - `rosrun image_view image_view image:=/px4flow/image/camera_image` allows me to see the image from the PX4FLOW however the image doesn't make much sense and the matrix interpretation seems wrong (e.g. when focusing on a person hands are twisted and upside down)

- Apparently this is harmless and connected to https://github.com/mavlink/mavros/issues/246. When launching `mavros_extras`, i can see a message saying. 
```bash
 WARN] [1426609661.574445541]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
```
 - Surprisingly, the `FLOW_ENABLE` parameter seems to bet set to 0 every time a GCS connects to the FCU (Erle-Brain in this case). We saw this behavior with at least MAVProxy and APM Planner 2.0.

mhkabir, 2015-04-10:
@vmayoral You should go with i2c. The i2c integral frame does not miss data unlike serial and accumulation is done on-sensor.
1. There was an option for sending the flow data to your flight controller (a relay) from mavros. I removed it to avoid the very issue we wish to mitigate by forcing people to use i2c. Currently on PX4, we only support i2c connections actively.
2. The APM codebase already has flow via i2c working. If you wish, give me a day and I'll get flow running on the Erle Brain ;) I tried previously but left it out as I had far more pressing jobs. It should work fine as long as the i2c drivers work. There should be no difference in running it on a different platform as long as the abstraction layer is done properly.
3. The image viewer isn't actually very useful for other tasks. Just for focusing the lens when the ```VIDEO_ONLY``` param is enabled. You should otherwise see the 64x64 pixel patch we use for flow computation. That said, we might need to check the image encodings.
4. You might need to write the ```FLOW_ENABLE ```param to memory ( IIRC APM should do that automatically)

vmayoral, 2015-04-12:
@mhkabir,

I've been looking into the APM codebase support for the PX4FLOW module and i have not been able to identify the drivers but solely [a simple interface](https://github.com/erlerobot/ardupilot/tree/master/libraries/AP_OpticalFlow). I assume that the drivers are within the PX4Firmware code.

`AP_HAL_Linux` does not use PX4Firmware thereby we probably need to port the driver to Linux. We definitely can use your help here.
Regards,

mhkabir, 2015-04-12:
Aah yes. APM uses the PX4 low-level device drivers for most peripherals. 
The driver shouldn't be too difficult. Just one i2c transfer. I'll be happy to assist via skype :)

vmayoral, 2015-04-12:
I put some time into a Linux userspace I2C driver. Please have a look at https://github.com/erlerobot/ardupilot/blob/px4flow/libraries/AP_OpticalFlow/AP_OpticalFlow_Linux.cpp.

PX4Firmware code is using a `transfer` to read a few bytes from a specific I2C register. I matched it with [readRegisters](https://github.com/erlerobot/ardupilot/blob/px4flow/libraries/AP_HAL_Linux/I2CDriver.h#L32) primitive but i might need to implement a new one. What's your opinion @mhkabir?

vooon, 2015-04-15:
@vmayoral just looked your source, and it looks ok. Only one: maybe better move semaphore release right after transfer.

But i think that this issue can be closed (as it not mavros related).

closed 2015-04-15 by vooon

vmayoral, 2015-04-29:
@vooon and @mhkabir,

We keep blocked on the Linux-driver side. @mhkabir mentioned that he previously used `mavros` to send the information (through the `mavlink` topic i guess). Could anyone point out which are these commits on how to bring this back again?

Thanks,

vooon, 2015-04-29:
That was working before FX4Flow update, not sure about present status.
Old code: https://github.com/mavlink/mavros/blob/901dae0a98e5da20fba6ff1ca1767e7ec98fd641/mavros_extras/src/plugins/px4flow.cpp