mavlink/mavros #272: Gyro raw messages

dani-carbonell, 2015-04-14:
Hi, I am currently working in a wheeled robot with ROS and we calculate the odometry using a high precision gyro and wheel encoders. We would like to use Pixhawk instead of this expensive gyro but using the orientation value (yaw) of the "imu/data" topic leads to a bad odomety if compared with the first one.

Could we have access to the gyro (raw - scaled) values in the topic "/imu/data_raw"?

After performing mag/gyro/accel calibration it still shows a non linearity when robot is turning in place.
[image]

vooon, 2015-04-14:
Not sure that it is mavros side, because for PX4 it is wery simple: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/imu_pub.cpp#L244

Try connect QGC and plot `HIGHRES_IMU.zgyro`.

vooon, 2015-06-03:
Updates?

dani-carbonell, 2015-06-04:
I've been able to configure the Pixhwak to output raw IMU data and now I can integrate gyro orientation at 150Hz. Additionally I have created a node that performs offset calibration and it shows relative good performance if compared with a much more expensive gyro. In the other hand I would like to confirm that internal gyro calibration on the Pixhawk does not calibrate scale of the gyro. In the image can be seen how scale is not the same than in the more expensive gyro and consecutive turns in place show deviation.

[image]

closed 2015-06-22 by mhkabir