mavlink/mavros #274: [param] In latest PX4 param indexes not a constant

vooon, 2015-04-14:
With latest master i got lots of:
```
[ WARN] [1429040837.846295115]: PR: Param TRIM_ROLL index(21->22)/count(361->362) changed! FCU changed?
[ WARN] [1429040837.885361130]: PR: Param TRIM_YAW index(19->20)/count(361->362) changed! FCU changed?
[ WARN] [1429040837.951246953]: PR: Param UAVCAN_ENABLE index(361->362)/count(361->362) changed! FCU changed?
```

It is not a bug, and i can ignore that now, but param update speed is slowed by this logic.

(I'm doing that man: https://pixhawk.org/peripherals/radio-control/opentx/single_channel_mode_switch )

LorenzMeier, 2015-04-14:
That's an issue on our end I need to fix. Thanks for letting me know. It should be stable after boot.

vooon, 2015-06-28:
Seems that issue still present in px4 stable. Also on companion link i often got rerequests about 90 params. So startup are long...

DonLakeFlyer, 2015-06-28:
> Also on companion link i often got rerequests about 90 params

Just FYI: If you are going over radio it's fairly common to get high re-request counts.

vooon, 2015-06-28:
@DonLakeFlyer direct UART connection to OBC on 921600.

DonLakeFlyer, 2015-06-28:
Something wrong then.

LorenzMeier, 2015-06-29:
I didn't get these via either radio or USB. Let me check your particular setup. I think the fix to this is to change the param find logic to a compile time thing. Working on that.

closed 2015-10-31 by vooon