mavlink/mavros #275: Without GPS, Does the /mavros/setpoint_position/local work?

wangsen1312, 2015-04-16:
Now I can arm the plane and I don't know how can I use the mavros to take off the plane?
I can use the vicon mocap system to capture the position?
Can I use vicon_bridge to publish it ? and does /mavros/setpoint_position/local works??
Thank you very much!

vooon, 2015-04-16:
It definitely works with PX4. What FCU do you use?

vooon, 2015-04-16:
But better address that question to FCU devs, but change topic with [`SET_POSITION_TARGET_LOCAL_NED`](https://pixhawk.ethz.ch/mavlink/#SET_POSITION_TARGET_LOCAL_NED).

wangsen1312, 2015-04-17:
@vooon   I use the native PX4 with ArduCopter V3.2.1.   I am the beginner with the ros and mavros, the topic such as SET_POSITION_TARGET_LOCAL_NED  in mavlink not in mavros I don't know how to call it. Can you give me a piece of example? Thank you very much!

vooon, 2015-04-17:
PX4 or APM? Pixhawk (hardware) can run PX4 native firmware or APM firmware (which uses some parts of PX4 on that platform).
From your answer i suggest APM, and i don't know. Support for `SET_POSITION_TARGET_LOCAL_NED` is a new feature in APM:Copter, and i'm lack of user reports about how this works.

Most of this setpoint plugins translates ros messages into `SET_POSITION_TARGET_LOCAL_NED`, there no magic (except converting orientation frames from #49).

As a example look code of [`mavsetp`](https://github.com/mavlink/mavros/blob/master/mavros/scripts/mavsetp) which uses [`mavros.setpoint`](https://github.com/mavlink/mavros/blob/master/mavros/src/mavros/setpoint.py).

wangsen1312, 2015-04-17:
@vooon  I'm sure  PX4 native firmware.  I use roslaunch px4.launch to connect it and safety arm to arm it.
I find plugins 6.14 setpoint_position (Sent LOCAL_NED_POSITION_SETPOINT_EXTERNAL to FCU) http://wiki.ros.org/mavros    Does it work now?
Thank you very much!

vooon, 2015-04-17:
Post here result of: `rostopic echo -n1 /diagnostics` mavros knows most of firmwares (by it's number in heartbeat, don't forget \`\`\` for code block).
`LOCAL_NED_POSITION_SETPOINT_EXTERNAL` renamed to `SET_POSITION_TARGET_LOCAL_NED`. It works, but i'm unsure about disabled GPS (at least requires proper configuration of FCU).
But @mhkabir does positioning without GNSS, so i think it is possible.

LorenzMeier, 2015-04-18:
Anyone interested in testing this one?
https://github.com/PX4/Firmware/pull/1983

wangsen1312, 2015-04-28:
@vooon  After echo the diagnostics: the result is
header:  
  seq: 42
  stamp: 
    secs: 1430235591
    nsecs: 65052247
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 0
      - 
        key: Tx sequence number:
        value: 0
      - 
        key: Rx total bytes:
        value: 37619
      - 
        key: Tx total bytes:
        value: 1177
      - 
        key: Rx speed:
        value: 935.000000
      - 
        key: Tx speed:
        value: 20.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 1
      - 
        key: EPH (m)
        value: 99.99
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Heartbeats since startup
        value: 37
      - 
        key: Frequency (Hz)
        value: 1.099252
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: ArduPilotMega
      - 
        key: Mode
        value: STABILIZE
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Sensor present
        value: 0x0061FC2F
      - 
        key: Sensor enabled
        value: 0x00619C2F
      - 
        key: Sensor helth
        value: 0x0061FC2F
      - 
        key: Sensor 3D Gyro
        value: Ok
      - 
        key: Sensor 3D Accel
        value: Ok
      - 
        key: Sensor 3D Mag
        value: Ok
      - 
        key: Sensor Abs Pressure
        value: Ok
      - 
        key: Sensor GPS
        value: Ok
      - 
        key: Sensor Ang Rate Control
        value: Ok
      - 
        key: Sensor Attitude Stab
        value: Ok
      - 
        key: Sensor Yaw Position
        value: Ok
      - 
        key: Sensor Motor Outputs
        value: Ok
      - 
        key: Sensor RC Receiver
        value: Ok
      - 
        key: AHRS health
        value: Ok
      - 
        key: Terrain health
        value: Ok
      - 
        key: CPU Load (%)
        value: 15.6
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 1
    name: mavros: Battery
    message: Low voltage
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Voltage
        value: 0.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 100.0
  - 
    level: 0
    name: mavros: 3DR Radio
    message: Normal
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: RSSI
        value: 149
      - 
        key: RSSI (dBm)
        value: -48.6
      - 
        key: Remote RSSI
        value: 155
      - 
        key: Remote RSSI (dBm)
        value: -45.4
      - 
        key: Tx buffer (%)
        value: 100
      - 
        key: Noice level
        value: 36
      - 
        key: Remote noice level
        value: 34
      - 
        key: Rx errors
        value: 4
      - 
        key: Fixed
        value: 3

PS:  I can get the data form IMU with n.subscribe("/mavros/imu/data", 1, localori)  in ROS
but I can't get anything from n.subscribe("/mavros/local_position/local", 1, localpos) 
Can you give me some suggestion?

vooon, 2015-04-28:
You should use `apm.launch`, and perhaps FCU don't send `LOCAL_POSITION_NED`. No message - no data on topic.

wangsen1312, 2015-04-28:
Thank you very much!  ''' apm.launch '' works well!
However   "/mavros/local_position/local" still not working, Does it mean FCU internal position estimator not works well or I need something to boot it?

wangsen1312, 2015-04-28:
@vooon   Thank you very much!!!   apm.launch   works well!
However "/mavros/local_position/local" still not working, Does it mean my FCU internal position estimator not works well or I need something to boot it because I want to use "/mavros/setpoint_position/local" to publish data in the near future.
Thank you in advance.

vooon, 2015-04-29:
Hmm, it strange, but you could check it with APM Planner 2. There exist widget with all messages.

No update on mavros topics almost always because FCU don't send that message.

wangsen1312, 2015-04-29:
@voon  I just awared that I used the pixhawk, so does apm.launch fine? 
PS: I can connect to APM planner 2, and see the status widget, which one is the local_position?  
Thank you very much!

vooon, 2015-04-29:
`LOCAL_POSITION_NED`. Pixhawk with **ArduCopter** firmware == APM.

wangsen1312, 2015-04-30:
@voon  Thank you very much! 
After Google several times, maybe We can't get the local_position without GPS. So maybe I also can't set local_position without GPS.
PS: Now I have get the local position from VICON, I can send the attitude to TOPIC mocap_pose_estimate and vision_pose_estimate.  I don't know whether these two topics have any difference and how can I use this data in the next step?

vooon, 2015-04-30:
Mocap uses new message, so may be unsupported by fcu. So try vision first.

wangsen1312, 2015-05-07:
@vooon   Now I can send the VICON position to vision_pose_estimation topic. What I do next is to replace the local_position/local data with these data.  Can you give me some suggestion?
Thank you very much!

closed 2015-05-14 by wangsen1312

vinhk, 2016-07-15:
this I supposed ? 
/mavros/setpoint_velocity/cmd_vel