mavlink/mavros #278: Connection URL has issues when selecting different nets

vmayoral, 2015-04-20:
When doing:
```
root@erlebrain:/lib/systemd/system# rosrun mavros gcs_bridge _gcs_url:="udp://127.0.0.1:6666@11.0.0.2:7000"
Error:   mavconn: udp0: sendto: Invalid argument
         at line 233 in /root/indigo/src/libmavconn/src/udp.cpp
Error:   mavconn: udp0: send: channel closed!
         at line 149 in /root/indigo/src/libmavconn/src/udp.cpp
Error:   mavconn: udp0: send: channel closed!
         at line 149 in /root/indigo/src/libmavconn/src/udp.cpp
Error:   mavconn: udp0: send: channel closed!
         at line 149 in /root/indigo/src/libmavconn/src/udp.cpp
Error:   mavconn: udp0: send: channel closed!
         at line 149 in /root/indigo/src/libmavconn/src/udp.cpp
Error:   mavconn: udp0: send: channel closed!
         at line 149 in /root/indigo/src/libmavconn/src/udp.cpp
terminate called after throwing an instance of 'std::system_error'
Error:   mavconn: udp0: send: channel closed!
         at line 149 in /root/indigo/src/libmavconn/src/udp.cpp
Error:   mavconn: udp0: send: channel closed!
         at line 149 in /root/indigo/src/libmavconn/src/udp.cpp
  what():  Resource deadlock avoided
Aborted

```

vooon, 2015-04-20:
Not sure that there no error in code, but you definitly should use 0.0.0.0 or (`:port`) here.
Bind address should be: any, machine ip (any IF).

vmayoral, 2015-04-23:
`0.0.0.0` seems to do the job. The problem can be also be bypassed using IPs in the same subnet.

closed 2015-04-23 by vooon