mavlink/mavros #279: mavros, more than one `gcs_bridge`

vmayoral, 2015-04-20:
As it is right now, it seems like we can't have more than one `gcs_bridge` (naming issue):

```bash
rosrun mavros gcs_bridge _gcs_url:="udp://11.0.0.1:6666@11.0.0.2:7000" &
rosrun mavros gcs_bridge _gcs_url:="serial:///dev/ttyO4"

[ WARN] [1400129174.731807037]: Shutdown request received.
[ WARN] [1400129174.744814704]: Reason given for shutdown: [new node registered with same name]
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost::lock_error
Aborted

```

We can probably bypass this by [renaming the node](https://github.com/mavlink/mavros/blob/master/mavros/src/gcs_bridge.cpp#L50) and recompiling the code. @vooon can we modify gcs_bridge to rename the node according to an argument (maybe the first one?) @ahcorde

vooon, 2015-04-20:
Rename it in your launch script.
FYI: `<node name="blahblahblah" />` declares it's name, and override init name.

vooon, 2015-04-20:
Also i don't recommend using background shells (trailing `&`) with ros nodes.
Best way - use launch files.

vmayoral, 2015-04-20:
Thanks @vooon that did it. In addition to launch files, `__name:="whatever-name"` works as well.

closed 2015-04-20 by vmayoral