mavlink/mavros #281: Connexion via serial 4/5 , optical flow implementation , GPS fake ?

Royalcheese, 2015-04-23:
Hi all ! 
i'm a new user of px4 and mavros since one week. I'm trying to make it works via vicon in order to implement a new optical flow controller. But i've some basics issues which keep me from going deeper :(
i'm a bit confused with some points, if someone could help me <3

1/ i've installed mavros and px4 as described in tutorial, but when i'm running Roscore and $roslaunch mavros px4.launch fcu_url:=/dev/ttyUSB0:57600 , it just receive 0 packet and appears not to be connected. 
Same result with the default $roslaunch mavros px4.launch , i can recover some data like data_raw or mag, but the /diagnotic say that : 
 level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0

So do i have to assume that the bridge does not work properly ? my mavros appears to be the only publisher and subscriber on the stream (thanks to rostopic info tool ) , mavlink does work properly (i know it through cutecom device ) but i cannot makes them communicate together using ACM0 and serial 4/5 port (related to USB0 ). 
I maybe does not understood something..

2/ About Vicon and optical flow, i've seen that some things have been implemented. I can recover a local position with optitrack node on ROS (x,y,z,r,p,y) , and i want to publish and override data on mavros local_pos topic,   in order to make mavlink using my local_pos_ned for the autopilot.
I would use "ros topic tool relay" but i'm not sure if this is the best way, or if the implemented features are already doing the same thing. i may modify local_pos topic on mavros too, but i would like to know if other ways have already been implemented that i've missed.

3/ About the GPS , if i chose "GPS : no " and "GPS fake no ", will the autopilot start properly or does it need to work with GPS fake ?

here i am, i'm sorry to bother you with basics questions , if anyone could help , or redirecting me on appropriated topics, i would be extremely grateful =)

thanks in advances, have a good day all !

vooon, 2015-04-23:
1. it's a bug #269, see rx/tx rates (second part of link report) and heartbeat diag (connection status based on heartbeat).
2. If you override position from mavros you don't send it to FCU => FCU does not use your data. See extras plugins for that. Also it require some setup on FCU side, google it on px4 lists.
3. Also google px4 lists (or ask if there no answer).

Royalcheese, 2015-04-23:
Ok thank you i'm on it ! :)
For 2, in fact i was looking to fake the actual local_position of mavros in order to avoid to many change in the program , cause i was expecting that he was already sending local_pos to mavlink, which use it in case of GPS fake, but i maybe got wrong.
i keep giving feedback on this .

vooon, 2015-04-23:
2: FCU sends `LOCAL_POSITION_NED` -> MR local_position translate and publish message.

Royalcheese, 2015-04-24:
Ok i've worked about those problems and that's what i deduced about my problems : 
Thanks you @vooon , so I have to modify attitude topic on mavros based on Vicon position instead of local position, i got confused with what is published and what is subscribed.
1/ Mavros and Mavlink can reach topic one each other but dont know when they are doing it, so that i can recover mavlink data for instance and mavros dont know who is the publisher with rostopic info.
So it seems not to be a problem at all finally.

I keep working on following subjects but the drone is working well in config (GPS no , GPS_fake yes).
I'm trying to publish something from mavros to mavlink through an embedded application on px4. i'll keep give news :)

Royalcheese, 2015-04-24:
Thank you @vooon  :) Just a question, as ROSTOPIC INFO is not reliable, do the "setpoint_position/local" is a published topic from mavros to mavlink as i expect ? Because the "imu/data_raw" appears to be published by mavros through rostopic info :/

1/ Mavros and Mavlink can reach topic one each other but dont know when they are doing it, so that i can recover mavlink data for instance and mavros dont know who is the publisher with rostopic info.
So it seems not to be a problem at all finally.
2/ keep working on it.
3/ the drone seems to boot properly in "GPS no GPS_fake yes ". But I have to check .

vooon, 2015-04-24:
Read plugin documentation on [ros wiki](http://wiki.ros.org/mavros#Plugins).
Data flow is fixed: fcu sends -> publisher, fcu listen -> subscriber.

Setpoint position is desired position that fcu will try to achieve. That uses FCU internal position estimator.

Royalcheese, 2015-04-27:
Ok i was confused ^^ Thank you @vooon !

Royalcheese, 2015-04-29:
So if I want to set the Local position of the drone I have to speak with local_position topic, and if I want to set a point to reach I have to speak with setpoint_position ?
But setpoint doesn't seems to get data from anyone , and local pos doesn't seems to send data to anyone.

As you told me : FCU NED-->local_pos ENU
setpoint ENU--->FCU NED

But what if I would set my own local pos (from the data I get from Vicon) and set my own setpoint ?

"Vicon_local_pos"--->FCU--->local_position
"myPoint"--->setpoint--->FCU 
Cause I expect that mavlink msg are going are unidirectional , or I am mistaken ?

Royalcheese, 2015-04-30:
Ok in order to get the local mocap frame you need to remap x,y to vision_pose/pose

Royalcheese, 2015-04-30:
just link the question to px4 user forum just in case: https://groups.google.com/forum/#!topic/px4users/yEBu74WEpqo

Royalcheese, 2015-04-30:
fixed with correcting z_vision weight

closed 2015-05-27 by Royalcheese