mavlink/mavros #282: Mavros_extra plugin not found

Royalcheese, 2015-04-24:
Hi everyone ! 
i experiment some problem with mavros extra plugins, mavros don’t find the associated library of the mavros_extra plugins in order to make mocap vision working. Any idea why does it not find them ? 
Thanks in advanced,
best regards.

vooon, 2015-04-24:
Did you do `source ~/your_ws/devel/setup.bash` after build?

vooon, 2015-04-26:
Updates?

Royalcheese, 2015-04-27:
The source setup.bash was done, and Mavros/ px4 have been updated 2 hours before..

vooon, 2015-04-27:
What ROS version used? I seen similar bug on Hydro when updated to REP 140 package.xml.
I recommend to update at least to Indigo (Jade release soon, but now it lack some useful packages like rviz).

Royalcheese, 2015-04-27:
You're right , upgrading to Indigo makes this problem not happening . Thank you !

closed 2015-04-27 by Royalcheese