mavlink/mavros #283: Not able to arm using telem2

Diamg, 2015-04-27:
Hi,

I'm using a raspberry 2 as onboard computer and I'm able to read the topic either from the usb as well as from the uart(rpi) to telem2(px4). The problem comes when sending an arming comand, it works with the us but not with the uart. 
The response are: 
$rosservice call /mavros/cmd/arming "value: true"
success: False
result: 4
$ rosrun mavros mavsafety arm
Request failed. Check mavros logs

On the px4.launch terminal there is no info message when it happens.

Moreover I put here the warnings I obtain on the mavros_node even though I think they don't have nothing to do.
[ WARN] [1430122958.486594299]: RAW_IMU: linear acceleration known on APM only
[ WARN] [1430122958.508094584]: TF to MSG: Quaternion Not Properly Normalized

[ WARN] [1430122972.703239245]: PR: request list timeout, retries left 2
[ WARN] [1430122972.740587117]: VER: request timeout, retries left 1
[ WARN] [1430122973.705465197]: PR: request list timeout, retries left 1
[ WARN] [1430122974.707253341]: PR: request list timeout, retries left 0
[ WARN] [1430122977.764099451]: VER: request timeout, retries left 0
[ WARN] [1430122982.789108852]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities

The firmware I'm using is  ArduCopter V3.2.1

Thanks in advance

vooon, 2015-04-27:
Please:

a. Update to 0.11.2 (yesterday release)
b. what launch script you used?
c. `rostopic echo -n1 /diagnostics`

Note: use triple ` for code blocks (read [github formatting page](https://guides.github.com/features/mastering-markdown/)).

Diamg, 2015-04-28:
Hi,
a. mavros is updated.
b. I'm using:
```roslaunch mavros px4.launch```
c. 
```rostopic echo -n1 /diagnostics```
```
header: 
  seq: 284
  stamp: 
    secs: 1430218701
    nsecs: 929121105
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyAMA0:115200
    values: 
      - 
        key: Received packets:
        value: 708
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 62
      - 
        key: Tx sequence number:
        value: 58
      - 
        key: Rx total bytes:
        value: 19296
      - 
        key: Tx total bytes:
        value: 73619
      - 
        key: Rx speed:
        value: 1059.000000
      - 
        key: Tx speed:
        value: 260.000000
  - 
    level: 0
    name: mavros: GPS
    message: 3D fix
    hardware_id: /dev/ttyAMA0:115200
    values: 
      - 
        key: Satellites visible
        value: 8
      - 
        key: Fix type
        value: 3
      - 
        key: EPH (m)
        value: 1.97
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: /dev/ttyAMA0:115200
    values: 
      - 
        key: Heartbeats since startup
        value: 20
      - 
        key: Frequency (Hz)
        value: 0.999110
      - 
        key: Vehicle type
        value: Octorotor
      - 
        key: Autopilot type
        value: ArduPilotMega
      - 
        key: Mode
        value: ALT_HOLD
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyAMA0:115200
    values: 
      - 
        key: Sensor present
        value: 0x0060FC2F
      - 
        key: Sensor enabled
        value: 0x0060FC2F
      - 
        key: Sensor helth
        value: 0x0060FC2F
      - 
        key: Sensor 3D Gyro
        value: Ok
      - 
        key: Sensor 3D Accel
        value: Ok
      - 
        key: Sensor 3D Mag
        value: Ok
      - 
        key: Sensor Abs Pressure
        value: Ok
      - 
        key: Sensor GPS
        value: Ok
      - 
        key: Sensor Ang Rate Control
        value: Ok
      - 
        key: Sensor Attitude Stab
        value: Ok
      - 
        key: Sensor Yaw Position
        value: Ok
      - 
        key: Sensor Z/Alt Control
        value: Ok
      - 
        key: Sensor X/Y Pos Control
        value: Ok
      - 
        key: Sensor Motor Outputs
        value: Ok
      - 
        key: AHRS health
        value: Ok
      - 
        key: Terrain health
        value: Ok
      - 
        key: CPU Load (%)
        value: 55.9
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: /dev/ttyAMA0:115200
    values: 
      - 
        key: Voltage
        value: 16.54
      - 
        key: Current
        value: 0.4
      - 
        key: Remaining
        value: 99.0
  - 
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: /dev/ttyAMA0:115200
    values: 
      - 
        key: Timesyncs since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000
---
```

vooon, 2015-04-28:
b. You should use `apm.launch`!
c. diagnostics looks good. just use right launch.

Diamg, 2015-04-29:
Hi,
I had tried the other launchers getting the same result. Any other idea? 
Which is de difference between the launchers?  They are used depending on the firmware and not the hardware I guess.
Thanks

vooon, 2015-04-29:
Demending on firmware, yes. But there different default target IDs and plugin configuration (and list).

```
      - 
        key: Autopilot type
        value: ArduPilotMega
```
-> apm.launch. px4 here at least have wrong ID's and ton of unused plugins. Also there unusable (PX4 only) timesync that eats telemetry bandwidth.

Also there at least should be several FCU messages at startup. You may find log in `~/.ros/log/<dir>`.
But i recommend do `rosclean purge` for easy finding last run.

Diamg, 2015-04-30:
I finally changed the firmware to px4 but still not able to use the telem 2. With px4 it doesn't get the heartbeat. However I'm working with the USB fine by the moment. 
Getting out of the topic, could you recommend me how to start working with offboard mode and local point topics? There is any example? 
As I understood the list of things I should do first using ROS are:
a. arm: done
b. change to offboard  mode: that is rejected
c. take off: which is also rejected
d. publish on the ```/mavros/setpoint_position/local``` topic

Thanks for all your help

vooon, 2015-04-30:
By default px4 dont start mavlink app at telem2. See documentation about `SYS_COMPANION` parameter (check name).
Next: takeoff may not work, i saw comments about commanding sp with increasing z.

a. arm
b. start sending setpoints
c. request  offboard

Diamg, 2015-05-04:
I've read it and I have the ```SYS_COMPANION``` parameter to 921600 but still not getting the heartbeat.
On the other hand, using the USB I'm aible to change to ```OFFBOARD``` by using ```rosrun mavros mavsys mode -c OFFBOARD``` if no rc signal is recived for first time. Otherwise it change the nav state but the main state remains on ```MANUAL``` so it change again the nav state to ```MANUAL```.
```
[ INFO] [1430728902.175722349]: FCU: [cmd] nav state: OFFBOARD
[ INFO] [1430728902.227245155]: FCU: [cmd] main state: MANUAL
[ INFO] [1430728902.278751814]: FCU: [cmd] nav state: MANUAL
```
However I can change the mode to ```OFFBOARD``` using ```rosrun mavros mavcmd guided``` but then the rc have no control on the pixhawk and there is no way of changing the mode again either with any ROS executable or service. 
Is that normal?

vooon, 2015-05-04:
Not normal, but this questions better ask at [px4users](https://groups.google.com/forum/#!forum/px4users).

Ensure that data comes to your uart adapter. Use `minicom -D <device> -b 921600`.
Then check your url.

Guided command is first implemented method, now better use `mavsys mode`.
Other possible way - configure OFFBOARD switch on your RC.

Diamg, 2015-05-04:
Ok thanks, I will ask there. 
The telem 2 should be a px4 firmware issue also because I'm not able to connect via screen or minicom.
Thanks

vooon, 2015-06-03:
Updates?

Diamg, 2015-06-08:
I finally change to px4 flight stack, now Im not able to detect the pixhawk using uart to uart connection. At the moment I'm working fine with a connection USB-USB. At some point I will need another solution in that case I'll publish the news.

LorenzMeier, 2015-06-08:
https://pixhawk.org/dev/companion_link

closed 2015-06-28 by vooon