mavlink/mavros #284: What is mavros/setpoint_mixin.h ?

Royalcheese, 2015-04-28:
Hello !
I'm not sure about what is exactly this feature. It does implements the function set_position_target_local_ned(), but how dit it work with setpoint_position.cpp? 
From now I understand that setpoint_position set the change of frame and setpoint_mixin is sending the desired position through the function, but are they ?
For context, I would write a similar plugin to catch Vicon data ( I already catch the topic) , reworking data and sending it to Mavlink. I think setpoint architecture could help me in that point
Thanks in advance,
Have a good day !

vooon, 2015-04-28:
It's just reduces copy-paste because several plugins uses same message.
Definition on [L31][L31]. It gets UAS from plugin and send to fcu link encoded message.

[L31]: https://github.com/mavlink/mavros/blob/master/mavros/include/mavros/setpoint_mixin.h#L31

Royalcheese, 2015-04-29:
Ok, thank you !

closed 2015-04-29 by Royalcheese