mavlink/mavros #285: SetpointAttitude plugin can not control throttle and attitude together

dvorak0, 2015-04-30:
I am using setpoint_attitude to control my quadrotor through mavros and pixhawk.

First, I tried to control attitude and throttle at the same time by publishing geometry_msgs::PoseStamped and std_msgs::Float64. This results that the speed of rotors jumps between normal speed and arming speed. It seems the throttle is set to 0 sometimes.

I fixed the problem by using geometry_msgs::PoseStamped.pose.position.z for throttle and changed the mask (ignore_all_except_q) from (1 << 6) | (7 << 0) to (7 << 0). It works well finally.

Could anyone give me some explanations? Or it is a bug possibly.

vooon, 2015-04-30:
@mhkabir any ideas?

dvorak0, 2015-05-01:
@vooon I got some ideas. The reason is removing 'static' incorrectly in

SHA: 1d283bf3c15b79ad2aa23ed3e2de7ff80f0261fe.

It disable the ability of setpoint_attititude to control thrust and attitude separately.

I've submit a pull request. https://github.com/PX4/Firmware/pull/2112.

closed 2015-05-04 by dvorak0