mavlink/mavros #287: Example raw mavlink message

joshvillbrandt, 2015-05-03:
Hey there. Could some provide me with an example of sending a raw mavlink message? Below is a simple send function that I've started. My specific questions at the moment are:

* How do I "pack" the payload64 attribute for various variable types? For the life of me, I can't figure out where functions like `mavlink_msg_set_actuator_control_target_pack_chan` are coming from in the [mavros plugins](https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/actuator_control.cpp).
* What is `fromlcm` and what should I be setting that to?
* Is my value from `compid` correct? Should I be grabbing this from somewhere else instead of explicitly stating it?

```cpp
void Pilot::setServo(int index, float value) {
  // thruster values should be between 1100 and 1900 microseconds (us)
  // values less than 1500 us are backwards; values more than are forwards
  int pulse_width = (value + 1) * 400 + 1100;

  // create raw mavlink message
  mavros::Mavlink mavlink_msg;
  mavlink_msg.header.stamp = ros::Time::now();
  mavlink_msg.len = 2;
  mavlink_msg.seq = 0; // i should technically increment this each time
  mavlink_msg.sysid = 1;
  mavlink_msg.compid = 250;
  mavlink_msg.msgid = 183; // https://pixhawk.ethz.ch/mavlink/
  mavlink_msg.fromlcm = true; // not sure what this is
  mavlink_msg.payload64 = [index, pulse_width];

  // publish message
  mavlink_pub.publish(mavlink_msg);
}
```

Eventually it would be nice for this type of example to live in the documentation. Maybe in a ROS wiki tutorials section like many ros packages have.

Thank you!! :)

joshvillbrandt, 2015-05-03:
Hmm... Well it seems I should be using the [command service](http://docs.ros.org/api/mavros/html/srv/CommandLong.html) for starters... Looking into that now.

vooon, 2015-05-03:
* `mavlink_msg_xyz` comes from mavlink c-library ([generated sources](https://github.com/mavlink/c_library)).
* `fromlcm` - unused flag leaved for compatible with `mavlink_ros`
* Your example is wrong. You should encode your message to `mavlink_message_t`, then copy it to Mavlink.msg using `mavros::copy_mavlink_to_ros()`.

Also i not recommend to use to/from topics for controlling. Better find or write plugin which does what you want.

Note: perhaps there a bug in [utils.37](https://github.com/mavlink/mavros/blob/master/mavros/include/mavros/utils.h#L37) - should i copy seq and ids? Need to check.

joshvillbrandt, 2015-05-03:
Hey @vooon, thanks for the reply!

I went ahead and switched to the using the command plugin. Here is my latest code:

```cpp
void Pilot::setServo(int index, float value) {
  // thruster values should be between 1100 and 1900 microseconds (us)
  // values less than 1500 us are backwards; values more than are forwards
  int pulse_width = (value + 1) * 400 + 1100;

  // send mavros command message
  // http://docs.ros.org/api/mavros/html/srv/CommandLong.html
  // CMD_DO_SET_SERVO: https://pixhawk.ethz.ch/mavlink/
  ROS_INFO("setServo(%d,%f->%d)", index, value, pulse_width);
  mavros::CommandLong srv;
  srv.request.command = mavros::CommandLongRequest::CMD_DO_SET_SERVO;
  srv.request.param1 = index + 1; // servos are 1-indexed here
  srv.request.param2 = pulse_width;
  bool result = command_client.call(srv);
}
```

Commands are now being sent and I can spin motors. However, the commanding is very unreliable. The biggest issue seems to be that the service calls sometimes seem to hang. I'll often issue a set of commands from my remote control, and then see the ROS_INFO logs for one of the motors immediately, but then the rest all go in a big batch a few seconds later. Sometimes they all go though. This seems to be more on the hardware side and now less to do with mavros, but do you have any suggestions? I'm using an APM 2.6 with APMrover2 firmware by the way.

Thanks for your help!

joshvillbrandt, 2015-05-03:
Also, the number of `FCU: command received: ` messages never seems to equal the number of SET_SERVO commands my program is actually trying to send. Any clue about this? I am definitely sending the commands in rapid succession. Maybe the FCU keeps a very tiny buffer?

joshvillbrandt, 2015-05-03:
Alright, well, after further review, it appears that I managed to have to active serial connections to the APM. The number of FCU messages still doesn't match the number of commands I'm actually sending (a buffer issue on the APM side?), but otherwise everything seems to be working well.

Thanks for your help, @vooon!

closed 2015-05-03 by joshvillbrandt

vooon, 2015-05-03:
Note that service not best way for continuously stream values. Topic are better in that case.
Also note that for each COMMAND_LONG plugin waits COMMAND_ACK (because APM always send them and `confirmation` does not matter) also it generates lots of "command received:" messages, which floods telemetry link if you send commands too often.

About "command received:" - i think that in past it prints command number, but not now.

For what you used set_servo? Perhaps you also may use `rc/override` (but be careful).

joshvillbrandt, 2015-05-03:
Ah, I think [rc/override](http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.rc_io) is exactly what I need! The documentation notes that this is for APM only though. I'd like my code base to work with both the APM and PX4 hardware. What is the best approach to directly set the RC out values for both platforms?

My application, by the way, is for an underwater vehicle (see the [BlueROV](https://github.com/bluerobotics/BlueROV) repo.) Since the existing APM and PX4 packages aren't well suited for underwater vehicles, I'm performing most of the high-level control on a RaspberryPi and attempting to directly override the output PWM values.

vooon, 2015-05-03:
For APM - `rc/override` (raw PWM values),
PX4 - actuator_control, but it run in OFFBOARD mode and normalized (pass trough mixer).