mavlink/mavros #288: frame of setpoint

MarcGyongyosi, 2015-05-03:
Hi, quick question - just to make sure everything will be working as expected during the next tests tomorrow: when I send a setpoint as PoseStamped on the mavros/setpoint_position/local topic does it take the frame into consideration? I.e. if I send a point x,y,z,yaw in the fcu frame, that is relative to the fcu itself, right? So If I were to send a PoseStamped in a different frame - e.g. local_origin, will it take that into consideration and transform properly?

Thanks!

Marc

vooon, 2015-05-03:
I think, its local_origin, not body frame.
Not sure does tf translate or not...

Royalcheese, 2015-05-06:
Are you overwriting the topic with your own talker ? Because i've done it, but i'm not sure if it just erase setpoint_pos topic or if it write data on it to make setpoint_pos plugin working. In that case you will have tf translating I suppose.

I'm interested about the results of your test, concerning what frame you have to send in PoseStamped if it do not bother you :)

MarcGyongyosi, 2015-05-19:
I will test it as soon as I have some for test data from my localization framework in!

Royalcheese, 2015-05-26:
I have z_fcu =  -z_sent , setting Posestamped.header.frame.id= "fcu", =1 ...
I must have misunderstood something...

vooon, 2015-05-26:
@Royalcheese Frame conversion #49.

Royalcheese, 2015-05-27:
Great ! thank you :)

vooon, 2015-06-03:
I thin i can close this.

closed 2015-06-03 by vooon