mavlink/mavros #289: Can VICON data replace local_postion data?

wangsen1312, 2015-05-04:
Now I have get the position from VICON, I can send the attitude and position to TOPIC vision_pose_estimate. Because I don't have GPS now, So can I transfer VICON data to local_position instead so that I can navigation indoor.

Royalcheese, 2015-05-05:
I don't see the question :p

wangsen1312, 2015-05-05:
@Royalcheese  Thank you for your attention. 
Now what I want to is using VICON data instead the local_position calculated by FCU. 
Can you give me some suggestion?

Royalcheese, 2015-05-05:
I've experimented same problem last week ^^ In fact the purpose is done, but you have to remap data in node.launch file.
Something like that : <remap from="/mavros/vision_pose/pose" to="/mavros/mocap/pose">
you have more explanation on remap here : http://wiki.ros.org/roslaunch/XML/remap
and some remap example here : https://pixhawk.org/dev/ros/visual_estimation
I Hope it will help :)
have a good day!

wangsen1312, 2015-05-05:
@Royalcheese Thank you very much! I will try it ASAP.

Royalcheese, 2015-05-07:
Take care of the weight of "vision sensor" in parameter of the fcu : you can set it via QGC . It must be very higher than the other parameter of position (GPS..) because it work like a vision sensor ( as i understood) and the weight is the credit accorded to the sensor. If you want to be full with vision sensor you should increase this weight high.
Take care at CBRK_no_vision = 0 too 
have a good day !

wangsen1312, 2015-05-07:
@Royalcheese   Thank you for your reminder!  There is still something wrong that why I still can't see the local_position.   I get the vicon data and publish it to vision_pose_estimate topic and I add the remap in launch file.  
Do you have the same setup as me?  Can you share what you do? Thank you very much!

wangsen1312, 2015-05-08:
@Royalcheese  Now I can get the local_position NOW. But I still cannot change the local_position data with the VICON when I use the remap?
Did your works well? Thank you very much!

Royalcheese, 2015-05-11:
Mine works, have you decreased the weight of other sensors in QGC ?

Royalcheese, 2015-05-11:
https://pixhawk.org/dev/ros/visual_estimation?s[]=odroid

wangsen1312, 2015-05-11:
@Royalcheese   In QGC, How can I find the weight of sensors parameters? I only find the CBRK_no_vision and change it to 0.
Thank you very much~

wangsen1312, 2015-05-12:
@Royalcheese  Thank you very much! I can make the local_position workable. Does this workable when use setposition/local topic?

vooon, 2015-05-12:
I think yes, as i understood commander uses same local position uORB that you can see in mavros topic.

Royalcheese, 2015-05-12:
I'm on it ;) like vooon I think it does, but offboard control behaviour is somehow.. strange for now ^^

wangsen1312, 2015-05-12:
@Royalcheese Thank you very much~ I will try it ASAP. If I can get the stable result, I will tell you soon.

Royalcheese, 2015-05-13:
Thanks you ! right now i've connected ground PC to ODROID to FCU. i will test vicon data and offboard soon

congleetea, 2015-05-22:
@Royalcheese Hi, i also send the mocap data to local position(use vision_pose no speed via mavros ) , but the z value drifts, how is your result?maybe you can give me some advice. i set all weight parameter to zero except for vision weight. and the flyer also can not track the setpoint position . thanks a lot in advanced.

Royalcheese, 2015-05-26:
Hi :)
please tell me :
I have : INAV_w_z_vis_p = 8 , INAV_w_z_vis_p = 7 ,CBRV_no_vision = 0 but I guess you've this too.
please tell me :
Where do you remap your mocap data ?
where do you remap your setpoint pos stream ?
Are x,y setting well ?

jgoppert, 2015-05-26:
Try using this instead of inav, mocap is directly supported. https://github.com/PX4/Firmware/pull/2096

wangsen1312, 2015-05-26:
@jgoppert  Thank you very much~ How Can I update my firmware and use it directly? Thank you very much~

jgoppert, 2015-05-26:
```bash
git clone https://github.com/PX4/Firmware.git
cd Firmware
git remote add jgoppert https://github.com/jgoppert/Firmware.git
git fetch jgoppert
git checkout -t jgoppert/cart_pos_estimator
git submodules init
git submodules update
J=4 make archives
make -j4 px4fmu-v2_default upload
```

congleetea, 2015-05-28:
@Royalcheese I set the vision weight max value 10, others are zero. I remap mocap data to /mavros/vision_pose/pose in launch file . the setpoints are directly publish on /mavros/setpoint_position_local/local. x y are well set. Now I can make the uav fly to setpoint , but I can not get vision_speed.  I just differential the pose simply, but the FCU have no any info on vision_speed.

wangsen1312, 2015-05-28:
@congleetea    Thank you for share your progress. You can use remap and setpoint well, so can you offer me the demo launch file and setpoint file. I may have something wrong. Thank you in advance

congleetea, 2015-05-28:
I use optitrack. just write a setpoint node , it is very simply . you can learn from ros tutorials.

Royalcheese, 2015-05-28:
Hum I need more time to master it, but you have to have a look on mc_pos_control module and INAV module (if you're not using jgoppert estimator, but it seems to be better for mocap) 
you have several PID controller, their parameters are the parameters you set on QGC. 
With a XY_vision_V set to 0, I expect the gain to be 0 and so you see nothing. But if this parameter too high , you will have extreme variation of estimated speed, and excessive thrust variation ( +40, -40, no saturation ).
by the way, what kind of precaution have you used for Offboard test ?

Royalcheese, 2015-05-28:
@congleetea  I have several question : 
What is the header.frame you're sending in setpoint pos ?
What header.fram you're sending in mocap, and vision_pos ?
All INAV parameter are set to 0 except vis_p_z and vis_p_xy and it works well ?
What are your remap 
by the way, what kind of precaution have you used for Offboard test ?
[image]
from now I got this with 10 vis_z and intial other value for parameter
I can get this :
[image]
but remap from local_position/local to vision_pose, and mocap to vision pose, but this is a mistake...
I can't test indoor without good simulation

Royalcheese, 2015-05-28:
I got it ! Matter of ros::time

Royalcheese, 2015-05-28:
[image]
with your config i got this..

What precaution have you taken for testing offboard indoor ?

Royalcheese, 2015-05-29:
This is because vel_z_derivator is default set to 0. And this is why we have no phase shift.
Any idea why @vooon ?

Royalcheese, 2015-06-01:
@congleetea , have you change parameter on MPC ?
Could you send me your graph of trust and z_pos when you are rea

TSC21, 2015-06-15:
@wangsen1312 are you able to test https://github.com/PX4/Firmware/pull/2361?

closed 2015-07-09 by vooon