mavlink/mavros #290: I can't run this

jzy0820, 2015-05-06:
I can't run this
rosdep install --from-paths src --ignore-src --rosdistro hydro -y

vooon, 2015-05-06:
Please tell detailed what you want to do? What error it prints?

vooon, 2015-05-08:
Updates?

jzy0820, 2015-05-15:
I'm so sorry too late to reply.
I want to link the MAVlink in ROS. The original words is as following. It's one step of all.
MAVLink

MAVLink is the communication API used by MAVCONN. It is a header-only library and thus does not need to be compiled. Just download it via GIT (you can also update it this way with git pull)

git clone git://github.com/mavlink/mavlink.git
cd mavlink
cd ..
MAVCONN

Download MAVCONN via GIT

git clone git://github.com/pixhawk/mavconn.git
Configure and compile it:

cd mavconn
mkdir build
cd build
cmake -D CMAKE_BUILD_TYPE=Release ..
make -j4
sudo make install
I run the commends above.When I run "cmake -D CMAKE_BUILD_TYPE=Release .."
The error is:
root@ubuntu:~/mav/mavconn/build# cmake -D CMAKE_BUILD_TYPE=Release ..
CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
  Could NOT find MAVLink protocol (mavlink.h) (missing: MAVLINK_INCLUDE_DIR
  MAVLINK_TYPES_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:315 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindMAVLINK.cmake:19 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/LibraryConfig.cmake:47 (FIND_PACKAGE)
  CMakeLists.txt:22 (INCLUDE)

-- Configuring incomplete, errors occurred!
See also "/root/mav/mavconn/build/CMakeFiles/CMakeOutput.log".
How can i solve this problem.
Thank you very much!

vooon, 2015-05-15:
Please don't open new issue (#295), when already opened this.

Please specify:
1. OS distibution (name and version)
2. Machine architecture (e.g. x86/amd64/armhf)
3. ROS distribution version

vooon, 2015-05-15:
1. Issue header non informative, next time provide a short description of your problem.
2. mavconn is totally unrelated to this project. **lib**mavconn is different. Did you read [installation instructions][1]?

[1]: https://github.com/mavlink/mavros/tree/master/mavros#installation

jzy0820, 2015-05-15:
I'm so sorry.I'm a newer and not good at speak English.
I just want to link a uav with pixhawk in the ros in Ubuntu.
Do you have a method to help me solve it?
Thank you

jzy0820, 2015-05-15:
I am trying to link by a web (https://pixhawk.ethz.ch/installation/ubuntu#get_your_ubuntu_up_and_running)

vooon, 2015-05-15:
> I'm not good at speak English.
I'm too :)

I hope that you used latest LTS 14.04? Because 11.10 is too old (4 years).
14.04 LTS is best option for you, because there available ROS Indigo and Jade soon, while on 15.04 only Jade (coming soon).

Next, this howto is little outdated, also you only need ROS for MAVROS.
Other listed software is offtopic here, but i think that minimal tool set is:
1. QGC or APM Planner 2 (depending on your FCU)
2. ROS RQT tools

About ROS installation you can read here: http://wiki.ros.org/indigo/Installation/Ubuntu

vooon, 2015-05-15:
Also some useful information on pixhawk.org:
- https://pixhawk.org/dev/ros/getting_started_tutorial?s[]=mavros
- https://pixhawk.org/dev/ros/mavros_offboard
- https://pixhawk.org/dev/software_architecture?s[]=mavros

jzy0820, 2015-05-18:
Is this useful "https://pixhawk.org/peripherals/onboard_computers/odroid_u3"?
My flight control is Pixhawk like this "https://pixhawk.org/modules/pixhawk".

vooon, 2015-05-18:
Yes if you used Odroid-U3.
And it is useful, but you shall keep in mind that this manual is for ARM Cortex (armhf) based computer.
Usually linux on any platform is pretty same, and only few differences like repo url, port names etc.

vooon, 2015-06-03:
Updates?

closed 2015-06-28 by vooon