mavlink/mavros #291: local_position/local can't not work?

wangsen1312, 2015-05-07:
when I type 
rostopic echo /mavros/local_position/local 
It shows:
WARNING: topic [/mavros/local_position/local] does not appear to be published yet

I can echo every published topic except local? Is that something wrong?

vooon, 2015-05-07:
Does FCU send [LOCAL_POSITION_NED](https://pixhawk.ethz.ch/mavlink/#LOCAL_POSITION_NED)?

wangsen1312, 2015-05-07:
Thank you very much! @vooon  How can I find this?

TSC21, 2015-05-07:
@wangsen1312 have you checked the firmware startup script?

wangsen1312, 2015-05-07:
@TSC21   Thank you for your attention. What do you mean about firmware startup script? Can you give me more hints

TSC21, 2015-05-07:
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.usb

TSC21, 2015-05-07:
If you are using APM, you probably need to activate it in Mission Planner. But that I'm not sure, cause I work with PX4, not with Arducopter.

wangsen1312, 2015-05-07:
@TSC21  How can I activate it? I will try this. Thank you for your sharing.

mhkabir, 2015-05-08:
Short story : It won't work unless you implement it in the APM codebase.
All the estimators and controllers work in the global frame for APM.

wangsen1312, 2015-05-08:
@mhkabir    Thank you very much!!  I am a new user for APM. So can you give me an example that when this example works I can get the local position? Thank you again!

vooon, 2015-05-08:
Probably easier switch to PX4, rather than implementation in APM...

TSC21, 2015-05-08:
@wangsen1312, @vooon is right! Probably it's better for you to switch to PX4 firmware than use the APM and have to implement this feature from scratch. Ang given you are a "new user", probably it will take much more time for you to get this done.
Anyway, if you still want to implement this on APM, you should check the APM repo on Github and open an issue on it, as this is off-topic and not related to mavros.

wangsen1312, 2015-05-08:
@vooon @TSC21 @mhkabir   Thank you for your sharing ideas!  I don't know because I use PIXHAWK with PX4 and my plane is quadcopter. So I don't know is PX4 for APM? Do I need to flash a firmware?

mhkabir, 2015-05-08:
Please see :

www.pixhawk.org/choice

Pixhawk is the hardware. APM and PX4 are two different software stacks for
it.

wangsen1312, 2015-05-08:
@mhkabir  Maybe I have tried it before. If I use Qgroundcontrol to flash it, it will be PX4, If I use Mission Planner, it will be APM? Is that all right?
PS: Can PX4 stack control my quadcopter? If it can, I will flash it as soon as possible!

mhkabir, 2015-05-08:
Yes. Please read the PX4 wiki and mailing list for more details.

Closing.

closed 2015-05-08 by mhkabir