mavlink/mavros #294: APM SITL - mavros don't publish any topic for ArduCopter

ramonmelo, 2015-05-12:
Hello everyone,

I'm trying to make a multi-robot system for a group of aerial and land robots using the APM autopilot system and ROS. Before any tests in real robots, I was using the simulation to test the algorithms.

I followed the tutorial on this [link](http://dev.ardupilot.com/wiki/simulation-2/sitl-simulator-software-in-the-loop/setting-up-sitl-on-linux/), and worked very well. Using the APM Planner 2 I can do all commands.

When the simulations start, it enable to communication via *tcp://localhost:5762*, so I just use this address to connect via mavros on the mavlink system to send commands.

The problem is, I can communicate with the **APMrover2** and **ArduPlane** firmwares, receive all the topics (battery, gps, state, ...) and can send commands via services, but with the **ArduCopter** only some topics are send (state, waypoints), but, whats is strange, is that some services also work, like *set_mode*.

I cloned all the code, the [ardupilot](https://github.com/diydrones/ardupilot) using the master branch, which is in the version *3.3-dev* for the *ArduCopter*, and the mavros, also the master.

Any ideas ?

(sorry with I doing something wrong, I tried to look for other similar questions before send it.)

molinsp, 2015-05-15:
Hi, 

I'm working on a similar idea. I was able to run simulations with multiple instances of ArduCopter connected to several mavros nodes.
Maybe your issue is also related to the fact that you connect using a tcp port. I had issues with this and switched to using udp.
http://answers.ros.org/question/208265/mavros-multi-vehicle-simulation/

Maybe we can work together on this if you are interested.

ramonmelo, 2015-05-16:
Hello,

Yes, that works, using the UDP connection. Thanks! 

And, yes, we can do some work. For now I making a simple module that will be over the mavros system, as a new layer of control, it's still in development, but will aggregate the important data from the UAV (gps, rel_alt, status...) in one place. Besides create a communication interface between the drones.

closed 2015-05-16 by ramonmelo