mavlink/mavros #295: I have a big problem! help

jzy0820, 2015-05-15:
I want to link the MAVlink in ROS. The original words is as following. It's one step of all.
MAVLink

MAVLink is the communication API used by MAVCONN. It is a header-only library and thus does not need to be compiled. Just download it via GIT (you can also update it this way with git pull)

git clone git://github.com/mavlink/mavlink.git
cd mavlink
cd ..
MAVCONN

Download MAVCONN via GIT

git clone git://github.com/pixhawk/mavconn.git
Configure and compile it:

cd mavconn
mkdir build
cd build
cmake -D CMAKE_BUILD_TYPE=Release ..
make -j4
sudo make install
I run the commends above.When I run "cmake -D CMAKE_BUILD_TYPE=Release .."
The error is:
root@ubuntu:~/mav/mavconn/build# cmake -D CMAKE_BUILD_TYPE=Release ..
CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
Could NOT find MAVLink protocol (mavlink.h) (missing: MAVLINK_INCLUDE_DIR
MAVLINK_TYPES_INCLUDE_DIR)
Call Stack (most recent call first):
/usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:315 (_FPHSA_FAILURE_MESSAGE)
cmake/FindMAVLINK.cmake:19 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
cmake/LibraryConfig.cmake:47 (FIND_PACKAGE)
CMakeLists.txt:22 (INCLUDE)

-- Configuring incomplete, errors occurred!
See also "/root/mav/mavconn/build/CMakeFiles/CMakeOutput.log".
How can i solve this problem.
Thank you very much!

vooon, 2015-05-15:
1. Issue header non informative, next time provide a short description of your problem.
2. mavconn is totally unrelated to this project. **lib**mavconn is different. Did you read [installation instructions][1]?

[1]: https://github.com/mavlink/mavros/tree/master/mavros#installation

vooon, 2015-05-15:
Related to #290.

closed 2015-05-15 by vooon