mavlink/mavros #297: Connection problem between Ground Control Station APM Planner and mavros using UDP link.

MauroPagliarella, 2015-05-18:
Hi all
I'm building a robot running ROS on board, and I would like to receive robot telemetries and send telecommands to the robot using the 
APM ground control station v2.0.17 using the mavros UDP link. I performed the test described here below but it didn't work.
Any help is appreciated. Thank you very much.
Mauro

SYSTEM CONFIGURATION:
1: Laptop running Ubuntu 14.04 & ROS Indigo with the following packages installed: 
	1262779 drwxr-xr-x   3 root  4096 mag 12 23:19 mavlink		
	1270356 drwxr-xr-x   6 root  4096 mag 12 23:19 mavros		
	1270615 drwxr-xr-x   5 root  4096 mag 12 23:19 mavros_extras	

2: The APM Planner v2.0.17 is installed and runs on the same laptop.

3: Arduino 1R3 with dummy firmware connectd to one of the laptop usb port.(Dummy only to have a connection on the serial port to allow ROS node to start). 

4: ROS Launch file (prova.launch) reported here below:

	<launch>
		
		

		<arg name="fcu_url" default="/dev/ttyACM0:57600" />
		<arg name="gcs_url" default="udp://@localhost:14550" />
		<arg name="tgt_system" default="1" />
		<arg name="tgt_component" default="50" />

		<include file="$(find mavros)/launch/node.launch">
			<arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
			<arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

			<arg name="fcu_url" value="$(arg fcu_url)" />
			<arg name="gcs_url" value="$(arg gcs_url)" />
			<arg name="tgt_system" value="$(arg tgt_system)" />
			<arg name="tgt_component" value="$(arg tgt_component)" />
		</include>
	</launch>

OBJECTIVE:
Connect the APM ground Control Station v2.0.17 to mavros UDP port 14550 to send commands to ROS and receive messages from ROS

PERFORMED OPERATIONS:
1: Open terminal session and start mavros on port 14550 using shell command: roslaunch mavros prova.launch 
2: start QGC: APM Planner v2.0.17 using shell command: ./release/apmplanner2

OPERATION 1 RESULTS:
mauro@mauro-HP-Pavilion-dv7-Notebook-PC:
mauro@mauro-HP-Pavilion-dv7-Notebook-PC:~$ roslaunch mavros prova.launch 
... logging to /home/mauro/.ros/log/8b90949a-fce5-11e4-80e1-6159cf7aef84/roslaunch-mauro-HP-Pavilion-dv7-Notebook-PC-4287.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mauro-HP-Pavilion-dv7-Notebook-PC:42874/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat: 5.0
 * /mavros/conn/system_time: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync: 0.1
 * /mavros/fcu_url: /dev/ttyACM0:57600
 * /mavros/gcs_url: udp://@localhost:...
 * /mavros/global_position/child_frame_id: fcu
[... 175 lines cut ...]
DEBUG 2015-05-18T00:41:48.074 QuickView: load settings END 
 INFO 2015-05-18T00:41:48.159 Using Alsa Audio driver 
 WARN 2015-05-18T00:41:48.159 Dir directory tmp_audio exists 
DEBUG 2015-05-18T00:41:48.170 qmlBaseDir "/home/mauro/Qt/MyProjects/apm_planner" 
DEBUG 2015-05-18T00:41:48.170  QUrl( "file:///home/mauro/Qt/MyProjects/apm_planner/qml/ApmToolBar.qml" )  
APMToolBar:  SELECTED
DEBUG 2015-05-18T00:41:48.210 QML Status: 1 
DEBUG 2015-05-18T00:41:48.219 Loading widget: "PRIMARY_FLIGHT_DISPLAY_QML_DOCKWIDGET" 
DEBUG 2015-05-18T00:41:48.220 Loading widget: "UAS_INFO_INFOVIEW_DOCKWIDGET" 
 INFO 2015-05-18T00:41:48.492 UDP Created  "UDP Link (port:14550)" 
 INFO 2015-05-18T00:41:48.494 UDP disconnect 
 INFO 2015-05-18T00:41:48.496 UDPLink::UDP connect   QHostAddress( "0.0.0.0" )  : 14550 
ERROR 2015-05-18T00:41:48.496 bind failed!   QHostAddress( "0.0.0.0" )  : 14550 
DEBUG 2015-05-18T00:41:48.498 Create Serial Connection: SerialConnection(0x32971c0) 
 INFO 2015-05-18T00:41:48.499 "ttyACM0" 115200 
DEBUG 2015-05-18T00:41:48.502 APMToolBar: new Serial Link Created 2 
DEBUG 2015-05-18T00:41:48.502 APMToolBar setConnection: false 
DEBUG 2015-05-18T00:41:48.503 APMToolBar: updateLinkDisplay "ttyACM0" : "115200"  connected: false 
DEBUG 2015-05-18T00:41:48.565 SCW: Link is disconnected 
DEBUG 2015-05-18T00:41:48.573 Baud rate: 115200 Expected: 4 
DEBUG 2015-05-18T00:41:53.168 "retrieve versionobject from server: http://firmware.diydrones.com/Tools/APMPlanner/apm_planner_version.json" 
DEBUG 2015-05-18T00:41:53.580 Downloading: 2780 / 2780 
DEBUG 2015-05-18T00:41:53.580 Downloading: 2780 / 2780 
DEBUG 2015-05-18T00:41:53.581 AutoUpdateCheck::httpFinished() 
DEBUG 2015-05-18T00:42:01.045 Dronshare: remind later

vooon, 2015-05-18:
Mavros desined to be just a bridge for FCU-ROS, not ROS-GCS. For GCS they provide transparent proxy, so it acts like direct FCU-GCS connection.
GCS messages even not passed to plugin system, and i don't want to change that (there lots of error prone).

So it you want to communicate ROS-GCS you should implement ROS node that mimic to FCU.
Mavros here not fit, but it's part **libmavconn** may be used.
As example you can look at [PX4 SITL mavlink node][1].

Other strategy may be used if you want to use real FCU (e.g. pixhawk) but react to few commands from GCS in ROS.
You may write your own **gcs_bridge** node with handling of that messages.
Look at [gcs_image_bridge.cpp][2].

[1]: https://github.com/PX4/Firmware/blob/master/src/platforms/ros/nodes/mavlink/mavlink.cpp
[2]: https://github.com/mavlink/mavros/blob/master/mavros_extras/src/gcs_image_bridge.cpp

MauroPagliarella, 2015-05-19:
I don't use an FCU. 
I control my robot by ROS (running on board), and I need to send some control commands to the robot from a ground control station. Moreover, I would like to receive robot telemetries (imu, encoder, gps etc.), by means of a control application running on a laptop,  so I thought to use for this purpose the GCS.  
I'll check the links in your answer.

Thank you very much for the clarification. 
Bye

vooon, 2015-06-28:
Updates?

MauroPagliarella, 2015-06-28:
I decided not to use the APM GCS, and developed a custom ground control station based on two tcp-ip links toward ROS. One link for TM messages and the other to send telecommands to my robot.
I created a ros-qt package, using the cmd "catkin_create_qt_package" and developed my ground control station using Qt 5.
At the moment it doesn't work very well. The TM messages, generated by my robot and sent to my ground control station, using the tcp-ip telemetry link, seem correctly sent (there is no telemetry packet lost), but the packet transmission appens at a rate minor w.r.t the rate of the packet generation. (The rate of the tm packet generation is 1pkt/sec while the rate of the received pkts is 1 packet every 2 seconds). Probably the tcp-ip socket accumulates the packets and sent them slowly. Now I'm investigating on this aspect. 

Do you have  any idea on the possible solutions?

Thank you very much
Mauro Pagliarella

vooon, 2015-06-28:
Use UDP, it is better for telemetry, but lossy.
But if you have IP telemetry, why not simply write RQT plugin (like dashboard) and connect to robot's roscore?
Than much simpler than reinventing protocol and write support nodes.

Anyway now it offtopic for mavros, so i close this issue.

MauroPagliarella, 2015-06-28:
I don't want to lose packets.
RQT seems to need ROS, and I would like to develop a ground control station running on a system without ros, e.g a tablet. 

I agree with you, my problem is not related to mavros, this issue can be closed.

Thank you for your help
Mauro

closed 2015-06-28 by vooon

vooon, 2018-10-24:
@AfshinMr what are you described is not transparent, as you want to modify messages coming trough proxy.
Mavros have build-in proxy that can be enabled by `gcs_url` parameter.
But to be able to modify messages it's easier to write your proxy that will decode and reencode messages back to mavlink stream.
I think that it should be relatively easy with pymavlink.

AfshinMr, 2018-10-25:
@vooon Thank you for your answer. In fact, I need to modify the messages if they are way-points or pass them without modification if they are not way-points.
can you give me a hint or any source code to do with pymavlink? or the steps I need to follow?
I appreciate it.

vooon, 2018-10-26:
@AfshinMr i don't have code, but what are you need is following

1. create 2 mavlink interfaces, one for FCU side and another on GCS
2. Check MSGID and if that is not `MISSION_*` pass from one link to another
3. In case of MISSION_* do whatever you want.

AfshinMr, 2018-10-29:
@vooon Thank you again for your answer.  I try to do that.

AfshinMr, 2018-10-31:
@vooon  Sorry Vladimir, I am completely confused when I tried to implement what you described. could you explain me in more details? I need to create an interface which listens to the GCS? Do I need to use a GCS like QGCS to send MAVLINK messages? if yes, How to connect them?
or I need to just create a MAVLINK message simulator to send the messages to the created interface?
I need another interface in FCU side to send te modified messages to the FCU. How to connect the two interfaces? I mean how to connect the interface in FCU side and the interface in GCS side? 
I am not sure if I got your description right ? Please explain in more details. I really appreciate your help.

vooon, 2018-11-06:
@AfshinMr i mean that you need instance that operate with udp/tcp/serial and parse and send messages.
I don't remember API of pymavlink, on libmavconn that will look like following (pseudocode):
```cpp

auto fcu_link = MAVConnInterface::open_url("serial:///dev/ttyACM0:57600");
auto gcs_link = MAVConnInterface::open_url("udp://localhost:14555@localhost:14550");

fcu_link->message_received_cb = [gcs_link] (const mavlink::mavlink_message_t *msg, const Framing framing) {
    gcs_link->send_message(msg);
};

gcs_link->message_received_cb = [fcu_link] (const mavlink::mavlink_message_t *msg, const Framing framing) {
    if (msg->msgid == OUR_MSGID) {
        DO_SOMETHING(msg);
        return;
    }
    // forward original message
    fcu_link->send_message(msg);
}
```

AfshinMr, 2018-11-09:
Thank you very much @vooon , appreciate your help. I try that :-)

AfshinMr, 2018-11-20:
Hi @vooon , 
I have created the interfaces as you described to me in Python. Here is the code:

          `gcs_conn = mavutil.mavlink_connection('tcpin:localhost:15795')
           gcs_conn.wait_heartbeat()
           print("Heartbeat from system (system %u component %u)" % (gcs_conn.target_system, gcs_conn.target_system))
           vehicle = mavutil.mavlink_connection('tcp:localhost:5760')
           vehicle.wait_heartbeat() # recieving heartbeat from the vehicle
           print("Heartbeat from system (system %u component %u)" % (vehicle.target_system, vehicle.target_system))
           while True:
                     gcs_msg = gcs_conn.recv_match()
                     if gcs_msg == None:
                            pass
                     else:
                            vehicle.mav.send(gcs_msg)
                            print(gcs_msg)

                     vcl_msg = vehicle.recv_match()
                     if vcl_msg == None:
                            pass
                     else:
                            gcs_conn.mav.send(vcl_msg)
                            print(vcl_msg)
`

First, Here I just want to receive and forward the messages !(without modification)
But, I still cannot send the messages from vehicle to the QGC ! and QGC does not recognize the vehicle !
Here is what I see when I run the code:
[image]

vooon, 2018-11-20:
I don't know, i don't use pymavlink for most time. Maybe you can find some hints in mavproxy, it's very powerful.

AfshinMr, 2018-11-21:
@vooon  Thank you for your response. You mean, I can use MAVProxy as a proxy that I explained for you?

vooon, 2018-11-21:
I don't know how hard it will be to add code to modify messages.
But it can be used as a good example on how to forward messages in pymavlink.