mavlink/mavros #298: Missing linear acceleration from /mavros/imu/data

MarcGyongyosi, 2015-05-19:
Hi,

so this is a weird problem that I haven't had so far: I suddenly cannot get the linear acceleration data on the /mavros/imu/data topic (/mavros/imu/data_raw is empty). I reflashed the PX4 firmware with the latest stable release and also recompiled mavros with the latest release. The system companion parameter is set to 921600 and I am connected via UART on Telem2. 

Does anyone have any insights into this? Thanks!

MarcGyongyosi, 2015-05-19:
Sorry about that. I had switched my SD cards and the streams on the one I was using was sending over the wrong port..........

closed 2015-05-19 by MarcGyongyosi

ahmadalahmar, 2024-09-26:
Hello, How did you solve this issue? I am having the exact same problem.
@MarcGyongyosi